Koenkk/zigbee2mqtt #17861: to topic #17857

vpaulos, 2023-06-01:
### Link

https://github.com/Koenkk/zigbee2mqtt/issues/17857

### Database entry

'TS0601' '_TZE204_5toc8efa' 0xa4c138e467d721fb 0xA849

### Comments

while device workings, where some errors

vpaulos, 2023-06-01:
Logger: homeassistant.util.logging
Source: util/logging.py:168
First occurred: 15:59:51 (66 occurrences)
Last logged: 17:59:17

Exception in message_received when handling msg on 'zigbee2mqtt/m.proom.wf': '{"child_lock":"UNLOCK","current_heating_setpoint":50,"deadzone_temperature":50,"heat":"OFF","linkquality":54,"local_temperature":22.5,"local_temperature_calibration":-4,"max_temperature":450,"max_temperature_limit":450,"min_temperature_limit":50,"preset":"hold","preset_mode":"hold","program":null,"running_state":"idle","sensor":"OU","system_mode":"off"}' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 44, in wrapper msg_callback(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/sensor.py", line 292, in message_received _update_state(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/sensor.py", line 243, in _update_state payload = self._template(msg.payload, PayloadSentinel.DEFAULT) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 251, in async_render_with_possible_json_value rendered_payload = self._value_template.async_render_with_possible_json_value( ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 735, in async_render_with_possible_json_value return _render_with_context(self.template, compiled, **variables).strip() ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 2179, in _render_with_context return template.render(**kwargs) ^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/jinja2/environment.py", line 1301, in render self.environment.handle_exception() File "/usr/local/lib/python3.11/site-packages/jinja2/environment.py", line 936, in handle_exception raise rewrite_traceback_stack(source=source) File "", line 1, in top-level template code File "/usr/local/lib/python3.11/site-packages/jinja2/filters.py", line 870, in do_truncate if len(s) <= length + leeway: ^^^^^^ TypeError: object of type 'NoneType' has no len()
Exception in message_received when handling msg on 'zigbee2mqtt/m.proom.wf': '{"child_lock":"UNLOCK","current_heating_setpoint":50,"deadzone_temperature":50,"heat":"OFF","linkquality":58,"local_temperature":21,"local_temperature_calibration":-4,"max_temperature":450,"max_temperature_limit":450,"min_temperature_limit":50,"preset":"hold","preset_mode":"hold","program":null,"running_state":"idle","sensor":"OU","system_mode":"off"}' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 44, in wrapper msg_callback(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/sensor.py", line 292, in message_received _update_state(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/sensor.py", line 243, in _update_state payload = self._template(msg.payload, PayloadSentinel.DEFAULT) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 251, in async_render_with_possible_json_value rendered_payload = self._value_template.async_render_with_possible_json_value( ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 735, in async_render_with_possible_json_value return _render_with_context(self.template, compiled, **variables).strip() ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 2179, in _render_with_context return template.render(**kwargs) ^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/jinja2/environment.py", line 1301, in render self.environment.handle_exception() File "/usr/local/lib/python3.11/site-packages/jinja2/environment.py", line 936, in handle_exception raise rewrite_traceback_stack(source=source) File "", line 1, in top-level template code File "/usr/local/lib/python3.11/site-packages/jinja2/filters.py", line 870, in do_truncate if len(s) <= length + leeway: ^^^^^^ TypeError: object of type 'NoneType' has no len()
Exception in message_received when handling msg on 'zigbee2mqtt/m.proom.wf': '{"child_lock":"UNLOCK","current_heating_setpoint":50,"deadzone_temperature":50,"heat":"OFF","linkquality":54,"local_temperature":21,"local_temperature_calibration":-4,"max_temperature":450,"max_temperature_limit":450,"min_temperature_limit":50,"preset":"hold","preset_mode":"hold","program":null,"running_state":"idle","sensor":"OU","system_mode":"off"}' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 44, in wrapper msg_callback(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/sensor.py", line 292, in message_received _update_state(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/sensor.py", line 243, in _update_state payload = self._template(msg.payload, PayloadSentinel.DEFAULT) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 251, in async_render_with_possible_json_value rendered_payload = self._value_template.async_render_with_possible_json_value( ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/src/homeassistant/homeassis [... cut ...] ^^^^^^ File "/usr/local/lib/python3.11/site-packages/jinja2/environment.py", line 1301, in render self.environment.handle_exception() File "/usr/local/lib/python3.11/site-packages/jinja2/environment.py", line 936, in handle_exception raise rewrite_traceback_stack(source=source) File "", line 1, in top-level template code File "/usr/local/lib/python3.11/site-packages/jinja2/filters.py", line 870, in do_truncate if len(s) <= length + leeway: ^^^^^^ TypeError: object of type 'NoneType' has no len()

buddy31130, 2023-06-14:
Same for me

buddy31130, 2023-07-02:
still having this error popping more than 10,000 times a day, even with latest update

Koenkk, 2023-07-02:
`Moes BHT-002: Unrecognized DP` will now be logged to `debug` instead of `warn`.

buddy31130, 2023-08-01:
sorry to put it back in front again, but the error is still showing up in the logs 
[image]
Even with the latest update (1.32.2-1)

Koenkk, 2023-08-02:
Ah sorry, this issue contains 2 issues:

```
2023-06-01 17:57:04zigbee-herdsman-converters:Moes BHT-002: Unrecognized DP https://github.com/Koenkk/zigbee2mqtt/issues/102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,210]}}- some errors
```
-> This one is fixed in z2m 1.32.2

```
TypeError: object of type 'NoneType' has no len()
```
-> I just pushed a fix for this one (this is the one that shows in HA)

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-09-09 by github-actions[bot]