Koenkk/zigbee2mqtt #17865: TuYa TS011F_plug_1 | Zigbee2MQTT not updating exposed power (low power devices only)

b2Homard, 2023-06-01:
### What happened?

Hello there,

The exposed "Power" value is not updated properly when I draw a small amount of power (e.g. 6W in my case). However, the current is updated and exposed correctly.

Take a look at this specific plug in the logs : "zigbee2mqtt/0xa4c138cf1e1a2f79"

Please note that this is a software issue, since i was able to read these small power draws before the last update (today), using the Dev Console.

Here are 2 screenshots, with the lamp OFF and ON.
[image]
[image]

### What did you expect to happen?

The exposed power should match the exposed current (multiplied by the voltage).

### How to reproduce it (minimal and precise)

Use a low power device (small LED lamp). Observe the current updating and the power staying at 0W.

### Zigbee2MQTT version

 1.31.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff | Zigbee 3.0 | Dongle Plus-E

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/11629798/log.txt)

v1k70rk4, 2023-06-02:
Please set the precision to 3 for everything in the settings (specific) menu. It just rounds it down to 0 because the ampere is so small.

v1k70rk4, 2023-06-02:
@b2Homard Please set the precision to 3 for everything in the settings (specific) menu. It just rounds it down to 0 because the ampere is so small.

b2Homard, 2023-06-02:
> @b2Homard Please set the precision to 3 for everything in the settings (specific) menu. It just rounds it down to 0 because the ampere is so small.

Hello,
I already did that, it doesn't change anything about the exposed power. But again, everything works like a charm with the exposed current.

I could use that to run my automations but I would like to understand why the exposed power wont show these 6-7W, since it was working before the update!

sakirsek, 2023-06-04:
After the last update it started showing inconsistent amperage value on my devices. It mostly shows zero.

v1k70rk4, 2023-06-05:
Try this, it solved my problem once.

[image]

sakirsek, 2023-06-05:
> Let's say @sakirsek's problem is different, I don't even know what might be the issue. Maybe try completely removing and re-pairing the device.

I tried, still the same. With version 1.31.0, TS011F_plug_1 devices started to work very inconsistently. I have 6 TS011F_plug_1 devices and they are all the same. It also gives wrong values.

oneseventhree, 2023-06-07:
I am also getting wrong values - I turn the plug off and it says 4 watts even when nothing is plugged into it. The values are updating much slower too for power

oneseventhree, 2023-06-16:
does anyone know if the new update fixes this? I'm too scared to update haha

sakirsek, 2023-06-16:
> does anyone know if the new update fixes this? I'm too scared to update haha

don't update. the problem still continues. Does anyone know if it's possible to downgrade?

oneseventhree, 2023-06-16:
> > does anyone know if the new update fixes this? I'm too scared to update haha
> 
> don't update. the problem still continues. Does anyone know if it's possible to downgrade?

I downgraded by selecting an old HA back up and then selecting partial restore and unselecting everything but Z2M

Koenkk, 2023-06-17:
- Please post the data/database.db entry of your device.
- If the issue about no 0 showing up, please post the debug log of having a load attached to the plug and removing the load (so power should be 0)

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

oneseventhree, 2023-06-17:
> * Please post the data/database.db entry of your device.

where do I find this?

> * If the issue about no 0 showing up, please post the debug log of having a load attached to the plug and removing the load (so power should be 0)
> 
> See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

Uploaded the log here:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/11778374/log.txt)

The plug goes to 0 when I unplug the load:
debug 2023-06-17 18:27:40: Received Zigbee message from 'PC SmartPlug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":0}' from endpoint 1 with groupID 0

Koenkk, 2023-06-17:
I see quickly after a load is added again (I guess you expect power 0 to reported after `{"rmsCurrent":0}`) but this can take some time (10-30 seconds), can you unplug the load for longer?

```
debug 2023-06-17 18:27:40: Received Zigbee message from 'PC SmartPlug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":0}' from endpoint 1 with groupID 0
...
17:51:58: Received Zigbee message from 'PC SmartPlug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":121}'
```

oneseventhree, 2023-06-18:
> I see quickly after a load is added again (I guess you expect power 0 to reported after `{"rmsCurrent":0}`) but this can take some time (10-30 seconds), can you unplug the load for longer?
>
> ``` [...]

I actually never connected the load again - its still disconnected.

The new log is with the load getting connected then I wait exactly one minute and disconnect it then I wait exactly one more minute and reconnect it - all by having the PC off.
I also waited a few minutes after to send this:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/11780249/log.txt)

Also during this time my Fan SmartPlug was showing 0watts with the fan connected with load but the fan not running with my other smartplugs(not zigbee) it would show 0.3w-2w.
The values are wrong when the fan is running. 
Hopefully the other people on here experiencing the same issue can show their logs as well

Koenkk, 2023-06-18:
> then I wait exactly one minute and disconnect it then I wait exactly one more minute and reconnect it - all by having the PC off.

Can you add annotations in the log file on where you did this?

oneseventhree, 2023-06-18:
> > then I wait exactly one minute and disconnect it then I wait exactly one more minute and reconnect it - all by having the PC off.
> 
> Can you add annotations in the log file on where you did this?

No problems - I edited out the other devices:

```txt
#### LOAD CONNECTED ####
debug 2023-06-18 09:54:39: Received Zigbee message from 'PC SmartPlug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":114}' from endpoint 1 with groupID 0
info  2023-06-18 09:54:39: MQTT publish: topic 'zigbee2mqtt/PC SmartPlug', payload '{"child_lock":"UNLOCK","current":0.06,"energy":17.56,"indicator_mode":"off/on","linkquality":160,"power":0,"power_outage_memory":"off","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":243}'

#### LOAD DISCONNECTED ####
debug 2023-06-18 09:55:40: Received Zigbee message from 'PC SmartPlug', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
debug 2023-06-18 09:55:40: Successfully pinged 'PC SmartPlug' (attempt 1/2)

debug 2023-06-18 09:55:44: Received Zigbee message from 'PC SmartPlug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":0}' from endpoint 1 with groupID 0

info  2023-06-18 09:55:55: MQTT publish: topic 'zigbee2mqtt/PC SmartPlug', payload '{"child_lock":"UNLOCK","current":0,"energy":17.56,"indicator_mode":"off/on","linkquality":116,"power":0,"power_outage_memory":"off","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":243}'

#### LOAD CONNECTED AND LEFT FOR A FEW MORE MINUTES ####
debug 2023-06-18 09:56:45: Received Zigbee message from 'PC SmartPlug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":114}' from endpoint 1 with groupID 0
info  2023-06-18 09:56:45: MQTT publish: topic 'zigbee2mqtt/PC SmartPlug', payload '{"child_lock":"UNLOCK","current":0.06,"energy":17.56,"indicator_mode":"off/on","linkquality":153,"power":0,"power_outage_memory":"off","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":243}'
debug 2023-06-18 09:56:45: Received Zigbee message from 'PC SmartPlug', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
debug 2023-06-18 09:56:45: Successfully pinged 'PC SmartPlug' (attempt 1/2)
debug 2023-06-18 09:56:57: Received Zigbee message from 'PC SmartPlug', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,1756]}' from endpoint 1 with groupID 0
info  2023-06-18 09:56:57: MQTT publish: topic 'zigbee2mqtt/PC SmartPlug', payload '{"child_lock":"UNLOCK","current":0.06,"energy":17.56,"indicator_mode":"off/on","linkquality":149,"power":0,"power_outage_memory":"off","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":243}'
debug 2023-06-18 09:56:58: Received Zigbee message from 'PC SmartPlug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":4,"rmsCurrent":121,"rmsVoltage":243}' from endpoint 1 with groupID 0
info  2023-06-18 09:56:58: MQTT publish: topic 'zigbee2mqtt/PC SmartPlug', payload '{"child_lock":"UNLOCK","current":0.06,"energy":17.56,"indicator_mode":"off/on","linkquality":145,"power":4,"power_outage_memory":"off","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":243}'
\debug 2023-06-18 09:57:03: lumi.motion.ac02: Processed data into payload {}

debug 2023-06-18 09:57:21: Received Zigbee message from 'PC SmartPlug', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID 0

debug 2023-06-18 09:58:26: Received Zigbee message from 'PC SmartPlug', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
debug 2023-06-18 09:58:26: Successfully pinged 'PC SmartPlug' (attempt 1/2)
info  2023-06-18 09:58:37: MQTT publish: topic 'zigbee2mqtt/0x00158d0006ed82f5', payload '{"action":null,"angle":13,"angle_x":5,"angle_x_absolute":85,"angle_y":-1,"angle_y_absolute":91,"angle_z":85,"battery":100,"device_temperature":22,"linkquality":123,"power_outage_count":236,"sensitivity":null,"strength":10,"vibration":false,"voltage":3005}'
```

Koenkk, 2023-06-19:
I don't understand the problem, the current goes to 0 correctly?

```

#### LOAD DISCONNECTED ####
info  2023-06-18 09:55:55: MQTT publish: topic 'zigbee2mqtt/PC SmartPlug', payload '{"child_lock":"UNLOCK","current":0,"energy":17.56,"indicator_mode":"off/on","linkquality":116,"power":0,"power_outage_memory":"off","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":243}'

```

oneseventhree, 2023-06-20:
> I don't understand the problem, the current goes to 0 correctly?
>
> ```
>
> #### LOAD DISCONNECTED ####
> info  2023-06-18 09:55:55: MQTT publish: topic 'zigbee2mqtt/PC SmartPlug', payload [...]

First of all, thank you so much for helping us :)

When the load is disconnected it goes to 0 which is fine in the latest update but the OP's post has been about showing the wrong value when at lower power values.

For me my computer is off but showing 4w. My other wifi plugs was showing much less like 1w

With my fan connected but not turned on I would get 0.3w-2w but it shows 0 watts now.
I know its drawing more power than 0 watts

Koenkk, 2023-06-20:
> For me my computer is off but showing 4w. My other wifi plugs was showing much less like 1w

- Did this work in any earlier z2m version? (I don't know of these plugs can measure that accurately)
- If yes, can you provide the debug log of this?

oneseventhree, 2023-06-22:
> > For me my computer is off but showing 4w. My other wifi plugs was showing much less like 1w
> 
> * Did this work in any earlier z2m version? (I don't know of these plugs can measure that accurately)

> * If yes, can you provide the debug log of this?

I am not sure - I've only just got these plugs a few weeks ago when the problems the OP post started.

I can see the polling is every 5 minutes - is there a way to speed this up so I can properly monitor what watts are showing?

sakirsek, 2023-06-22:
hey @Koenkk , the case of showing wrong current value continues. wattage is correct but current is wrong. 
It should be 2A but it shows 1A. 
5v * 2A = 11W is correct.

Zigbee2MQTT version 1.31.2

[image]
[image]
[image]

sakirsek, 2023-06-22:
> @sakirsek can you confirm the current is always divided by 2 (also when connecting higher loads to it?) Then we can just multiply the current by 2
> 
> Also, please provide the data/database.db entry of your device

I'm not sure about the current is always divided by 2. In the coming days, I can measure with different adapters and different multimeters and give more detailed information.

[database.txt](https://github.com/Koenkk/zigbee2mqtt/files/11839506/database.txt)

sakirsek, 2023-06-23:
> @sakirsek can you confirm the current is always divided by 2 (also when connecting higher loads to it?) Then we can just multiply the current by 2
> 
> Also, please provide the data/database.db entry of your device

I made different measurements with the power meter. I found that with the following specific settings it gives an approximate result.

Also, when calculating watt = volt * amp, I was always doing it according to DC. I learned that the current is calculated differently in AC. 😅 

You don't need to make any changes for my problem.

[image]
[image]
[image]
[image]

closed 2023-07-31 by github-actions[bot]