Koenkk/zigbee2mqtt #17867: zigbee2mqtt 1.31.0 failed to start after the latest update.

choinw, 2023-06-02:
### What happened?

The add-on itself failed to start with the error below shown in log

/app/node_modules/zigbee-herdsman-converters/index.js:79
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:79:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)

### What did you expect to happen?

wake up normally.

### How to reproduce it (minimal and precise)

reproduction rate 100%

### Zigbee2MQTT version

1.31.0-1

### Adapter firmware version

20201127

### Adapter

sonoff CC2561

### Debug log

/app/node_modules/zigbee-herdsman-converters/index.js:79
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:79:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)

kufischer-hacs, 2023-06-02:
Same problem here. 
Disabling
`external_converters:`
in 
`configuration.yaml` 
allows for a normal start. Maybe need updated converters?

Martin-Dolan, 2023-06-02:
Same here. If I comment out the external converter for my light switch, if will start.

choinw, 2023-06-02:
Startup problem resolved by disabling external converter from 'configuration.yaml' file.
Then what should users do to use external converter configuration in the new version?

Koenkk, 2023-06-02:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/17863#issuecomment-1573279515

closed 2023-06-02 by Koenkk