Koenkk/zigbee2mqtt #17870: ZG-204ZL not reporting LUX value

aaronpowell, 2023-06-02:
### What happened?

I've added a few [ZG-204ZL](https://www.zigbee2mqtt.io/devices/ZG-204ZL.html) devices to my network and while they are reporting occupancy correctly, the illuminance value is reported as `null` (or `N/A` in the HA add-on UI):

[image]

Here's the message in MQTT Explorer:

```json
{
  "battery": 100,
  "illuminance": null,
  "keep_time": "10",
  "linkquality": 255,
  "occupancy": false,
  "sensitivity": "high"
}
```

Naturally this results in HA seeing it as Unknown and thus automations aren't able to be run using the value.

### What did you expect to happen?

I expect that the LUX value is reported from the sensors.

### How to reproduce it (minimal and precise)

1. Follow pairing instructions from https://www.zigbee2mqtt.io/devices/ZG-204ZL.html
2. Wait for device to pair and be configured
3. Observe the values in either the Z2M UI or via monitoring the MQTT streams
4. Cover sensor to simulate a change in LUX value
5. Observe `null` being reported

### Zigbee2MQTT version

1.31.0-1

### Adapter firmware version

0x26580700

### Adapter

ConBee2/RaspBee2

### Debug log

```
2023-06-02T04:37:08.398Z zigbee-herdsman:deconz:frameParser response payload: 9,242,2,0,59,1,4,0,1,1
2023-06-02T04:37:08.404Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":242,"manufacturerCode":null,"commandIdentifier":2},"Payload":{"seq":15104,"dpValues":[{"dp":1,"datatype":4,"data":{"type":"Buffer","data":[1]}}]},"Command":{"ID":2,"parameters":[{"name":"seq","type":33},{"name":"dpValues","type":1011}],"name":"dataReport"}},"address":3667,"endpoint":1,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}'
2023-06-02T04:37:08.408Z zigbee-herdsman:controller:endpoint DefaultResponse 0xa4c1385c9624822f/1 61184(2, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2023-06-02T04:37:08.410Z zigbee-herdsman:deconz:adapter no response expected (242)
2023-06-02T04:37:08.410Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
Zigbee2MQTT:info  2023-06-02 14:37:08: MQTT publish: topic 'zigbee2mqtt/Stairs - top', payload '{"battery":100,"illuminance":null,"keep_time":"10","linkquality":255,"occupancy":false,"sensitivity":"high"}'
2023-06-02T04:37:08.425Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xe53 EP:1 SeqNr. 54 request id: 9
2023-06-02T04:37:08.430Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 9 status: 0
2023-06-02T04:37:08.430Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2023-06-02T04:37:08.438Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
2023-06-02T04:37:08.438Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2023-06-02T04:37:08.443Z zigbee-herdsman:deconz:driver query aps data confirm
2023-06-02T04:37:08.448Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 55
2023-06-02T04:37:08.451Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xe53 request id: 9 confirm status: 0
2023-06-02T04:37:08.451Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2023-06-02T04:37:08.451Z zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 242
2023-06-02T04:37:08.452Z zigbee-herdsman:deconz:adapter false, true, false, 10000
2023-06-02T04:37:08.452Z zigbee-herdsman:deconz:adapter resolve request (242)
2023-06-02T04:37:08.562Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2023-06-02T04:37:08.562Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2023-06-02T04:37:08.566Z zigbee-herdsman:deconz:driver query aps data indication
2023-06-02T04:37:08.571Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 56
2023-06-02T04:37:08.574Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 56 srcAddr: 0xe53 destAddr: 0x0 profile id: 0x104 cluster id: 0xef00 lqi: 255
2023-06-02T04:37:08.574Z zigbee-herdsman:deconz:frameParser response payload: 9,243,2,0,60,1,4,0,1,0
2023-06-02T04:37:08.580Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":243,"manufacturerCode":null,"commandIdentifier":2},"Payload":{"seq":15360,"dpValues":[{"dp":1,"datatype":4,"data":{"type":"Buffer","data":[0]}}]},"Command":{"ID":2,"parameters":[{"name":"seq","type":33},{"name":"dpValues","type":1011}],"name":"dataReport"}},"address":3667,"endpoint":1,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}'
2023-06-02T04:37:08.584Z zigbee-herdsman:controller:endpoint DefaultResponse 0xa4c1385c9624822f/1 61184(2, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2023-06-02T04:37:08.585Z zigbee-herdsman:deconz:adapter no response expected (243)
2023-06-02T04:37:08.585Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
Zigbee2MQTT:info  2023-06-02 14:37:08: MQTT publish: topic 'zigbee2mqtt/Stairs - top', payload '{"battery":100,"illuminance":null,"keep_time":"10","linkquality":255,"occupancy":true,"sensitivity":"high"}'
2023-06-02T04:37:08.592Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xe53 EP:1 SeqNr. 57 request id: 10
2023-06-02T04:37:08.595Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 10 status: 0
2023-06-02T04:37:08.596Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2023-06-02T04:37:08.604Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
2023-06-02T04:37:08.604Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2023-06-02T04:37:08.608Z zigbee-herdsman:deconz:driver query aps data confirm
2023-06-02T04:37:08.613Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 58
2023-06-02T04:37:08.616Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xe53 request id: 10 confirm status: 0
2023-06-02T04:37:08.616Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2023-06-02T04:37:08.616Z zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 243
2023-06-02T04:37:08.617Z zigbee-herdsman:deconz:adapter false, true, false, 10000
2023-06-02T04:37:08.617Z zigbee-herdsman:deconz:adapter resolve request (243)
2023-06-02T04:37:09.829Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 59
2023-06-02T04:37:09.832Z zigbee-herdsman:deconz:frameParser device state: 10100010
2023-06-02T04:37:09.832Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
```

aaronpowell, 2023-07-04:
Still having this with 1.32.0-2

closed 2023-08-13 by github-actions[bot]

Gil80, 2023-08-27:
Having the same exact problem.

outkastm, 2024-01-24:
Same problem and the battery stay always at 100%

Koenkk, 2024-01-26:
Can you try triggering the illuminance sensor by shining a light into it and provide the debug log of that?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

aaronpowell, 2024-01-30:
@Koenkk Done, here's what I see in the logs from when occupancy is detected through to cleared, and I was shining a torch at it:

```
Zigbee2MQTT:info  2024-01-30 11:26:23: MQTT publish: topic 'zigbee2mqtt/Stairs - top', payload '{"battery":100,"illuminance":null,"illuminance_interval":null,"keep_time":"10","linkquality":255,"occupancy":true,"sensitivity":"high"}'
2024-01-30T00:26:23.378Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x6f57 EP:1 SeqNr. 38 request id: 6
2024-01-30T00:26:23.383Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 6 status: 0
2024-01-30T00:26:23.383Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2024-01-30T00:26:23.435Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
2024-01-30T00:26:23.436Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2024-01-30T00:26:23.436Z zigbee-herdsman:deconz:driver query aps data confirm
2024-01-30T00:26:23.441Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 39
2024-01-30T00:26:23.443Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0x6f57 request id: 6 confirm status: 0
2024-01-30T00:26:23.444Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2024-01-30T00:26:23.444Z zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 99
2024-01-30T00:26:23.445Z zigbee-herdsman:deconz:adapter false, true, false, 10000
2024-01-30T00:26:23.445Z zigbee-herdsman:deconz:adapter resolve request (99)
2024-01-30T00:26:30.485Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 40
2024-01-30T00:26:30.487Z zigbee-herdsman:deconz:frameParser device state: 10100010
2024-01-30T00:26:30.488Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2024-01-30T00:26:40.487Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 41
2024-01-30T00:26:40.489Z zigbee-herdsman:deconz:frameParser device state: 10100010
2024-01-30T00:26:40.490Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2024-01-30T00:26:50.485Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 42
2024-01-30T00:26:50.487Z zigbee-herdsman:deconz:frameParser device state: 10100010
2024-01-30T00:26:50.487Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2024-01-30T00:26:53.376Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2024-01-30T00:26:53.376Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2024-01-30T00:26:53.378Z zigbee-herdsman:deconz:driver query aps data indication
2024-01-30T00:26:53.379Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 43
2024-01-30T00:26:53.381Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 43 srcAddr: 0x6f57 destAddr: 0x0 profile id: 0x104 cluster id: 0xef00 lqi: 255
2024-01-30T00:26:53.382Z zigbee-herdsman:deconz:frameParser response payload: 9,100,2,0,34,1,4,0,1,1
2024-01-30T00:26:53.388Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":100,"manufacturerCode":null,"commandIdentifier":2},"Payload":{"seq":8704,"dpValues":[{"dp":1,"datatype":4,"data":{"type":"Buffer","data":[1]}}]},"Command":{"ID":2,"parameters":[{"name":"seq","type":33},{"name":"dpValues","type":1011}],"name":"dataReport"}},"address":28503,"endpoint":1,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}'
2024-01-30T00:26:53.391Z zigbee-herdsman:controller:endpoint DefaultResponse 0xa4c1385c9624822f/1 61184(2, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2024-01-30T00:26:53.392Z zigbee-herdsman:deconz:adapter no response expected (100)
2024-01-30T00:26:53.392Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
Zigbee2MQTT:info  2024-01-30 11:26:53: MQTT publish: topic 'zigbee2mqtt/Stairs - top', payload '{"battery":100,"illuminance":null,"illuminance_interval":null,"keep_time":"10","linkquality":255,"occupancy":false,"sensitivity":"high"}'
2024-01-30T00:26:53.398Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x6f57 EP:1 SeqNr. 44 request id: 7
2024-01-30T00:26:53.401Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 7 status: 0
2024-01-30T00:26:53.401Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2024-01-30T00:26:53.489Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
2024-01-30T00:26:53.489Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[... cut ...]
2024-01-30T00:26:53.497Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0x6f57 request id: 7 confirm status: 0
2024-01-30T00:26:53.498Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2024-01-30T00:26:53.498Z zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 100
2024-01-30T00:26:53.498Z zigbee-herdsman:deconz:adapter false, true, false, 10000
2024-01-30T00:26:53.498Z zigbee-herdsman:deconz:adapter resolve request (100)
```

Koenkk, 2024-01-30:
I see it only reports datapoint 1, which is the `occupancy`, so not illuminance update indeed. Do you know how this device behaves with the TuYa hub?

aaronpowell, 2024-01-31:
I don't have a TuYa hub unfortunately, only a ConBee II stick

jammi, 2024-07-29:
I'm also having the same issue, only occupancy/clear, but no illuminance.

XRedShark, 2024-11-12:
Same problem here. Can I help you to send debug?

XRedShark, 2024-11-13:
> I'm also having the same issue, only occupancy/clear, but no illuminance.

https://github.com/Koenkk/zigbee2mqtt/issues/12678#issuecomment-1160448226

Koenkk, 2024-11-14:
@XRedShark I see illuminance in your screenshot? This shows the lux value

XRedShark, 2024-11-18:
> @XRedShark I see illuminance in your screenshot? This shows the lux value

The problem was that the entity needs to be activated. By default it is disabled. Problem solved.

aaronpowell, 2024-11-18:
> > @XRedShark I see illuminance in your screenshot? This shows the lux value
> 
> On Home Assistant the problem was that the entity needs to be activated. By default it is disabled. Problem solved.

How do you activate the entity?

XRedShark, 2024-11-19:
> > > @XRedShark I see illuminance in your screenshot? This shows the lux value
> > 
> > On Home Assistant the problem was that the entity needs to be activated. By default it is disabled. Problem solved.
> 
> How do you activate the entity?

In zigbee2mqtt, after pairing the entity are disable. I have enabled and after 30 sec. lux appear working

jammi, 2024-11-19:
> > @XRedShark I see illuminance in your screenshot? This shows the lux value
> 
> On Home Assistant the problem was that the entity needs to be activated. By default it is disabled. Problem solved.

Not my experience, and I fail to see how it's related unless your device worked out of the box, and only had HA-side entity disabled. I haven't had it disabled in HA and Zigbee2MQTT receives null state for it. In the HA entities, it's "Unknown".

[image]

Eaglefire1, 2024-12-24:
Same problem for me. Is there a solution? I trained two new color devices. both do not display lux values

pawel1830, 2025-01-03:
@XRedShark What version of Zigbee2MQTT do you have?

XRedShark, 2025-01-03:
> @XRedShark What version of Zigbee2MQTT do you have? 

1.42.0

pawel1830, 2025-01-03:
> > @XRedShark What version of Zigbee2MQTT do you have?
> 
> 1.42.0

I also have 1.42.0 And data in logs
info 2025-01-03 21:45:36z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/czujnik ruchu hol', payload '{"battery":100,"illuminance":null,"illuminance_interval":1,"keep_time":"30","linkquality":152,"occupancy":true,"sensitivity":"high"}'

Do you have any idea?

tlogik, 2025-01-05:
I have version 1.42.0-2 and the problem is still present.

pavleec, 2025-01-10:
I got the same problem with the following sensor. Only occupancy is properly reported and illuminance is always null. 
Running on the latest HA with Zigbee2MQTT v.2.0.0-2. 

```z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c13842c107cf24', payload '{"battery":100,"illuminance":null,"illuminance_interval":2,"keep_time":"10","linkquality":40,"occupancy":false,"sensitivity":"high"}'```

[image]

Endrut, 2025-01-15:
I have the same isssue, illuminance not reported by device.

m4ary, 2025-04-05:
still have

> > I got the same problem with the following sensor. Only occupancy is properly reported and illuminance is always null. Running on the latest HA with Zigbee2MQTT v.2.0.0-2.
> > `z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c13842c107cf24', payload '{"battery":100,"illuminance":null,"illuminance_interval":2,"keep_time":"10","linkquality":40,"occupancy":false,"sensitivity":"high"}'`
> > [image]
> 
> I have the same issue with Z2M 2.1.1-1. I have two new installation (NUC and RPI). Looks like translation is not configured.
> 
> When i use the devconsole to generated external definition i cannot see lux definition. Is this correct?
> 
> ```
> 
> const definition = {
>     zigbeeModel: ['TS0601'],
>     model: 'TS0601',
>     vendor: '_TZE200_bh3n6gk8',
>     description: 'Automatically generated definition',
>     extend: [m.iasZoneAlarm({"zoneType":"generic","zoneAttributes":["alarm_1","alarm_2","tamper","battery_low"]}), m.battery()],
>     meta: {},
> };
> 
> module.exports = definition;
> ```

i have exactly the same device and issue and no luck to fix it

tureka, 2025-06-05:
Anything changed in this matter? No luck for me either to run this sensor with luminance readings.

 _TZE200_bh3n6gk8 - marked as eWelink on the box, EZ-P1                                         - does not work 

_TZE200_3towulqd - marked as Tuya with Light Sensor on the box, TZ-P1-G_CA         - works well 

Both use the same zigbee-herdsman-converters fingerprint.

I've checked the latest dev branch (05.06.2025)

Any chance to take antoher look???

rmaher001, 2025-06-23:
@tureka I have a mixture of both types:
3 out of 5 _TZE200_3towulqd report
0 out of 2 _TZE200_bh3n6gk8 report