Koenkk/zigbee2mqtt #17871: Philips MasterConnect T8 -     LEDtube T8 5FT V2.0

tech2mar, 2023-06-02:
### Link

https://www.lighting.philips.com/main/prof/led-lamps-and-tubes/led-masterconnect/masterconnect-t8

### Database entry

{"id":120,"type":"Router","ieeeAddr":"0x0017880101937916","nwkAddr":31267,"manufId":4107,"manufName":"Philips Lighting","powerSource":"DC Source","modelId":"LEDtube T8 5FT V2.0","epList":[64,66,80,242],"endpoints":{"64":{"profId":260,"epId":64,"devId":257,"inClusterList":[0,3,21,21,4,5,6,8,769,1794,64771,1030,64770,64774],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"LEDtube T8 5FT V2.0","manufacturerName":"Philips Lighting","powerSource":4,"zclVersion":2,"appVersion":53,"stackVersion":0,"hwVersion":16,"dateCode":"20180318","swBuildId":"3.5.12"}}},"binds":[],"configuredReportings":[],"meta":{}},"66":{"profId":260,"epId":66,"devId":263,"inClusterList":[3,0,4],"outClusterList":[1030],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"80":{"profId":260,"epId":80,"devId":7,"inClusterList":[0,3,64768,64769],"outClusterList":[3,64768],"clusters":{"genBasic":{"attributes":{"stackVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":53,"stackVersion":0,"hwVersion":16,"dateCode":"20180318","swBuildId":"3.5.12","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1685605772483,"defaultSendRequestWhen":"immediate"}

### Comments

Hey guys, I found this in my house, after one of the tubes failed I replaced it with this one, at the same time I was configuring and pairing some devices, On the list there are 2 new devices that pop up that made me think, now I know what Philips are doing with these (smart) tubes, its funny because Philips don't documented it as a Zigbee certified product, they also sell a very expensive (gateway) that can connect a lot of these tubes in car parks, so basically they are Zigbee lights, very simple things. But maybe this could help someone.

Cheers and keep up the good work!
[image]

tech2mar, 2023-07-04:
Should the converter not being the same for the Osram substitube? That would be great.

tech2mar, 2023-08-05:
> ### Link
>
> https://www.lighting.philips.com/main/prof/led-lamps-and-tubes/led-masterconnect/masterconnect-t8
>
> ### Database entry [...]

@Koenkk do you know someone that could help here? I'm not that skilled in coding 😞.
I'm sorry that I can't help more then that.

Best regards

Koenkk, 2023-08-06:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/ef06ca2c555426ce374ba7efb1bd9abc

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

tech2mar, 2023-08-06:
Works like a charm, this is just ridiculous what Philips is doing, they are marketing it as a completely different service and "technology", is this even legal? Does the Zigbee standard not need to be mentioned on these products?

Koenkk, 2023-08-06:
Great, I've added the device!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-08-06 by Koenkk

tech2mar, 2023-08-06:
Thank you so much ✨