Koenkk/zigbee2mqtt #17875: Moes BHT series Thermostat exeptions in version 1.31.0

IgorSimic, 2023-06-02:
### What happened?

After update to version 1.31.0 my home assistant started log is flooded with folowing:
Exception in message_received when handling msg on 'zigbee2mqtt/0x5c0272fffedd2137': '{"child_lock":"LOCK","current_heating_setpoint":21,"deadzone_temperature":null,"heat":"OFF","last_seen":"2023-06-02T13:49:32+02:00","linkquality":185,"local_temperature":23,"local_temperature_calibration":null,"max_temperature_limit":null,"min_temperature_limit":null,"preset":"hold","preset_mode":"hold","program":null,"running_state":"idle","sensor":"IN","system_mode":"off"}' Traceback (most recent call last): File "/srv/hass311/lib/python3.11/site-packages/homeassistant/components/mqtt/debug_info.py", line 44, in wrapper msg_callback(msg) File "/srv/hass311/lib/python3.11/site-packages/homeassistant/components/mqtt/sensor.py", line 321, in message_received _update_state(msg) File "/srv/hass311/lib/python3.11/site-packages/homeassistant/components/mqtt/sensor.py", line 272, in _update_state payload = self._template(msg.payload, PayloadSentinel.DEFAULT) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/srv/hass311/lib/python3.11/site-packages/homeassistant/components/mqtt/models.py", line 251, in async_render_with_possible_json_value rendered_payload = self._value_template.async_render_with_possible_json_value( ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/srv/hass311/lib/python3.11/site-packages/homeassistant/helpers/template.py", line 716, in async_render_with_possible_json_value return _render_with_context(self.template, compiled, **variables).strip() ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/srv/hass311/lib/python3.11/site-packages/homeassistant/helpers/template.py", line 2164, in _render_with_context return template.render(**kwargs) ^^^^^^^^^^^^^^^^^^^^^^^^^ File "/srv/hass311/lib/python3.11/site-packages/jinja2/environment.py", line 1301, in render self.environment.handle_exception() File "/srv/hass311/lib/python3.11/site-packages/jinja2/environment.py", line 936, in handle_exception raise rewrite_traceback_stack(source=source) File "<template>", line 1, in top-level template code File "/srv/hass311/lib/python3.11/site-packages/jinja2/filters.py", line 870, in do_truncate if len(s) <= length + leeway: ^^^^^^ TypeError: object of type 'NoneType' has no len()

Device,0x5c0272fffedd2137 is Moes BHT series Thermostat TS0601.
Before this update I never got this message and my thermostat works without any problem.

### Zigbee2MQTT version

1.31.0

### Adapter firmware version

20221226

### Adapter

Slaesh's CC2652RB stick

DarwinData, 2023-06-03:
I am getting this exact same error after updating to 1.31.1.  It was ok prior to that.

Lots of errors, for example:
```
Logger: homeassistant.util.logging
Source: util/logging.py:168
First occurred: 7:43:36 AM (215 occurrences)
Last logged: 9:10:35 AM
```

IgorSimic, 2023-06-07:
Have new findings about the problem.
Updated to z2m 1.31.1 and same message in home assistant log keep appeared:

Exception in message_received when handling msg on 'zigbee2mqtt/0x5c0272fffedd2137': '{"child_lock":null,"current_heating_setpoint":21,"deadzone_temperature":null,"heat":"OFF","last_seen":"2023-06-07T17:42:17+02:00","linkquality":163,"local_temperature":23,"local_temperature_calibration":null,"max_temperature_limit":null,"min_temperature_limit":null,"preset":"hold","preset_mode":"hold","program":null,"running_state":"idle","sensor":null,"system_mode":"off"}' Traceback (most recent call last): File "/srv/hass311/lib/python3.11/site-packages/homeassistant/components/mqtt/debug_info.py", line 44, in wrapper msg_callback(msg) File "/srv/hass311/lib/python3.11/site-packages/homeassistant/components/mqtt/sensor.py", line 321, in message_received _update_state(msg) File "/srv/hass311/lib/python3.11/site-packages/homeassistant/components/mqtt/sensor.py", line 272, in _update_state payload = self._template(msg.payload, PayloadSentinel.DEFAULT) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/srv/hass311/lib/python3.11/site-packages/homeassistant/components/mqtt/models.py", line 251, in async_render_with_possible_json_value rendered_payload = self._value_template.async_render_with_possible_json_value( ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/srv/hass311/lib/python3.11/site-packages/homeassistant/helpers/template.py", line 716, in async_render_with_possible_json_value return _render_with_context(self.template, compiled, **variables).strip() ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/srv/hass311/lib/python3.11/site-packages/homeassistant/helpers/template.py", line 2164, in _render_with_context return template.render(**kwargs) ^^^^^^^^^^^^^^^^^^^^^^^^^ File "/srv/hass311/lib/python3.11/site-packages/jinja2/environment.py", line 1301, in render self.environment.handle_exception() File "/srv/hass311/lib/python3.11/site-packages/jinja2/environment.py", line 936, in handle_exception raise rewrite_traceback_stack(source=source) File "<template>", line 1, in top-level template code File "/srv/hass311/lib/python3.11/site-packages/jinja2/filters.py", line 870, in do_truncate if len(s) <= length + leeway: ^^^^^^ TypeError: object of type 'NoneType' has no len()

Then I find out that program sensor in home assistant config device is unknown. Previously this sensor showed thermostat's program choosed.
I disabled this sensor and entries in log stopped comming.
Obviously, in 1.31.0 structure of message is changed and this somehow results with this problem.

closed 2023-06-07 by IgorSimic

IgorSimic, 2023-06-07:
Here is a screenshot.
________________________________________
Šalje: DarwinData ***@***.***>
Poslano: 7. lipnja 2023. 19:00
Prima: Koenkk/zigbee2mqtt
Kopija: IgorSimic; State change
Predmet: Re: [Koenkk/zigbee2mqtt] Moes BHT series Thermostat BHT-002-GCLZB (_TZE200_aoclfnxz) exceptions in version 1.31.0 (Issue #17875)

Sorry but what program sensor did you disable?

IgorSimic, 2023-06-07:
Here you can see screenshot:
https://community.home-assistant.io/t/moes-bht-series-thermostat-bht-002-gclzb-tze200-aoclfnxz-exceptions-in-zigbee2mqtt-v1-31-0-and-newer/579334

DarwinData, 2023-06-09:
Thanks, disabling the program sensor eliminated the error messages.  Is this really a 'fix' or a work around?  The underlying problem is still there so I wonder if this issue should stay open?

alejovea9, 2023-07-07:
> For me this is almost fix because I don't need thermostat's programs nor the program sensor. But yes, underlaying problem is probably still there.
> […](#)
> --- Igor Simic ---

I have exactrly the same issue, and and fix it by disabling that sensor, but I think this is a temporary solution

Jarikk, 2023-08-14:
Sorry, but why did you close the issue? The problem has not been fixed yet.
I thought the problem was related to the MQTT integration and created the ticket there first ([#98344](https://github.com/home-assistant/core/issues/98344)). But in the discussion it was found out that it is an issue of Zigbee2Mqtt.

The problem is in the value template for object "program" because it cannot handle NULL value correctly.
**{{ value_json.program|default("") | truncate(254, True, '', 0) }}**

`
2023-08-14 10:03:06.773 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"child_lock":null,"current_heating_setpoint":23,"deadzone_temperature":null,"heat":"OFF","last_seen":"2023-08-14T10:03:03+03:00","linkquality":58,"local_temperature":23,"local_temperature_calibration":null,"max_temperature_limit":null,"min_temperature_limit":null,"preset":"hold","preset_mode":"hold","program":null,"running_state":"idle","sensor":null,"system_mode":"off"}' with variables {'entity_id': 'sensor.0x60a423fffeb27fa1_program', 'name': 'program', 'this': <template TemplateStateFromEntityId(sensor.0x60a423fffeb27fa1_program)>} with default value 'default' and Template<template=({{ value_json.program|default("") | truncate(254, True, '', 0) }}) renders=3240>

2023-08-14 10:03:06.794 ERROR (MainThread) [homeassistant.util.logging] Exception in message_received when handling msg on 'zigbee2mqtt_livingroom/Balcony, thermostat': '{"child_lock":null,"current_heating_setpoint":23,"deadzone_temperature":null,"heat":"OFF","last_seen":"2023-08-14T10:03:03+03:00","linkquality":58,"local_temperature":23,"local_temperature_calibration":null,"max_temperature_limit":null,"min_temperature_limit":null,"preset":"hold","preset_mode":"hold","program":null,"running_state":"idle","sensor":null,"system_mode":"off"}'
Traceback (most recent call last):
  File "/lsiopy/lib/python3.11/site-packages/homeassistant/components/mqtt/debug_info.py", line 44, in wrapper
    msg_callback(msg)
  File "/lsiopy/lib/python3.11/site-packages/homeassistant/components/mqtt/sensor.py", line 293, in message_received
    _update_state(msg)
  File "/lsiopy/lib/python3.11/site-packages/homeassistant/components/mqtt/sensor.py", line 244, in _update_state
    payload = self._template(msg.payload, PayloadSentinel.DEFAULT)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/lsiopy/lib/python3.11/site-packages/homeassistant/components/mqtt/models.py", line 251, in async_render_with_possible_json_value
    rendered_payload = self._value_template.async_render_with_possible_json_value(
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/lsiopy/lib/python3.11/site-packages/homeassistant/helpers/template.py", line 735, in async_render_with_possible_json_value
    return _render_with_context(self.template, compiled, **variables).strip()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/lsiopy/lib/python3.11/site-packages/homeassistant/helpers/template.py", line 2179, in _render_with_context
    return template.render(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/lsiopy/lib/python3.11/site-packages/jinja2/environment.py", line 1301, in render
    self.environment.handle_exception()
  File "/lsiopy/lib/python3.11/site-packages/jinja2/environment.py", line 936, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "<template>", line 1, in top-level template code
  File "/lsiopy/lib/python3.11/site-packages/jinja2/filters.py", line 870, in do_truncate
    if len(s) <= length + leeway:
       ^^^^^^
TypeError: object of type 'NoneType' has no len()
`

Could you re-open the issue?