Koenkk/zigbee2mqtt #17883: [Wrong device]: TS011F_plug_1

jorgeveca, 2023-06-02:
### Link

https://www.amazon.com/dp/B0B9SR4LCL?ref=ppx_yo2ov_dt_b_product_details&th=1

### Model

SM-AZ713

### Description

Zigbee Water Valve, Tuya Smart Water Valve Shutoff

### Vendor

Tuya

### Picture (link)

https://m.media-amazon.com/images/I/61aVAPiL2zL._AC_SL1500_.jpg

### Database entry

{"id":48,"type":"Router","ieeeAddr":"0xa4c1389a4459e49a","nwkAddr":60056,"manufId":4417,"manufName":"_TZ3000_tvuarksa","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":>

jorgeveca, 2023-06-02:
Tuya Smart Valve TS011F being detected as smart plug after last update to 1.31.0.

Before this update it was working as expected. Now, the image change to smart plug and I am seeing errors in the logs about getting state of the device.

Jun 02 11:00:46 Zigbee2MQTTPi4-PB npm[243375]: Zigbee2MQTT:error 2023-06-02 11:00:46: Failed to configure 'Llave Tanque del Agua', attempt 4 (Error: Bind 0xa4c1389a4459e49a/1 haElectricalMeasurement from '0x00212effff075e46/1' failed (Error: status: 130)
Jun 02 11:00:46 Zigbee2MQTTPi4-PB npm[243375]: at DeconzAdapter.bind (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:766:19)
Jun 02 11:00:46 Zigbee2MQTTPi4-PB npm[243375]: at Endpoint.bind (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:645:13)
Jun 02 11:00:46 Zigbee2MQTTPi4-PB npm[243375]: at Object.bind (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/reporting.js:40:9)
Jun 02 11:00:46 Zigbee2MQTTPi4-PB npm[243375]: at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/tuya.js:3170:13)
Jun 02 11:00:46 Zigbee2MQTTPi4-PB npm[243375]: at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:117:13)
Jun 02 11:00:46 Zigbee2MQTTPi4-PB npm[243375]: at Configure.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/configure.ts:55:21))
Jun 02 11:03:08 Zigbee2MQTTPi4-PB npm[243375]: Zigbee2MQTT:error 2023-06-02 11:03:08: Publish 'get' 'state' to 'Llave Tanque del Agua' failed: 'Error: Read 0xa4c1389a4459e49a/1 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'WRITE_ONLY')'

Does anyone know how to fix it? Thanks.

Koenkk, 2023-06-03:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-06-03 by Koenkk

jorgeveca, 2023-06-03:
Thank you very much for your help. Really appreciate it.

I will be checking when it is available to update.

jorgeveca, 2023-06-03:
It seems, the update on dev branch is not working, please see below. Do I have to wait more time?:

jcastro@Zigbee2MQTTPi4-PB:/opt/zigbee2mqtt $ git pull
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint:
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint:
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
Already up to date.

jorgeveca, 2023-06-04:
Jun 04 11:36:04 Zigbee2MQTTPi4-PB npm[268133]: Zigbee2MQTT:error 2023-06-04 11:36:04: Failed to configure 'Llave Tanque del Agua', attempt 1 (Error: Bind 0xa4c1389a4459e49a/1 haElectricalMeasurement from '0x00212effff075e46/1' failed (Error: status: 130)
Jun 04 11:36:04 Zigbee2MQTTPi4-PB npm[268133]:     at DeconzAdapter.bind (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:766:19)
Jun 04 11:36:04 Zigbee2MQTTPi4-PB npm[268133]:     at Endpoint.bind (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:645:13)
Jun 04 11:36:04 Zigbee2MQTTPi4-PB npm[268133]:     at Object.bind (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/reporting.js:40:9)
Jun 04 11:36:04 Zigbee2MQTTPi4-PB npm[268133]:     at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/tuya.js:3170:13)
Jun 04 11:36:04 Zigbee2MQTTPi4-PB npm[268133]:     at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:117:13)
Jun 04 11:36:04 Zigbee2MQTTPi4-PB npm[268133]:     at Immediate.<anonymous> (/opt/zigbee2mqtt/lib/extension/configure.ts:69:17))

It did nor work.

Thanks

jorgeveca, 2023-06-04:
not***

Koenkk, 2023-06-07:
Fix will be included in z2m 1.31.1 (will be released today)

jorgeveca, 2023-06-07:
Thanks for your confirmation!!!

jorgeveca, 2023-06-07:
It worked. Thanks again, Really appreciate your expertise on this.......

2302Dave, 2023-09-08:
I'm running version 1.33.0 and have a similar problem at the moment. I tried to add a TS011F_plug_1, but it recognizes it as a BSD29. Is there any way to fix this? Can the device be changed manually?