Koenkk/zigbee2mqtt #17886: Gledopto GL-C-009P shows as usupported

bramboot, 2023-06-02:
### What happened?

I tried to join two Gledopto GL-C-009P to my Zigbee network, but both show up as unsupported.

### What did you expect to happen?

I thought they would be recognized as the device was added in 2021
https://github.com/Koenkk/zigbee2mqtt/issues/6648

### Zigbee2MQTT version

1.18.1

### Adapter firmware version

0x26580700

### Adapter

ConBee II

Koenkk, 2023-06-03:
You are using an outdated z2m version.

If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

closed 2023-06-03 by Koenkk

bramboot, 2023-06-04:
Thank you. This is how it installed in HA. I will try to update to the latest.