Koenkk/zigbee2mqtt #17887: _TZ3000_dziaict4 doesn't send button number anymore

jomuc, 2023-06-02:
### What happened?

_TZ3000_dziaict4 is a 4 button and 3 state remote control

in release < 1.31.0 json messages were sent with action attribute having the syntax: [button-number]_[single/double/hold]
e.g. "action: "**1**_single" or "**4**_hold"

since 1.31.0 action doesn't contain any button numbers anymore. 

messages look like this:
```
{
    "battery": 100,
    "last_seen": "2023-06-02T19:52:24.793Z",
    "linkquality": 163,
    "voltage": 3000,
    "action": "double"
}
```

### What did you expect to happen?

action should have old format
or
message should have a new attribute with button number

### Zigbee2MQTT version

1.31.0

### Adapter firmware version

20221226

### Adapter

zStack3x0

jackchalat, 2023-06-03:
Can confirm. One interesting note is I have another same remote but it identifies as follow:

`Zigbee Model TS0044`
`Zigbee Manufacturer _TZ3000_ygvf9xzp`

This is working normally after the update. While the other that identifies as `_TZ3000_dziaict4` have the problem this issue describes.

Koenkk, 2023-06-03:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/17862

closed 2023-06-03 by Koenkk

rccoleman, 2023-06-03:
Came here to report the same for my TuYa TS0043 - even with the `latest-dev` container as of right now, it only reports "single" or "double", rather than "1_single" or "1_double".

Koenkk, 2023-06-03:
- Can you provide the data/database.db entry of your device?
- can you make a screenshot of the z2m frontend -> settings -> about page?

rccoleman, 2023-06-03:
Sure.  I confirmed that it has the change from https://github.com/Koenkk/zigbee-herdsman-converters/commit/6e91fbb2947d0d272df6941092b490c2fcf6d5bc by digging into the container.

`database.db` entry:

```
{"id":52,"type":"EndDevice","ieeeAddr":"0xa4c138eefd4d6ea6","nwkAddr":41372,"manufId":4417,"manufName":"_TZ3000_mjwg6a52","powerSource":"Battery","modelId":"TS0043","epList":[1,2,3,4],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[1,6,57344,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0011N\u0000\u0000\u0000\u0012","65506":49,"65508":1,"65534":0,"modelId":"TS0043","manufacturerName":"_TZ3000_mjwg6a52","powerSource":3,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":194,"batteryVoltage":29}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":0,"inClusterList":[6,1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":0,"inClusterList":[6,1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":0,"inClusterList":[6,1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1440156621},"lastSeen":1685807311290,"defaultSendRequestWhen":"immediate"}
```

[image]

Koenkk, 2023-06-04:
What if you add `console.log(msg.device)`?

rccoleman, 2023-06-04:
Yep, seems to work!

jomuc, 2023-06-04:
Sorry, was not at home over the weekend.
I tried zigbee-herdsman-converters 15.17.0 and it works also for my _TZ3000_dziaict4!
Perfect, thanks a lot for your quick support and fix.

Fougues, 2023-10-02:
the buttons go up in action but there are systematically 2 actions which go up at the same time.
One empty and another for example 1_single.

Except that the automation sees only one, either the empty one or the other (1_single) when it sees the empty one then nothing happens.

so you can press the button and see nothing happen. it's quite random.
I couldn't find a way to resolve this.