Koenkk/zigbee2mqtt #17888: Ikea E2103 TREDANSEN Incorrect Battery Reporting

FriedCheese2006, 2023-06-02:
### What happened?

After upgrading to firmware 24.4.11, all five E2103s I have are reporting what appears to be doubled battery values.

### What did you expect to happen?

No battery report over 100%

### How to reproduce it (minimal and precise)

Check battery report after upgrade.

### Zigbee2MQTT version

1.31.0

### Adapter firmware version

20210708

### Adapter

Sonoff Dongle P

### Debug log

[image]
[image]

Koenkk, 2023-06-03:
Pushed a possible fix, can you check if this is fixed now? 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

After updating wait 1 day to make the device report a new battery value, if the issue is not fixed provide me the data/database.db entry of this device.

MattWestb, 2023-06-03:
All released 24.X firmware is doing the batter reporting as Zigbee standard so all IKEA blinds have the same function then they is using the same firmware.
Also new and the old motion sensor and all remotes with 24.X firmware (Styrbar is having one 255 bug) shall not using doubling (i think some is fixed).

FriedCheese2006, 2023-06-03:
Looks like it may be good. I cycled them up/down to force a battery report.

[image]

Koenkk, 2023-06-04:
Great! I assume this can be closed now.

closed 2023-06-04 by Koenkk