Koenkk/zigbee2mqtt #17893: [New device support]:

rednet1, 2023-06-03:
### Link

https://lynktomydevice.com

### Database entry

{"id":31,"type":"EndDevice","ieeeAddr":"0xa4c1384db95f5c61","nwkAddr":15370,"manufId":4417,"manufName":"_TZ3040_bb6xaihh","powerSource":"Battery","modelId":"TS0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,3,4,1280,0],"outClusterList":[4,6,4096,25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0202","manufacturerName":"_TZ3040_bb6xaihh","powerSource":3,"zclVersion":3,"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0xe0798dfffe719556","zoneState":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1685765705086,"defaultSendRequestWhen":"immediate"}

### Comments

Model type : TS0202
Variant : _TZ3040_bb6xaihh

regards

Anashost, 2023-06-05:
Already supported and fixed, will be added in next release.
Here's a working external converter by @Koenkk
https://github.com/Koenkk/zigbee2mqtt/issues/17364#issuecomment-1566715618

rednet1, 2023-06-05:
Work fine! Thank

closed 2023-06-05 by rednet1