Koenkk/zigbee2mqtt #17896: [New device support]:

Hammerhand17, 2023-06-03:
### Link

https://es.aliexpress.com/item/1005004608535149.html?spm=a2g0o.order_list.order_list_main.42.3c66194dLNcLzl&gatewayAdapt=glo2esp

### Database entry

{"id":56,"type":"EndDevice","ieeeAddr":"0xa4c138792c5afd8a","nwkAddr":10456,"manufId":4417,"manufName":"_TZE200_mgxy2d9f","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1685787025589,"defaultSendRequestWhen":"immediate"}

### Comments

Pretty new to adding new devices to Zigbee2MQTT, I've seen the instructions on "support new devices" and added a external converter, but only "linkquality" is correctly exposed.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0601'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: '_TZE200_mgxy2d9f', // Vendor model number, look on the device for a model number
    vendor: 'Tuya', // Vendor of the device (only used for documentation and startup logging)
    description: 'Passive Infrared Detector', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.occupancy, fz.battery, fz.ignore_basic_report, fz.ias_occupancy_alarm_1_report], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.occupancy(), e.battery_low(), e.battery(), e.linkquality(), e.tamper(), e.battery_voltage()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;
```

Hammerhand17, 2023-07-22:
Still very interested but can't figure out what I'm doing wrong in the configuration.

Hammerhand17, 2023-08-24:
I have finally make it work almost completely with this converter:

`
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_mgxy2d9f',
        },
    ],
    model: 'TS0601_new',
    vendor: 'TuYa',
    description: 'Sensor de Presencia',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [e.occupancy(), e.battery_low(), e.tamper(), e.battery(),
        exposes.enum('sensitivity', ea.ALL, ['low', 'medium', 'high']).withDescription('PIR sensor sensitivity'),
        exposes.enum('keep_time', ea.ALL, [30, 60, 120, 600, 1800]).withDescription('PIR keep time in seconds'),
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
		    [1, 'occupancy', tuya.valueConverter.trueFalse0],
            [3, 'battery_low', tuya.valueConverter.trueFalse0],
            [4, 'battery', tuya.valueConverter.raw],			
            [5, 'tamper', tuya.valueConverter.raw],
            [9, 'radar_sensitivity', tuya.valueConverter.raw],
            [10, 'keep_time', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;
`
Keep_time and radar_sensitivity doesn't work and I'm not sure if I can go further on my own.

closed 2023-10-01 by github-actions[bot]

jbltpcblt, 2023-12-21:
I just received the same type of device, have you made any progress in your research ?