Koenkk/zigbee2mqtt #17897: [New device support]: TS0202

milandzuris, 2023-06-03:
### Link

https://vi.aliexpress.com/item/1005005099780623.html

### Database entry

*

### Comments

Hi i have one TuYa TS0202 Motion Sensor and is supported and other is look same is not supported how fix this?
[image]
[image]

Koenkk, 2023-06-03:
`_TZ3040_bb6xaihh` is supported in z2m 1.31.0

closed 2023-06-03 by Koenkk

milandzuris, 2023-06-04:
> `_TZ3040_bb6xaihh` is supported in z2m 1.31.0

i have 1.31.0-1 and nothing

milandzuris, 2023-06-04:
[image]
i dont have settings, and bad image: https://i5.walmartimages.com/asr/022424c9-e5c9-4baa-a88a-7bdeff800248.2e0697d86d6ab382c8e1b04c1d653be9.jpeg

milandzuris, 2023-06-04:
@Koenkk

Anashost, 2023-06-05:
@milandzuris It's already supported/fixed, and will be available in the next zigbee2mqtt update.
For now you can use this external converter which will work perfectly:
https://github.com/Koenkk/zigbee2mqtt/issues/17364#issuecomment-1566715618