Koenkk/zigbee2mqtt #17901: IKEA Fyrtur roller blind E1757 constantly updating status after latest firmware 24.4.11

syncnj, 2023-06-03:
### What happened?

After updating to the latest firmware for IKEA Fyrtur roller blinds, E1757, the device itself is updating its status multiple times per second.

The firmware update stated:
```
◆ Optimised battery performance.
◆ Stability improvements.
```
However, any blinds that have been updated to latest version, are constantly sending the traffic of commandCheckIn, compared to other blinds that are almost silent unless acted upon.

It seems they are expecting something from the hub to enter some other mode? 

Maybe we should warn users against updating to latest firmware as it might drain the battery too fast?

### What did you expect to happen?

Latest firmware shouldn't generate too much noisy traffic on network

### How to reproduce it (minimal and precise)

Update to latest firmware 24.4.11: https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html

### Zigbee2MQTT version

1.31.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Zigbee 3.0 Dongle Plus - E (newer version)

### Debug log

```
Info 2023-06-03 13:19:21MQTT publish: topic 'zigbee2mqtt/Blind1', payload '{"battery":50,"last_seen":"2023-06-03T13:19:21-07:00","linkquality":144,"position":0,"state":"CLOSE","update":{"installed_version":604241937,"latest_version":604241937,"state":"idle"},"update_available":null}'
Debug 2023-06-03 13:19:21Received Zigbee message from 'Blind2', type 'commandCheckIn', cluster 'genPollCtrl', data '{}' from endpoint 1 with groupID 0
Info 2023-06-03 13:19:21MQTT publish: topic 'zigbee2mqtt/Blind2', payload '{"battery":38,"last_seen":"2023-06-03T13:19:21-07:00","linkquality":220,"position":85,"state":"OPEN","update":{"installed_version":604241937,"latest_version":604241937,"state":"idle"},"update_available":null}'
Info 2023-06-03 13:19:21MQTT publish: topic 'zigbee2mqtt/Blind2', payload '{"battery":38,"last_seen":"2023-06-03T13:19:21-07:00","linkquality":224,"position":85,"state":"OPEN","update":{"installed_version":604241937,"latest_version":604241937,"state":"idle"},"update_available":null}'
Debug 2023-06-03 13:19:21Received Zigbee message from 'Blind1', type 'commandCheckIn', cluster 'genPollCtrl', data '{}' from endpoint 1 with groupID 0
Info 2023-06-03 13:19:21MQTT publish: topic 'zigbee2mqtt/Blind1', payload '{"battery":50,"last_seen":"2023-06-03T13:19:21-07:00","linkquality":148,"position":0,"state":"CLOSE","update":{"installed_version":604241937,"latest_version":604241937,"state":"idle"},"update_available":null}'
Info 2023-06-03 13:19:21MQTT publish: topic 'zigbee2mqtt/Blind3', payload '{"battery":100,"last_seen":"2023-06-03T13:19:21-07:00","linkquality":176,"position":98,"state":"OPEN","update":{"installed_version":604241937,"latest_version":604241937,"state":"idle"},"update_available":null}'
Info 2023-06-03 13:19:21MQTT publish: topic 'zigbee2mqtt/Blind1', payload '{"battery":50,"last_seen":"2023-06-03T13:19:21-07:00","linkquality":148,"position":0,"state":"CLOSE","update":{"installed_version":604241937,"latest_version":604241937,"state":"idle"},"update_available":null}'
Debug 2023-06-03 13:19:21Received Zigbee message from 'Blind3', type 'commandCheckIn', cluster 'genPollCtrl', data '{}' from endpoint 1 with groupID 0
Info 2023-06-03 13:19:21MQTT publish: topic 'zigbee2mqtt/Blind3', payload '{"battery":100,"last_seen":"2023-06-03T13:19:21-07:00","linkquality":172,"position":98,"state":"OPEN","update":{"installed_version":604241937,"latest_version":604241937,"state":"idle"},"update_available":null}'
Debug 2023-06-03 13:19:21Received Zigbee message from 'Blind4', type 'commandCheckIn', cluster 'genPollCtrl', data '{}' from endpoint 1 with groupID 0
Info 2023-06-03 13:19:21MQTT publish: topic 'zigbee2mqtt/Blind4', payload '{"battery":76,"last_seen":"2023-06-03T13:19:21-07:00","linkquality":192,"position":0,"state":"CLOSE","update":{"installed_version":604241937,"latest_version":604241937,"state":"idle"},"update_available":null}'
```

Skeletorjus, 2023-06-04:
https://github.com/Koenkk/zigbee2mqtt/issues/17860

syncnj, 2023-06-04:
> #17860

Thanks! Updated the poll interval and the traffic is back to normal now!

closed 2023-06-04 by syncnj