Koenkk/zigbee2mqtt #17902: [New device support]: AVATTO ZWSM-16-2 2gang switch module

anharald, 2023-06-03:
### Link

https://www.aliexpress.com/item/1005003051876791.html?spm=a2g0o.order_list.order_list_main.10.754d5c5fDaM397

### Database entry

{"id":34,"type":"Router","ieeeAddr":"0xa4c13876767232c8","nwkAddr":32631,"manufId":4417,"manufName":"_TZ3000_mtnpt6ws","powerSource":"Mains (single phase)","modelId":"TS0002","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"kI\u000e,\u0012?J\u000e,\u0012","65506":56,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_mtnpt6ws","zclVersion":3,"appVersion":72,"modelId":"TS0002","powerSource":1}},"manuSpecificBosch":{"attributes":{"53251":"AAAAAgAA"}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":0}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00259088d4","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00259088d4","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1030697865},"lastSeen":1685816569289,"defaultSendRequestWhen":"immediate"}

### Comments

device gets fingerprinted in Z2M as this device: https://www.zigbee2mqtt.io/devices/TS0002.html#tuya-ts0002
Problem is only switching the two channels is possible, other exposes are missing (power_on_behavior, switch_type)
The right device seems that one: https://www.zigbee2mqtt.io/devices/TS0002_switch_module.html
i already tried to use a external converter but it seems because the device gets already recognized from Z2M the external converter 
gets ignored :(
iam using the z2m addon in hassio so its really difficult to change devices.js local. 
is there anything i can do to force the use of the external converter file ? 
thanks in advance

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['_TZ3000_mtnpt6ws'],
    model: 'TS0002',
    vendor: 'Tuya',
    description: '2 gang Zigbee switch module',
        extend: tuya.extend.switch({switchType: true, endpoints: ['l1', 'l2']}),
        endpoint: (device) => {
            return {'l1': 1, 'l2': 2};
        },
        meta: {multiEndpoint: true},
        configure: async (device, coordinatorEndpoint, logger) => {
            await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
            await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
            await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        },
};
module.exports = definition;
```

Koenkk, 2023-06-04:
Fixed, thanks!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-06-04 by Koenkk

DaN66O, 2023-06-07:
Version 1.31.1commit: 3572c607 still has not been fixed

Koenkk, 2023-06-07:
Its available in the dev branch only (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

FeikoJoosten, 2023-09-05:
I've been trying to connect this device to my network but it doesn't even show up in the logs. 

Permit join is enabled. I'm running 1.33.0-dev commit: 91a9e32 <-- homeassistant addon. 
I factory reset the device to try and connect it to my network. 

I have multiple router devices surrounding the switch and the coordinator itself is quite nearby. Any ideas why it wouldn't even show in the logs? 

I've already tried another switch (I have 3 of these laying around) to see if it may be caused by the switch itself, but both aren't showing in the logs.