Koenkk/zigbee2mqtt #17906: _TZ3000_pnzfdr9y: power_on_behavior unsupported

Velines, 2023-06-04:
### What happened?

Hi,
the device is a LIDL/Silvercrest outdoor plug, Zigbee Model TS0101. It seems like it does not support setting the power_on_behavior:

`Publish 'set' 'power_on_behavior' to '0x9035eafffeb8f9d5' failed: 'Error: Write 0x9035eafffeb8f9d5/1 genOnOff({"moesStartUpOnOff":2}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`

### Zigbee2MQTT version

1.31.0

### Adapter firmware version

20210708

### Adapter

Slaesh's CC2652RB

closed 2023-07-12 by github-actions[bot]

zjean, 2023-09-02:
I have the same issue with two of these plugs. Anything I can do to help to get this fixed?

bertdeelman, 2023-12-20:
Hi,

I have the same issue. Anything I can do to help to get this fixed?

Regards,
Bert