Koenkk/zigbee2mqtt #17908: TS130F blind switch commands not sent to device

rbickel, 2023-06-04:
### What happened?

After a short while (minutes/hours) the commands are not sent anymore to TS130F (Girier: _TZ3210_dwytrmda). Sniffing the ZB network we can see that not message are sent to the device. The error is a MQTT timeout error (see logs belows)

The device is consistently broadcasting its link status every 15s. When opening/closing the device from the wall switch, it return the state accurately, without any issue.

### What did you expect to happen?

Commands sent and reaching the device

### Zigbee2MQTT version

1.31.0

### Adapter firmware version

6.7.9.0 build 405

### Adapter

SONIOFF ZBBridge EZSP v8

### Debug log

Herdsman:
2023-06-04T11:36:30.589Z zigbee-herdsman:adapter:ezsp:ezsp ==> setExtendedTimeout: {"remoteEui64":{"_value":{"type":"Buffer","data":[164,193,56,18,54,68,240,143]}},"extendedTimeout":true}
2023-06-04T11:36:30.589Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"setExtendedTimeout","_id_":126,"_isRequest_":true,"remoteEui64":{"_value":{"type":"Buffer","data":[164,193,56,18,54,68,240,143]}},"extendedTimeout":true}
2023-06-04T11:36:30.590Z zigbee-herdsman:adapter:ezsp:uart --> DATA (6,1,0): 9e00017e008ff044361238c1a401
2023-06-04T11:36:30.591Z zigbee-herdsman:adapter:ezsp:uart --> [61dc21a92a2a9a421da2581d6bf1934ab37e]
2023-06-04T11:36:30.592Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (7)
2023-06-04T11:36:30.628Z zigbee-herdsman:adapter:ezsp:uart <-- [17dca1a92a2ab3d67e]
2023-06-04T11:36:30.628Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (1,7,0): 17dca1a92a2ab3d67e
2023-06-04T11:36:30.629Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (2)
2023-06-04T11:36:30.629Z zigbee-herdsman:adapter:ezsp:uart --> [82503a7e]
2023-06-04T11:36:30.630Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 9e80017e00
2023-06-04T11:36:30.630Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x7e: {"_cls_":"setExtendedTimeout","_id_":126,"_isRequest_":false}
2023-06-04T11:36:30.630Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (7) success
2023-06-04T11:36:30.632Z zigbee-herdsman:adapter:ezsp:ezsp ==> sendUnicast: {"type":0,"indexOrDestination":25265,"apsFrame":{"clusterId":258,"profileId":260,"sequence":38,"sourceEndpoint":1,"destinationEndpoint":1,"groupId":0,"options":256},"messageTag":39,"message":{"type":"Buffer","data":[1,14,0]}}
2023-06-04T11:36:30.633Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"sendUnicast","_id_":52,"_isRequest_":true,"type":0,"indexOrDestination":25265,"apsFrame":{"clusterId":258,"profileId":260,"sequence":38,"sourceEndpoint":1,"destinationEndpoint":1,"groupId":0,"options":256},"messageTag":39,"message":{"type":"Buffer","data":[1,14,0]}}
2023-06-04T11:36:30.634Z zigbee-herdsman:adapter:ezsp:uart --> DATA (7,2,0): 9f0001340000b16204010201010100010000262703010e00
2023-06-04T11:36:30.634Z zigbee-herdsman:adapter:ezsp:uart --> [72dd21a9602a15033b904b27ab5493499d4e278dcacd6685fde7337e]
2023-06-04T11:36:30.635Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (0)
2023-06-04T11:36:30.682Z zigbee-herdsman:adapter:ezsp:uart <-- [20dda1a9602a15f6ac287e]
2023-06-04T11:36:30.682Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (2,0,0): 20dda1a9602a15f6ac287e
2023-06-04T11:36:30.682Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (3)
2023-06-04T11:36:30.682Z zigbee-herdsman:adapter:ezsp:uart --> [83401b7e]
2023-06-04T11:36:30.684Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 9f800134000044
2023-06-04T11:36:30.685Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x34: {"_cls_":"sendUnicast","_id_":52,"_isRequest_":false,"status":0,"sequence":68}
2023-06-04T11:36:30.685Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (0) success
2023-06-04T11:36:32.278Z zigbee-herdsman:adapter:ezsp:uart <-- [30ddb1a96b2a15033b904b27ab]
2023-06-04T11:36:32.279Z zigbee-herdsman:adapter:ezsp:uart <-- [5493499c4e27efcaa86746807e]
2023-06-04T11:36:32.280Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (3,0,0): 30ddb1a96b2a15033b904b27ab5493499c4e27efcaa86746807e
2023-06-04T11:36:32.280Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (4)
2023-06-04T11:36:32.281Z zigbee-herdsman:adapter:ezsp:uart --> [8430fc7e]
2023-06-04T11:36:32.281Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 9f90013f0000b1620401020101010000000044276600
2023-06-04T11:36:32.288Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x3f: {"_cls_":"messageSentHandler","_id_":63,"_isRequest_":false,"type":0,"indexOrDestination":25265,"apsFrame":{"profileId":260,"clusterId":258,"sourceEndpoint":1,"destinationEndpoint":1,"options":0,"groupId":0,"sequence":68},"messageTag":39,"status":102,"message":{"type":"Buffer","data":[]}}
2023-06-04T11:36:33.301Z zigbee-herdsman:adapter:ezsp:uart <-- [40ddb1a90d2a7c4381]
2023-06-04T11:36:33.303Z zigbee-herdsman:adapter:ezsp:uart <-- [e1e0db554569790c8e26abede8367e]
2023-06-04T11:36:33.303Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (4,0,0): 40ddb1a90d2a7c4381e1e0db554569790c8e26abede8367e
2023-06-04T11:36:33.303Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (5)
2023-06-04T11:36:33.304Z zigbee-herdsman:adapter:ezsp:uart --> [8520dd7e]
2023-06-04T11:36:33.305Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 9f9001590069f1d875aafeff10fb3090c0010000
2023-06-04T11:36:33.305Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x59: {"_cls_":"incomingRouteRecordHandler","_id_":89,"_isRequest_":false,"source":61801,"longId":{"type":"Buffer","data":[48,251,16,255,254,170,117,216]},"lastHopLqi":144,"lastHopRssi":-64,"relay":{"type":"Buffer","data":[0]}}
2023-06-04T11:36:33.306Z zigbee-herdsman:adapter:ezsp:driv handleRouteRecord: nwk=61801, ieee=0����u�, lqi=144, rssi=-64, relays=

Logs:
[image]

closed 2023-07-15 by github-actions[bot]

LeFlairGoD, 2023-09-29:
Have the same problem, first ran problem-free, now the blinds no longer respond

rbickel, 2023-09-29:
> Have the same problem, first ran problem-free, now the blinds no longer respond

When my blind stops responding I have to re-pair them. Pressing 10 times the down button put the device in pairing mode. I don't know why but sometimes they disconnect and I have to re-pair them

LeFlairGoD, 2023-10-02:
> > Have the same problem, first ran problem-free, now the blinds no longer respond
> 
> When my blind stops responding I have to re-pair them. Pressing 10 times the down button put the device in pairing mode. I don't know why but sometimes they disconnect and I have to re-pair them

This works, but for me only for 1-2 days.
So unfortunately not a solution for me