Koenkk/zigbee2mqtt #17910: Ikea Fyrtur and Kadrilj reports status once a second after update

mattias1972, 2023-06-04:
### What happened?

After firmware update all my Fyrtur and Kadrilj blinds reports status once a second.

### Zigbee2MQTT version

1.31.0-1

### Adapter firmware version

20210708

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

sjorge, 2023-06-04:
Already fixed in the dev branch https://github.com/Koenkk/zigbee-herdsman-converters/commit/17a92fae4695fee39656b13e6fec534eae2786d4

You can manually write the value using the dev console in the frontend too if you cant get on dev.

mattias1972, 2023-06-04:
Perfekt, problem solved!

closed 2023-06-04 by mattias1972

segdy, 2023-06-06:
Hi @sjorge amazing this sounds promising!

My z2m runs on a RPiZero and CPU is at 100% so I have a very hard time to to set these via the console.

Is this in the main branch already? How can I confirm?

```
zigbee2mqtt@rfpi:~$ git pull
Already up to date.
zigbee2mqtt@rfpi:~$ git log -1
commit f440c860193a5daa71265f9b3245f78224e42c72 (HEAD -> master, tag: 1.31.0, origin/master, origin/HEAD)
Author: Koen Kanters <koenkanters94@gmail.com>
Date:   Thu Jun 1 11:06:39 2023 +0200

    chore(dev): release 1.31.0 (#17772)
zigbee2mqtt@rfpi:~$
```

segdy, 2023-06-06:
Additional question: For one, I mistakenly set `checkinIntervalMin` (I believe). Is this a permanent setting? Is there a way I can check if I messed something up and revert it?