Koenkk/zigbee2mqtt #17911: [New device support]: Müller Licht Smart Socket Tower Outdoor 404078

NSimpsonsK, 2023-06-04:
### Link

https://www.mueller-licht.de/produktinformationen/artikel/?q=404078

### Database entry

{"id":68,"type":"Router","ieeeAddr":"0xa4c13800aa404f2a","nwkAddr":54418,"manufId":4635,"manufName":"MLI","powerSource":"Mains (single phase)","modelId":"tint smart power strip","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,4096,4111],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"tint smart power strip","manufacturerName":"MLI","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"MPLUG01","deviceEnabled":1}},"genOnOff":{"attributes":{"onOff":0,"onTime":0}},"genIdentify":{"attributes":{"identifyTime":0}},"genGroups":{"attributes":{"nameSupport":0}},"genOta":{"attributes":{"upgradeServerId":"0xffffffffffffffff","manufacturerId":4635}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c9ea46","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"MPLUG01","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1466202208},"lastSeen":1685912617976,"defaultSendRequestWhen":"immediate"}

### Comments

404078 Identifies as 45391 - Its the same product, but used for outdoor with a diffrent look.

closed 2023-07-12 by github-actions[bot]

Sweti934, 2024-06-12:
Is the device above now supported?

NSimpsonsK, 2024-06-12:
I don't know how it is now but I use it with the wrong image and Product name and it works fine

Sweti934, 2024-06-23:
Ok thank you! Can you control both of the outputs separately ? Thanks !

NSimpsonsK, 2024-06-23:
Unfortunately you can only control 2 of the 4 sockets and from the two only booth on or booth off