Koenkk/zigbee2mqtt #17914: [Wrong device]: KTNNKG ZB1248-10A present as TuYa TS011F_plug_1

swex, 2023-06-05:
### Link

https://www.zigbee2mqtt.io/devices/ZB1248-10A.html

### Model

TS011F_plug_1

### Description

Smart plug (with power monitoring)

### Vendor

KTNNKG

### Picture (link)

https://www.zigbee2mqtt.io/devices/ZB1248-10A.html

### Database entry

{"id":6,"type":"Router","ieeeAddr":"0x842e14fffe0fd167","nwkAddr":59915,"manufId":4098,"manufName":"_TZ3000_8fdayfch","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,57344,57345],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005","65506":31,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_8fdayfch","powerSource":1,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightSwitch":1,"tuyaBacklightMode":1,"moesStartUpOnOff":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cc4a54","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cc4a54","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cc4a54","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1685934164725,"defaultSendRequestWhen":"immediate"}

### Notes

[this](https://www.zigbee2mqtt.io/devices/ZB1248-10A.html) device now detected as [another one](https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html#tuya-ts011f_plug_1). State endpoint works but as z2m trying to reach other endpoints and device acts slowly

Koenkk, 2023-06-05:
should be fixed now, can you confirm?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

swex, 2023-06-05:
its good now, thanks!

closed 2023-06-06 by Koenkk