Koenkk/zigbee2mqtt #17919: NAS-AB02B2 Alarm mqtt /set only accepts first parameter

weltmeyer, 2023-06-05:
### What happened?

When sending a /set to an "[NEO NAS-AB02B2](https://www.zigbee2mqtt.io/devices/NAS-AB02B2.html)" (alias _TZE204_t1blo2bj/TS0601) it only accepts the first parameter. Example:

```
{
    "melody": 3,
    "volume": "low",
    "alarm": false
}
```
Here only "melody" is changed. volume and alarm are not changed on the device.

When sending 3 consecutive messages with only 1 parameter each, they all are accepted.

### What did you expect to happen?

Accept all sent parameters

### How to reproduce it (minimal and precise)

as seen in "What happened"

### Zigbee2MQTT version

1.31.0

### Adapter firmware version

20221226

### Adapter

POE Zigbee Stick, tested also on slaesh's directly via USB

weltmeyer, 2023-06-05:
seems like stale/closed issue #12542, #12220

Nafanya10, 2023-07-31:
This problem still exists.
The issue the device has is it will randomly ignore commands.
I send each command (duration, melody, volume, alarm) one after another, but the random failures still happen.
Please help fix this issue.

Model: Alarm (NAS-AB02B2)
Manufacture: Zigbee TZE204_t1blo2bj
Zigbee2MQTT Current version: 1.32.1
Home Assistant Core 2023.7.2

This is my automation code and it always works differently, in most cases works not correctly (not all parameters change):
```

alias: Audible alarm - alarm
description: ""
trigger:
  - platform: state
    entity_id:
      - input_button.virtual_button_02_alarm
    enabled: true
condition: []
action:
  - service: mqtt.publish
    data:
      qos: 0
      retain: false
      topic: zigbee2mqtt/Audible alarm/set
      payload: "{\"volume\": \"medium\"}"
  - delay:
      hours: 0
      minutes: 0
      seconds: 0
      milliseconds: 40
  - service: mqtt.publish
    data:
      qos: 0
      retain: false
      topic: zigbee2mqtt/Audible alarm/set
      payload: "{\"melody\": 3}"
    enabled: true
  - delay:
      hours: 0
      minutes: 0
      seconds: 0
      milliseconds: 40
  - service: mqtt.publish
    data:
      qos: 0
      retain: false
      topic: zigbee2mqtt/Audible alarm/set
      payload: "{\"duration\": 15}"
  - delay:
      hours: 0
      minutes: 0
      seconds: 0
      milliseconds: 40
  - service: mqtt.publish
    data:
      qos: 0
      retain: false
      topic: zigbee2mqtt/Audible alarm/set
      payload: "{\"alarm\": \"true\"}"
mode: single

```

Nafanya10, 2023-08-01:
Today I changed the way of automation, but the new way is also not stable. Automation works unpredictably, some parameters are updated, and some remain the same.
The result of automation triggering is unpredictable, each time automation triggers differently.

Please help fix this issue.

My new code:

```

alias: Audible alarm - bells
description: ""
trigger:
  - platform: state
    entity_id:
      - input_button.virtual_button_04_bells
condition: []
action:
  - device_id: d756b4873254559f9d2c983ca2d10496
    domain: number
    entity_id: ae1aaf8d259d9deaaaae9a83da5e533a
    type: set_value
    value: 5
  - type: turn_on
    device_id: d756b4873254559f9d2c983ca2d10496
    entity_id: 8a614924a59f8a74253890b7561f421f
    domain: switch
  - device_id: d756b4873254559f9d2c983ca2d10496
    domain: select
    entity_id: 2ef0c1f8cedc25418acfaada3d7c0dc5
    type: select_option
    option: "1"
  - device_id: d756b4873254559f9d2c983ca2d10496
    domain: select
    entity_id: b400f1158ef903d7ab4dcbef00c2787b
    type: select_option
    option: medium
mode: single

```

closed 2023-09-09 by github-actions[bot]

Dreyk007, 2023-11-15:
Not stale. Same issue with this device

wojtekzw, 2024-03-20:
Issue is still present. Even sending command one by one does not always help. Commands are ignored randomly

iMartien, 2024-09-18:
Just wanted to add that I have the same issue. Using the following HomeAssistant script to modify the parameters:

```
  sequence:
    - service: number.set_value
      data:
        value: 28
      target:
        entity_id: number.woonkamer_sirene_duration
    - delay:
        seconds: 1
    - service: select.select_option
      data:
        option: 8
      target:
        entity_id: select.woonkamer_sirene_melody
    - delay:
        seconds: 1
    - service: switch.turn_on
      entity_id: switch.woonkamer_sirene_alarm
```

The added delays seem to help reduce the issue, but sometimes one or more values still do not get set. Also using this for an alarm a 2 second delay before the alarm sounds isn't ideal.

Zigbee2Mqtt version: 1.40.1
HomeAssistant: 2024.9.1
Conbee2 Zigbee stick