Koenkk/zigbee2mqtt #17920: when trying to change adapter from Sonoff E to Sonoff P i get error

denveronly, 2023-06-05:
### What happened?

I am using sonoff dongle E adapter, and recently want to try out P version, because some of the devices fail to connect (they do connect and immediatly disconnect on any action) this encounters on some sonoff and aqara switches, also some motion sensors fall out for some time.
So i wanted to try if this happens on P version because E is not fully supported.

I stoped the addon
Plugged new usb adapter
filled in serial port new /dev information
deleted string adapter:ezsp
and i get this error which is strange im sure that there are no networks with same PAN iD

E```
rror 2023-06-05 13:12:18: Failed to start zigbee
error 2023-06-05 13:12:18: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2023-06-05 13:12:18: Exiting...
error 2023-06-05 13:12:18: Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby
    at ZnpAdapterManager.beginCommissioning (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:347:23)
    at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:92:17)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)
```

### What did you expect to happen?

work normaly

### Zigbee2MQTT version

1.31.0-dev commit: fd1622b

### Adapter firmware version

 6.10.3.0 build 297

### Adapter

sonoff dongle E and P

denveronly, 2023-06-10:
im having this issue
zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
can someone help please

closed 2023-07-18 by github-actions[bot]

MadMirza, 2023-08-08:
Just a ping to keep this alive.
Also @denveronly i saw you mentioned you were running Zigbee dongle E decently for a while, but I am trying to setup mine and I am having pairing issues.
- How many devices did you have in your network? I Can't pair more than 28 devices. I can't even the same type of device that I paired on time.
- How was the responsiveness? I see the LQI is so much better using the E than a P but i did not really get the chance to see it in action.
Any help regarding this is greatly appreciated, I am trying to stick the dongle E since it looks future proof than the P version.