Koenkk/zigbee2mqtt #17921: [New device support]: TS0052 _TZ3000_vluyhemf

hACKrus, 2023-06-05:
### Link

https://aliexpress.ru/item/1005002784552404.html

### Database entry

{"id":23,"type":"Router","ieeeAddr":"0xa4c138ded54d0139","nwkAddr":65402,"manufId":4417,"manufName":"_TZ3000_vluyhemf","powerSource":"Mains (single phase)","modelId":"TS0052","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,8,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0052","manufacturerName":"_TZ3000_vluyhemf","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6,57345,8],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1685978620876,"defaultSendRequestWhen":"immediate"}

### Comments

I bought this double dimmer some time ago. The other day I noticed that it started to be recognized in zigbee2mqtt, but as an incorrect model, apparently after this issue: https://github.com/Koenkk/zigbee2mqtt/issues/17239

Please separate these two devices and add a double dimmer.
I have a Taya gateway, I found how to get Data Endpoint, but I couldn't build the converter myself.

Thank you very much!

Koenkk, 2023-06-06:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/aa8a92591513931c8c72ddee52026fe8

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

Koenkk, 2023-06-07:
- Unfortunately, no. Switching output from Z2M doesn't change actual light > can you provide the debug log of this?
- Pushing button on case change both "state_l1" and "state_l2", but not common "state". > can you also provide the debug log of this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

hACKrus, 2023-06-07:
ID of device: 0xa4c138ded54d0139

Changed both states and brightness in Z2M and pushed button on case.

`Zigbee2MQTT:info  2023-06-07 20:18:48: MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139/l1', payload '{"brightness":128,"state":"OFF"}'
Zigbee2MQTT:info  2023-06-07 20:18:48: MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139/l2', payload '{"brightness":127,"state":"OFF"}'
Zigbee2MQTT:debug 2023-06-07 20:19:10: Received MQTT message on 'zigbee2mqtt/0xa4c138ded54d0139/set' with data '{"state_l1":"ON"}'
Zigbee2MQTT:debug 2023-06-07 20:19:10: Publishing 'set' 'state' to '0xa4c138ded54d0139'
Zigbee2MQTT:info  2023-06-07 20:19:10: MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139', payload '{"brightness":254,"brightness_l1":128,"brightness_l2":127,"level_config":{"on_off_transition_time":50},"linkquality":93,"state":"OFF","state_l1":"ON","state_l2":"OFF"}'
Zigbee2MQTT:info  2023-06-07 20:19:10: MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139/l1', payload '{"brightness":128,"state":"ON"}'
Zigbee2MQTT:info  2023-06-07 20:19:10: MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139/l2', payload '{"brightness":127,"state":"OFF"}'
Zigbee2MQTT:debug 2023-06-07 20:19:12: Received MQTT message on 'zigbee2mqtt/0xa4c138ded54d0139/set' with data '{"brightness_l1":254,"state_l1":"ON"}'
Zigbee2MQTT:debug 2023-06-07 20:19:12: Publishing 'set' 'brightness' to '0xa4c138ded54d0139'
Zigbee2MQTT:debug 2023-06-07 20:19:12: Received Zigbee message from '0xa4c138ded54d0139', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-06-07 20:19:12: MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139', payload '{"brightness":254,"brightness_l1":128,"brightness_l2":127,"level_config":{"on_off_transition_time":50},"linkquality":93,"state":"OFF","state_l1":"ON","state_l2":"OFF"}'
Zigbee2MQTT:info  2023-06-07 20:19:12: MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139/l1', payload '{"brightness":128,"state":"ON"}'
Zigbee2MQTT:info  2023-06-07 20:19:12: MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139/l2', payload '{"brightness":127,"state":"OFF"}'
Zigbee2MQTT:info  2023-06-07 20:19:12: MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139', payload '{"brightness":254,"brightness_l1":254,"brightness_l2":127,"level_config":{"on_off_transition_time":50},"linkquality":93,"state":"OFF","state_l1":"ON","state_l2":"OFF"}'
Zigbee2MQTT:info  2023-06-07 20:19:12: MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139/l1', payload '{"brightness":254,"state":"ON"}'
Zigbee2MQTT:info  2023-06-07 20:19:12: MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139/l2', payload '{"brightness":127,"state":"OFF"}'
Zigbee2MQTT:debug 2023-06-07 20:19:13: Received MQTT message on 'zigbee2mqtt/0xa4c138ded54d0139/set' with data '{"brightness_l1":3,"state_l1":"ON"}'
Zigbee2MQTT:debug 2023-06-07 20:19:13: Publishing 'set' 'brightness' to '0xa4c138ded54d0139'
Zigbee2MQTT:debug 2023-06-07 20:19:13: Received Zigbee message from '0xa4c138ded54d0139', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-06-07 20:19:13: MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139', payload '{"brightness":254,"brightness_l1":254,"brightness_l2":127,"level_config":{"on_off_transition_time":50},"linkquality":87,"state":"OFF","state_l1":"ON","state_l2":"OFF"}'
Zigbee2MQTT:info  2023-06-07 20:19:13: MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139/l1', payload '{"brightness":254,"state":"ON"}'
Zigbee2MQTT:info  2023-06-07 20:19:13: MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139/l2', payload '{"brightness":127,"state":"OFF"}'
Zigbee2MQTT:info  2023-06-07 20:19:14: MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139', payload '{"brightness":254,"brightness_l1":3,"brightness_l2":127,"level_config":{"on_off_transition_time":50},"linkquality":87,"state":"OFF","state_l1":"ON","state_l2":"OFF"}'
Zigbee2MQTT:info  2023-06-07 20:19:14: MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139/l1', payload '{"brightness":3,"state":"ON"}'
Zigbee2MQTT:info  2023-06-07 20:19:14: MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139/l2', payload '{"brightness":127,"state":"OFF"}'
Zigbee2MQTT:debug 2023-06-07 20:19:14: Received MQTT message on 'zigbee2mqtt/0xa4c138ded54d0139/set' with data '{"brightness_l1":3,"state_l1":"OFF"}'
Zigbee2MQTT:debug 2023-06-07 20:19:14: Publishing 'set' 'brightness' to '0xa4c138ded54d0139'
Zigbee2MQTT:info  2023-06-07 20:19:14: MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139', payload '{"brightness":254,"brightness_l1":3,"brightness_l2":127,"level_config":{"on_off_transition_time":50},"linkquality":90,"state":"OFF","state_l1":"OFF","state_l2":"OFF"}'
Zigbee2MQTT:info  2023-06-07 20:19:14: MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139/l1', payload '{"brightness":3,"state":"OFF"}'
Zigbee2MQTT:info  2023-06-07 20:19:14: MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139/l2', payload '{"brightness":127,"state":"OFF"}'
Zigbee2MQTT:debug 2023-06-07 20:19:14: Received Zigbee message from '0x00178801040928cc', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1107324829,"imageType":265,"manufacturerCode":4107}' from endpoint 2 with groupID 0
Zigbee2MQTT:debug 2023-06-07 20:19:14: Device '0x00178801040928cc' requested OTA
Zigbee2MQTT:debug 2023-06-07 20:19:14: Check if update available for '0x00178801040928cc' (RWL021)
[... cut ...]
Zigbee2MQTT:info  2023-06-07 20:19:21: MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139', payload '{"brightness":254,"brightness_l1":3,"brightness_l2":60,"level_config":{"on_off_transition_time":50},"linkquality":87,"state":"OFF","state_l1":"OFF","state_l2":"OFF"}'
Zigbee2MQTT:info  2023-06-07 20:19:21: MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139/l1', payload '{"brightness":3,"state":"OFF"}'
Zigbee2MQTT:info  2023-06-07 20:19:21: MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139/l2', payload '{"brightness":60,"state":"OFF"}'
`

Koenkk, 2023-06-08:
Can you provide me the datapoints from the TuYa hub?

Koenkk, 2023-06-12:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/b92e425556fa48d1ca2cefaa3bcb4b47

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

hACKrus, 2023-06-12:
Now controls in place, but they don't change things. If brightness was changed - on/off will generate error. Physical button also stopped working.

`Debug 2023-06-12 19:57:03Received MQTT message on 'zigbee2mqtt/0xa4c138ded54d0139/set' with data '{"state_l1":"OFF"}'
Debug 2023-06-12 19:57:03Publishing 'set' 'state' to '0xa4c138ded54d0139'
Info 2023-06-12 19:57:03MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139', payload '{"brightness_l1":254,"brightness_l2":"ON","countdown_l1":null,"countdown_l2":null,"linkquality":87,"state_l1":"OFF","state_l2":"ON"}'
Info 2023-06-12 19:57:03MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139/l1', payload '{"brightness":254,"countdown":null,"state":"OFF"}'
Info 2023-06-12 19:57:03MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139/l2', payload '{"brightness":"ON","countdown":null,"state":"ON"}'
Debug 2023-06-12 19:57:04Received MQTT message on 'zigbee2mqtt/0xa4c138ded54d0139/set' with data '{"state_l1":"ON"}'
Debug 2023-06-12 19:57:04Publishing 'set' 'state' to '0xa4c138ded54d0139'
Info 2023-06-12 19:57:04MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139', payload '{"brightness_l1":254,"brightness_l2":"ON","countdown_l1":null,"countdown_l2":null,"linkquality":87,"state_l1":"ON","state_l2":"ON"}'
Info 2023-06-12 19:57:04MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139/l1', payload '{"brightness":254,"countdown":null,"state":"ON"}'
Info 2023-06-12 19:57:04MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139/l2', payload '{"brightness":"ON","countdown":null,"state":"ON"}'
Debug 2023-06-12 19:57:05Received MQTT message on 'zigbee2mqtt/0xa4c138ded54d0139/set' with data '{"brightness_l1":179,"state_l1":"ON"}'
Debug 2023-06-12 19:57:05Publishing 'set' 'brightness' to '0xa4c138ded54d0139'
Info 2023-06-12 19:57:05MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139', payload '{"brightness_l1":"ON","brightness_l2":"ON","countdown_l1":null,"countdown_l2":null,"linkquality":87,"state_l1":"ON","state_l2":"ON"}'
Info 2023-06-12 19:57:05MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139/l1', payload '{"brightness":"ON","countdown":null,"state":"ON"}'
Info 2023-06-12 19:57:05MQTT publish: topic 'zigbee2mqtt/0xa4c138ded54d0139/l2', payload '{"brightness":"ON","countdown":null,"state":"ON"}'
Debug 2023-06-12 19:57:06Received MQTT message on 'zigbee2mqtt/0xa4c138ded54d0139/set' with data '{"brightness_l1":179,"state_l1":"ON","state_l2":"OFF"}'
Debug 2023-06-12 19:57:06Publishing 'set' 'brightness' to '0xa4c138ded54d0139'
Error 2023-06-12 19:57:06Publish 'set' 'brightness' to '0xa4c138ded54d0139' failed: 'Error: No datapoint defined for 'state_l2''
Debug 2023-06-12 19:57:06Error: No datapoint defined for 'state_l2' at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.js:1436:27) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:247:52) at EventEmitter.emit (node:events:525:35) at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22) at MQTT.onMessage (/app/lib/mqtt.ts:140:27) at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:126:27) at WebSocket.emit (node:events:513:28) at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1184:20) at Receiver.emit (node:events:513:28) at Receiver.dataMessage (/app/node_modules/ws/lib/receiver.js:541:14)
Debug 2023-06-12 19:57:07Received MQTT message on 'zigbee2mqtt/0xa4c138ded54d0139/set' with data '{"brightness_l1":179,"state_l1":"ON","state_l2":"ON"}'
Debug 2023-06-12 19:57:07Publishing 'set' 'brightness' to '0xa4c138ded54d0139'
Error 2023-06-12 19:57:07Publish 'set' 'brightness' to '0xa4c138ded54d0139' failed: 'Error: No datapoint defined for 'state_l2''
Debug 2023-06-12 19:57:07Error: No datapoint defined for 'state_l2' at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.js:1436:27) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:247:52) at EventEmitter.emit (node:events:525:35) at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22) at MQTT.onMessage (/app/lib/mqtt.ts:140:27) at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:126:27) at WebSocket.emit (node:events:513:28) at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1184:20) at Receiver.emit (node:events:513:28) at Receiver.dataMessage (/app/node_modules/ws/lib/receiver.js:541:14)
Debug 2023-06-12 19:57:08Received MQTT message on 'zigbee2mqtt/0xa4c138ded54d0139/set' with data '{"brightness_l1":179,"brightness_l2":202,"state_l1":"ON","state_l2":"ON"}'
Debug 2023-06-12 19:57:08Publishing 'set' 'brightness' to '0xa4c138ded54d0139'
Error 2023-06-12 19:57:08Publish 'set' 'brightness' to '0xa4c138ded54d0139' failed: 'Error: No datapoint defined for 'state_l2''
Debug 2023-06-12 19:57:08Error: No datapoint defined for 'state_l2' at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.js:1436:27) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:247:36)
Debug 2023-06-12 19:57:09Received MQTT message on 'zigbee2mqtt/0xa4c138ded54d0139/set' with data '{"brightness_l1":179,"brightness_l2":74,"state_l1":"ON","state_l2":"ON"}'
Debug 2023-06-12 19:57:09Publishing 'set' 'brightness' to '0xa4c138ded54d0139'
Error 2023-06-12 19:57:09Publish 'set' 'brightness' to '0xa4c138ded54d0139' failed: 'Error: No datapoint defined for 'state_l2''
Debug 2023-06-12 19:57:09Error: No datapoint defined for 'state_l2' at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.js:1436:27) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:247:3 [... cut ...]
Debug 2023-06-12 19:57:29Received Zigbee message from '0xa4c138ded54d0139', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Debug 2023-06-12 19:57:29No converter available for 'TS0052_dimmer_2' with cluster 'genOnOff' and type 'attributeReport' and data '{"onOff":1}'
Debug 2023-06-12 19:57:29Received Zigbee message from '0xa4c138ded54d0139', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 2 with groupID 0
Debug 2023-06-12 19:57:29No converter available for 'TS0052_dimmer_2' with cluster 'genOnOff' and type 'attributeReport' and data '{"onOff":1}'`

Koenkk, 2023-06-13:
> Now controls in place, but they don't change things.

To confirm, even setting just the state doesn't work? e.g.

```
Received MQTT message on 'zigbee2mqtt/0xa4c138ded54d0139/set' with data '{"state_l1":"ON"}'
```

closed 2023-07-22 by github-actions[bot]