Koenkk/zigbee2mqtt #17926: [New device support]: LoraTap ZigBee 3.0 Tuya Smart Life Roller Shutter Curtain Light US Switch

dbenyakar, 2023-06-05:
### Link

https://www.aliexpress.com/item/1005005590013359.html?spm=a2g0o.order_list.order_list_main.16.60461802ZJ5NBp

### Database entry

{"id":8,"type":"Router","ieeeAddr":"0xa4c138ac04ee9cc6","nwkAddr":12732,"manufId":4417,"manufName":"_TZE204_pxbjch8m","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"��\u0010,f��\u0010,\u0012��\u0010,e","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":74}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1685986280452,"defaultSendRequestWhen":"immediate"}

### Comments

Im new to this and got lost on the new device and converter part.
I searched for Loratap device under the devices list and found only one. But this one is with light.

dbenyakar, 2023-06-07:
Tried to create an external converter for the new device for the first time using the instructions and some examples here.
Not sure at all if this is how it should be but I ended up getting an error:

Starting Zigbee2MQTT...
/app/node_modules/zigbee-herdsman-converters/index.js:79
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:79:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)

The converter is:

`const legacy = require('zigbee-herdsman-converters/lib/legacy');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = {...require('zigbee-herdsman-converters/converters/toZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').toZigbee};
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
        fingerprint: [{modelID: 'TS0601', manufacturerName:'_TZE204_pxbjch8m'}],
        zigbeeModel: ['TS0601'],
        model: 'TS0601_curtain',
        vendor: 'Loratap',
        description: 'Curtain/blind switch with 1 Gang switch',
		fromZigbee: [fz.datapoints],
		toZigbee: [tz.datapoints],
        exposes: [tuya.exposes.switch().withEndpoint('l1'),
				  e.cover_position().setAccess('position', ea.STATE_SET), exposes.enum('backlight', ea.STATE_SET, ['OFF', 'ON']),
				  exposes.enum('calibration', ea.STATE_SET, ['OFF', 'ON']), exposes.enum('motor_reversal', ea.STATE_SET, ['OFF', 'ON'])],
        endpoint: (device) => {
            return {'l1': 1};
        },
        meta: {
            multiEndpoint: true,
            tuyaDatapoints: [
                [1, 'state', tuya.valueConverterBasic.lookup({'OPEN': tuya.enum(0), 'STOP': tuya.enum(1), 'CLOSE': tuya.enum(2)})],
                [2, 'position', tuya.valueConverter.coverPosition],
                [3, 'calibration', tuya.valueConverterBasic.lookup({'START': tuya.enum(0), 'END': tuya.enum(1)})],
                [7, 'backlight_mode', tuya.valueConverter.onOff],
                [8, 'motor_steering', tuya.valueConverterBasic.lookup({'FORWARD': tuya.enum(0), 'BACKWARD': tuya.enum(1)})],
                [101, 'state_l1', tuya.valueConverter.onOff],
                [103, 'child_lock', tuya.valueConverter.onOff],
            ],
        },
    };

module.exports = definition;`

When started the Z2M add on I got the following error:

dbenyakar, 2023-07-08:
Anything?

closed 2023-08-17 by github-actions[bot]

erezagami, 2023-11-09:
I was able to get the device to work, 
it is not perfect, I do not think it support curtain position.

This is my current external convertor , hope it help

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_pxbjch8m',
        },
    ],
    model: 'SC400ZB-US-JD-WHI',
    vendor: 'LoraTap',
    description: 'LoraTap Curtain Light Switch',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    options: [],
    exposes: [
        // Here you should put all functionality that your device exposes
        tuya.exposes.switch().withEndpoint('l1'),
        e.cover_position().withEndpoint('l2'),
        exposes.enum('motor_direction', ea.STATE_SET, ['FORWARD', 'BACKWARD']).withDescription('Set the motor direction').withEndpoint('l3'),
    ],
    endpoint: (device) => {
        return {'l1': 1, 'l2': 1, 'l3': 1};
    },
    meta: {
        multiEndpoint: true,
        tuyaDatapoints: [
            [1, 'state_l2', tuya.valueConverterBasic.lookup({'OPEN': tuya.enum(0), 'STOP': tuya.enum(1), 'CLOSE': tuya.enum(2)})],
            [7, 'state_l1' , tuya.valueConverter.onOff],
            [8, 'motor_direction_l3' , tuya.valueConverterBasic.lookup({'FORWARD': tuya.enum(0), 'BACKWARD': tuya.enum(1)})],
        ],
    },
};

module.exports = definition;`

dekela, 2024-05-07:
> I was able to get the device to work, it is not perfect, I do not think it support curtain position.
> 
> This is my current external convertor , hope it help
> 
> ```
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const extend = require('zigbee-herdsman-converters/lib/extend');
> const e = exposes.presets;
> const ea = exposes.access;
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> 
> const definition = {
>     // Since a lot of TuYa devices use the same modelID, but use different datapoints
>     // it's necessary to provide a fingerprint instead of a zigbeeModel
>     fingerprint: [
>         {
>             // The model ID from: Device with modelID 'TS0601' is not supported
>             // You may need to add \u0000 at the end of the name in some cases
>             modelID: 'TS0601',
>             // The manufacturer name from: Device with modelID 'TS0601' is not supported.
>             manufacturerName: '_TZE204_pxbjch8m',
>         },
>     ],
>     model: 'SC400ZB-US-JD-WHI',
>     vendor: 'LoraTap',
>     icon: 'https://www.zigbee2mqtt.io/images/devices/TS0601_cover_with_1_switch.jpg',
>     description: 'LoraTap Curtain Light Switch',
>     fromZigbee: [tuya.fz.datapoints],
>     toZigbee: [tuya.tz.datapoints],
>     onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
>     configure: tuya.configureMagicPacket,
>     options: [exposes.options.invert_cover()],
>     exposes: [
>         // Here you should put all functionality that your device exposes
>         tuya.exposes.switch().withEndpoint('l1'),
>         e.cover_position(),
>         // exposes.enum('calibration', ea.STATE_SET, ['START', 'END']).withDescription('Calibration'),
>         exposes.enum('motor_steering', ea.STATE_SET, ['FORWARD', 'BACKWARD']).withDescription('Motor Steering'),
>         // exposes.binary('child_lock', ea.STATE_SET, 'ON', 'OFF').withDescription('Child Lock'),
>     ],
>     endpoint: (device) => {
>         return {'l1': 1};
>     },
>     meta: {
>         multiEndpoint: true,
>         tuyaDatapoints: [
>             [1, 'state', tuya.valueConverterBasic.lookup({'OPEN': tuya.enum(0), 'STOP': tuya.enum(1), 'CLOSE': tuya.enum(2)})],
>             // [2, 'position', tuya.valueConverter.coverPosition],
>             // [3, 'calibration', tuya.valueConverterBasic.lookup({'START': tuya.enum(0), 'END': tuya.enum(1)})],
>             [7, 'state_l1' , tuya.valueConverter.onOff],
>             [8, 'motor_steering', tuya.valueConverterBasic.lookup({'FORWARD': tuya.enum(0), 'BACKWARD': tuya.enum(1)})],
>             // [103, 'child_lock', tuya.valueConverter.onOff],
>         ],
>     },
> };
> 
> module.exports = definition;
> ```

I tried to add this convertor and zigbee2mqtt doesnt seem to read from the file. What I am missing? I placed it in the zigbee2mq22 folder and enabled it in the options.

erezagami, 2024-06-23:
this is my current version
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_pxbjch8m',
        },
    ],
    model: 'SC400ZB-US-JD-WHI',
    vendor: 'LoraTap',
    description: 'LoraTap Curtain Light Switch',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    //onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    //configure: tuya.configureMagicPacket,
    exposes: [
        tuya.exposes.switch().withEndpoint('l1'),
        exposes.enum('state', ea.STATE_SET, ['OPEN', 'STOP', 'CLOSE']).withDescription('Cover State').withEndpoint('m1'),
        exposes.enum('motor_steering', ea.STATE_SET, ['FORWARD', 'BACKWARD']).withDescription('Motor Steering').withEndpoint('m1')
        // Here you should put all functionality that your device exposes
    ],
    endpoint: (device) => {
        return {'m1': 1, 'l1': 1};
    },
    meta: {
        multiEndpoint: true,
        tuyaDatapoints: [
            [1, 'state_m1', tuya.valueConverterBasic.lookup({'OPEN': tuya.enum(0), 'STOP': tuya.enum(1), 'CLOSE': tuya.enum(2)})],
            [7, 'state_l1' , tuya.valueConverter.onOff],
            [8, 'motor_steering_m1', tuya.valueConverterBasic.lookup({'FORWARD': tuya.enum(0), 'BACKWARD': tuya.enum(1)})],
        ],
    },
    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;
```

you need to place it in zigbee2mqtt direcotry, add it in the config as external converter and restart zigbee2mqtt

dekela, 2024-06-25:
@erezagami  Thank you. I tried it and I was able to get it into zigbee2mqtt but I cant change the state or any of the switches.

erezagami, 2024-06-30:
hmm, try to delete the device and add it again, maybe it still have old attributes

jony6484, 2024-11-06:
Any news anyone? is it possible to get the position (%)?

ArikShemesh, 2025-02-14:
> Any news anyone? is it possible to get the position (%)?

Did you able to find a solution for getting the position and calibration maybe?