Koenkk/zigbee2mqtt #17928: [New device support]: New TS0003 switch module version

mersadk, 2023-06-05:
### Link

-

### Database entry

-

### Comments

This is new variant of TS003 switch with model identifier "_TZ3000_nnwehhst".
I created external converter by copying TS0003_switch_module_1 definition from https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts
Everything works without issues.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition =  {
	fingerprint: tuya.fingerprint('TS0003', ['_TZ3000_nnwehhst']),
	model: 'TS0003_switch_module_4',
	vendor: 'TuYa',
	description: '3 gang switch module',
	whiteLabel: [{vendor: 'OXT', model: 'SWTZ23'}],
	extend: tuya.extend.switch({switchType: true, backlightModeOffOn: true, endpoints: ['l1', 'l2', 'l3']}),
	endpoint: (device) => {
		return {'l1': 1, 'l2': 2, 'l3': 3};
	},
	meta: {multiEndpoint: true},
	configure: async (device, coordinatorEndpoint, logger) => {
		await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
		await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
		await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
		await reporting.bind(device.getEndpoint(3), coordinatorEndpoint, ['genOnOff']);
	},
};

module.exports = definition;
```

Koenkk, 2023-06-06:
Integrated the changes, thanks!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-06-06 by Koenkk

mersadk, 2023-06-06:
Thanks!

When I updated Zigbee2MQTT in Home Assistant, will it be enough for me to delete my external converter and during next boot my existing device should be recognized as TS0003_switch_module_1? Or I need to remove device before update, and add it again later?

Koenkk, 2023-06-07:
Removing ext converter after updating is enough.