Koenkk/zigbee2mqtt #17931: ptvo.switch stops working after Z2MQTT upgrade (from 1.30.4-1 to 1.31.0-1)

OskarL83, 2023-06-06:
### What happened?

Device (ptvo.switch) do not show anything more than linkquality after upgrade of Z2MQTT.
After restore from backup, to previous version of Z2MQTT, it's working again - and everything is exposed. 

### What did you expect to happen?

That the device kept working after Z2MQTT upgrade. 

### How to reproduce it (minimal and precise)

Install 1.31.0-1

### Zigbee2MQTT version

1.31.0-1

### Adapter firmware version

20211217

### Adapter

SONOFF ZigBee 3.0 USB DONGLE PLUS (TI CC2652P+CP2002N)

Dennisd80, 2023-06-10:
Same issue, under status it only displays link quality. Contacts are not working and only action or off (general) are displayed.

C09038, 2023-06-15:
I have the same problem.

jvan1268, 2023-06-15:
Same issue here. Also with Sonoff stick which is on version 20220219.

Update to Zigbee2MQTT 1.31.2 did not solve the problem.
Reconfiguration of the device did not fix the problem either.

Logs display no problems. Latest log info:
2023-06-15 22:14:09MQTT publish: topic 'zigbee2mqtt/fuse box relay', payload '{"last_seen":"2023-06-15T22:14:09+02:00","linkquality":47,"state":"ON","state_action":"OFF","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF","state_l5":"OFF","state_l6":"OFF","state_l7":"OFF","state_l8":"OFF"}'

My board has push buttons. Manually turning on either of the 4 relays seem to trigger "state" from "OFF" to "ON". It doesn't matter which of the 4 relays and how many are turned on. "state_I1" through "state_I8" never seem to change.

Koenkk, 2023-06-17:
@ptvoinfo I think this broke due to https://github.com/Koenkk/zigbee-herdsman-converters/pull/5792

evilworm, 2023-07-18:
the fix does not work for me, commented in the other issue: https://github.com/Koenkk/zigbee-herdsman-converters/pull/5894#issuecomment-1640917805

PetrP2B, 2023-08-19:
Not stale.

PetrP2B, 2023-09-05:
Everything works correctly in the latest version of Edge and Stable. Thanks

OskarL83, 2023-09-05:
> Everything works correctly in the latest version of Edge and Stable. Thanks

OK. Without wipe and re-pair of devices?

PetrP2B, 2023-09-05:
I installed the latest versions of Z2M and everything was running. I didn't have to remove and pair the device again.

OskarL83, 2023-09-05:
> I installed the latest versions of Z2M and everything was running. I didn't have to remove and pair the device again.

Didn't work here. :-(

PetrP2B, 2023-09-05:
It worked with the Edge version, available on 29/08/2023, but could not get to the interface, see the image above, in this version (1.33.0-dev commit: 3285850) it works correctly for me.
[image]
[image]

triosniolin, 2023-09-12:
I just had [one of these](https://www.tindie.com/products/a_lab_technology/zigbee-4-channel-relay) delivered and was intending to use it to monitor a few switch inputs; however, Z2M registers it as a ptvo.switch and only exposes the 4 digital outputs. A change of state on the digital inputs comes through as an `action` only, and that action is immediately overwritten by an empty string or the next thing to occur so I can't view the current state of any input.

Is there something I'm missing to get these inputs available in Z2M and therefore Home Assistant?

closed 2023-10-21 by github-actions[bot]

OskarL83, 2023-10-21:
Not stale

ptvoinfo, 2023-10-21:
@OskarL83 Please describe again what is not working in your case. I'm sorry, but I cannot read error messages in your language from the screenshot.

OskarL83, 2023-10-21:
After upgrade, the multi channel relay is not working.

Before upgrade:
[image]
[image]
[image]
[image]

After upgrade:
[image]
[image]
[image]

OskarL83, 2023-10-21:
Above was after this upgrade was done.
[image]