Koenkk/zigbee2mqtt #17937: Samotech SM323- new model ID

jdiviney, 2023-06-06:
### What happened?

Bought new Samotech SM323- listed as supported, however there is a new modelID: "SM323" rather than the prior 'HK_DIM_A'
When added as a custom converter with the correct modelID it works perfectly.

### What did you expect to happen?

To be recognised by Z2M natively

### Zigbee2MQTT version

1.31.0

### Adapter firmware version

0x26580700

### Adapter

ConBee2

Koenkk, 2023-06-07:
Can you provide the data/database.db entry of this device?

jdiviney, 2023-06-07:
{"id":45,"type":"Router","ieeeAddr":"0xf4b3b1fffef939e3","nwkAddr":7008,"manufId":4107,"manufName":"Samotech","modelId":"SM323","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"powerSource":18,"appVersion":0,"stackVersion":0,"hwVersion":0,"swBuildId":"1.00","zclVersion":8}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":147}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff07d3fe","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00212effff07d3fe","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":8,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"20230318","swBuildId":"1.00","zclVersion":8,"interviewCompleted":true,"meta":{"configured":-526939472},"lastSeen":1686164235324,"defaultSendRequestWhen":"immediate"}

Koenkk, 2023-06-07:
Fixed!

Changes will be available in the dev branch tomorrow. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-06-07 by Koenkk