Koenkk/zigbee2mqtt #17940: [New device support]: LTECH TY-12-100-400-W1Z (Tuya LED CC driver)

apollolucas, 2023-06-07:
### Link

http://www.ltechonline.com/html/en/products/Intelligence/zigbee/TY-12-100-400-W1Z.html

### Database entry

{"id":9, "type":"Router", "ieeeAddr":"0x50325ffffe404b6e", "nwkAddr":54946,"manufId":4098, "manufName":"_TZE200_86nbew0j", "powerSource":"Mains (single phase)", "modelId":"TS0601", "epList":[1], "endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0004","65506":31,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_86nbew0j","stackVersion":0,"dateCode":"","appVersion":72,"zclVersion":3,"powerSource":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0029de14e3","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0029de14e3","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-24894193},"lastSeen":1685993790462,"defaultSendRequestWhen":"immediate"}

### Comments

This is a high quality constant current LED driver from the manufacturer LTECH. It's a Tuya compatible device that identifies as model ID TS0601. I tried repurposing 2 different existing TS0601 dimmer converters: 'TS0601_dimmer' and 'TS0601_dimmer_1'. 

The 'TS0601_dimmer' converter worked OK although there was a bug where if the brightness was set to 254 it will immediately jump to -4 (seemed similar to https://github.com/Koenkk/zigbee2mqtt/issues/13800#issuecomment-1263592783). Also, the mix and max brightness settings didn't seem to work as intended.

The 'TS0601_dimmer_1' converter seemed to work a little better but the brightness was being controlled by the min brightness setting instead of the brightness setting. The brightness setting, max brightness setting, countdown, light_type, and power_on_behaviour didn't seem to have any effect.

I changed the 'TS0601_dimmer_1' converter (see the code attached here) to swap dp of brightness with min_brightness and to remove some of the datapoints that didn't seem to do anything. This code works but there are two strange behaviors:

1. State (on/off) gets replaced by the brightness level. This happens only occasionally when the brightness is being adjusted. I can't seem to figure out what triggers this. This happens both when controlling the device from the Zigbee2MQTT GUI and from Apple Homekit (through HomeBridge). Turning the device off and on again reverts to the correct state.

[image]

[image]

2. The on/off state gets send multiple times.

[image]

Apart from these 2 issues I'm also wondering if the device exposes other Tuya dps (e.g. power on setting, transition, etc.) but I don't have a Tuya hub to try and find out. I've sent an email to the manufacturer to ask for DPIDs and UART and I'll update the case if I hear back from them (unlikely).

Any help to figure this out will be much appreciated!

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
	fingerprint: [
		{
			// The model ID from: Device with modelID 'TS0601' is not supported
			// You may need to add \u0000 at the end of the name in some cases
			modelID: 'TS0601',
			// The manufacturer name from: Device with modelID 'TS0601' is not supported.
			manufacturerName: '_TZE200_86nbew0j',
		},
	],    
	model: 'TY-12-100-400-W1Z', // Vendor model number, look on the device for a model number
	vendor: 'Tuya', // Vendor of the device (only used for documentation and startup logging)
	description: 'LTECH LED driver',
	fromZigbee: [tuya.fz.datapoints],
	toZigbee: [tuya.tz.datapoints],
	configure: tuya.configureMagicPacket,
	exposes: [tuya.exposes.lightBrightness(), e.power_on_behavior()],
	meta: {
		tuyaDatapoints: [
			[1, 'state', tuya.valueConverter.onOff, {skip: tuya.skip.stateOnAndBrightnessPresent}],
			[3, 'brightness', tuya.valueConverter.scale0_254to0_1000],
			[14, 'power_on_behavior', tuya.valueConverter.powerOnBehavior],
		],
	},
}
	
module.exports = definition;
```

lychee512, 2023-06-08:
I am working on a similar device from LTECH, #17953, you might want to try out the converter that I wrote. It faces some other issues compared to yours though.

larndoc, 2023-07-04:
Hi, I have the same device and used your converter from above and can confirm the issue. Sometimes after setting the brightness the "state" reports brightness instead of "on/off". Also I noticed once that I could not turn the light off when this happened (other than removing the power). 
Would be great to solve the issue to make this device officially supported.

closed 2023-08-12 by github-actions[bot]