Koenkk/zigbee2mqtt #17942: TuYa TS004F (_TZ3000_xabckq1v): action field ist always "single" and not "1_single", "2_single"...

ikarisan, 2023-06-07:
### What happened?

After upgrading to "1.31.0-dev commit: [73c19bf](https://github.com/Koenkk/zigbee2mqtt/commit/73c19bf)" my two TuYa TS004F switches (_TZ3000_xabckq1v) stopped working. 

Before the update the "action" was always "1_single", "1_double", "2_single", "2_double"...
After the updates every button reports "action": "single" or "action": "double" only. :(

```

Received Zigbee message from 'TuYa_Vierfach_Schalter2', type 'raw', cluster 'genOnOff', data '{"data":[1,57,253,0],"type":"Buffer"}' from endpoint 1 with groupID 0

MQTT publish: topic 'zigbee2mqtt/TuYa_Vierfach_Schalter2', payload '{"action":"single","battery":89,"device":{"applicationVersion":65,"dateCode":"","friendlyName":"TuYa_Vierfach_Schalter2","hardwareVersion":1,"ieeeAddr":"0x5c0272fffe050e42","manufacturerID":4098,"manufacturerName":"_TZ3000_xabckq1v","model":"TS004F","networkAddress":37360,"powerSource":"Battery","stackVersion":0,"type":"EndDevice","zclVersion":3},"last_seen":"2023-06-07T12:48:38+00:00","linkquality":99,"voltage":2900}'

Received Zigbee message from 'TuYa_Vierfach_Schalter2', type 'raw', cluster 'genOnOff', data '{"data":[1,57,253,0],"type":"Buffer"}' from endpoint 1 with groupID 0

MQTT publish: topic 'zigbee2mqtt/TuYa_Vierfach_Schalter2', payload '{"action":"double","battery":89,"device":{"applicationVersion":65,"dateCode":"","friendlyName":"TuYa_Vierfach_Schalter2","hardwareVersion":1,"ieeeAddr":"0x5c0272fffe050e42","manufacturerID":4098,"manufacturerName":"_TZ3000_xabckq1v","model":"TS004F","networkAddress":37360,"powerSource":"Battery","stackVersion":0,"type":"EndDevice","zclVersion":3},"last_seen":"2023-06-07T12:48:42+00:00","linkquality":99,"voltage":2900}'
```

### What did you expect to happen?

The four buttons should report individual "action" properties.

### How to reproduce it (minimal and precise)

Install commit [73c19bf](https://github.com/Koenkk/zigbee2mqtt/commit/73c19bf), pair a TS004F and press one of the four buttons. 

### Zigbee2MQTT version

1.31.0-dev

### Adapter firmware version

20201010

### Adapter

CC2538+CC2592, zStack30x

Koenkk, 2023-06-07:
Should be fixed in z2m 1.31.1

narmiak, 2023-06-07:
Just updated to z2m 1.31.1 and now I have same issue with TS0044. It was working fine under 1.31.0

djgrazzy, 2023-06-07:
Same issue with TS0044

> can you check if the latest-dev is fine? (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)
Happy  to test edge version in HA but unsure how to using the addon?

"Home Assistant addon
Use the edge version."

ikarisan, 2023-06-07:
> Should be fixed in z2m 1.31.1

Yes, works fine with 1.31.1-dev commit: [86898e5](https://github.com/Koenkk/zigbee2mqtt/commit/86898e5)

narmiak, 2023-06-07:
> > Should be fixed in z2m 1.31.1
> 
> Yes, works fine with 1.31.1-dev commit: [86898e5](https://github.com/Koenkk/zigbee2mqtt/commit/86898e5) (docker)

I can confirm.

transon123A, 2023-06-07:
Tôi cũng bị lỗi này sau khi nâng cấp lên 1.31.1.1. Xin hướng dẫn cách giải quyết vấn đề

pingu96, 2023-06-07:
Just came to say that I'm having the same issue with the TuYa TS0042 Since the update to 1.31.1-1 it now only shows "single" when either button is pressed rather than "1_single" or "2_single" depending on which button is pressed. I have rolled back to 1.31.0-1 and it's working fine again - not sure how to upgrade to the dev version with the home assistant OS addon to check if it's fixed.

fjjking, 2023-06-07:
same problem on TS0042. Rolling back to 1.31.0-1 fixes the issue

djgrazzy, 2023-06-07:
Can confirm working in Edge for me.

To run Edge on HA you will need to backup your xml configuration from the current installation (click configuration tab, then edit in xml (3 dots top right) and save somewhere safe (like a saved txt file on your pc)
Uninstall the current add-on
Find "Zigbee2MQTT Edge" in the add-on store and install.
Paste your saved config from above into configuration and start the add-on.

Koenkk, 2023-06-07:
Should be fixed in z2m 1.31.2

Iroamstaff, 2023-06-07:
Same problem with TS0044
_TZ3000_dziaict4 
Zigbee 4 button remote - 12 scene

closed 2023-07-15 by github-actions[bot]