Koenkk/zigbee2mqtt #17944: Ikea Symfonisk E1744: can not upgrade from 2.3.80 to 24.4.5

bartplessers, 2023-06-07:
### What happened?

Running
Home Assistant 2023.5.4
Supervisor 2023.06.1
Operating System 10.2
Frontend 20230503.3 - latest

with

Zigbee2MQTT
Current version: 1.31.0-1 

I'm trying to update my Ikea Symfonisk E1744 OTA
Update starts, but fails almost immediatly

logs
```
Info 2023-06-07 16:44:17MQTT publish: topic 'zigbee2mqtt/my/R15', payload '{"action":"","battery":60,"brightness":150,"linkquality":105,"update":{"installed_version":587728433,"latest_version":604241925,"state":"available"},"update_available":true}'
Info 2023-06-07 16:44:17MQTT publish: topic 'zigbee2mqtt/my/R15/action', payload 'toggle'
Info 2023-06-07 16:44:29Update of 'my/R15' at 0.00%
Info 2023-06-07 16:44:29MQTT publish: topic 'zigbee2mqtt/my/R15', payload '{"action":null,"battery":60,"brightness":150,"linkquality":105,"update":{"installed_version":587728433,"latest_version":604241925,"progress":0,"state":"updating"},"update_available":false}'
Info 2023-06-07 16:44:29MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'my/R15' at 0.00%","meta":{"device":"my/R15","progress":0,"status":"update_progress"},"type":"ota_update"}'
Info 2023-06-07 16:44:31MQTT publish: topic 'zigbee2mqtt/my/R15', payload '{"action":null,"battery":60,"brightness":150,"linkquality":105,"update":{"installed_version":587728433,"latest_version":604241925,"state":"available"},"update_available":true}'
Info 2023-06-07 16:44:31MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'my/R15' failed (Update failed with reason: 'aborted by device')","meta":{"device":"my/R15","status":"update_failed"},"type":"ota_update"}'
Info 2023-06-07 16:44:31MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"my/R15"},"error":"Update of 'my/R15' failed (Update failed with reason: 'aborted by device')","status":"error","transaction":"od7gh-1"}'
Error 2023-06-07 16:44:31Update of 'my/R15' failed (Update failed with reason: 'aborted by device')
```

### What did you expect to happen?

update succeeds

### Zigbee2MQTT version

1.31.0-1 

### Adapter firmware version

CC2652R_coordinator_20220219

### Adapter

CC2652 (Electrolama zig-a-zig-ah! (zzh!))

### Debug log

relevant debug log:

```
debug 2023-06-07 16:54:43: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/update' with data '{"id":"my/R15","transaction":"n6o3b-3"}'
info  2023-06-07 16:54:43: Updating 'my/R15' to latest firmware
info  2023-06-07 16:54:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Updating 'my/R15' to latest firmware","meta":{"device":"my/R15","status":"update_in_progress"},"type":"ota_update"}'
debug 2023-06-07 16:54:49: Received Zigbee message from 'my/n02/p01/power/P02/multistekker_LIDL', type 'attributeReport', cluster 'genBasic', data '{"65506":30,"appVersion":69}' from endpoint 1 with groupID 0
debug 2023-06-07 16:54:51: Received Zigbee message from 'my/R15', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
info  2023-06-07 16:54:52: MQTT publish: topic 'zigbee2mqtt/my/R15', payload '{"action":"toggle","battery":60,"brightness":255,"linkquality":102,"update":{"installed_version":587728433,"latest_version":604241925,"state":"available"},"update_available":true}'
info  2023-06-07 16:54:52: MQTT publish: topic 'zigbee2mqtt/my/R15', payload '{"action":"","battery":60,"brightness":255,"linkquality":102,"update":{"installed_version":587728433,"latest_version":604241925,"state":"available"},"update_available":true}'
info  2023-06-07 16:54:52: MQTT publish: topic 'zigbee2mqtt/my/R15/action', payload 'toggle'
debug 2023-06-07 16:54:52: Received Zigbee message from 'my/R15', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20210727","swBuildId":"2.3.080"}' from endpoint 1 with groupID 0
debug 2023-06-07 16:54:52: Updating to latest '0x680ae2fffe1022ae' (SYMFONISK Sound Controller)
debug 2023-06-07 16:54:52: Using endpoint '1'
debug 2023-06-07 16:54:55: Received Zigbee message from 'my/R15', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":587728433,"imageType":4554,"manufacturerCode":4476}' from endpoint 1 with groupID 0
debug 2023-06-07 16:54:55: Got OTA request '{"fieldControl":1,"manufacturerCode":4476,"imageType":4554,"fileVersion":587728433}'
debug 2023-06-07 16:54:55: getNewImage for '0x680ae2fffe1022ae', meta {"fileVersion":604241925,"url":"http://fw.ota.homesmart.ikea.net/global/GW1.0/01.20.065/bin/tradfri_dimmer-24.4.5-prod.ota.ota.signed"}
debug 2023-06-07 16:54:55: getNewImage for '0x680ae2fffe1022ae', image header {"otaUpgradeFileIdentifier":{"type":"Buffer","data":[30,241,238,11]},"otaHeaderVersion":256,"otaHeaderLength":56,"otaHeaderFieldControl":0,"manufacturerCode":4476,"imageType":4554,"fileVersion":604241925,"zigbeeStackVersion":2,"otaHeaderString":"GBL GBL_tradfri_dimmer\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","totalImageSize":214692}
debug 2023-06-07 16:54:55: Got new image for '0x680ae2fffe1022ae'
debug 2023-06-07 16:54:55: Starting upgrade
debug 2023-06-07 16:54:56: Received Zigbee message from 'my/R15', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":587728433,"imageType":4554,"manufacturerCode":4476}' from endpoint 1 with groupID 0
debug 2023-06-07 16:55:01: Received Zigbee message from 'my/R15', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
info  2023-06-07 16:55:01: MQTT publish: topic 'zigbee2mqtt/my/R15', payload '{"action":"toggle","battery":60,"brightness":255,"linkquality":102,"update":{"installed_version":587728433,"latest_version":604241925,"state":"available"},"update_available":true}'
[... 10 lines cut ...]
info  2023-06-07 16:55:34: Update of 'my/R15' at 0.54%, ≈ 119 minutes remaining
info  2023-06-07 16:55:34: MQTT publish: topic 'zigbee2mqtt/my/R15', payload '{"action":null,"battery":60,"brightness":255,"linkquality":105,"update":{"installed_version":587728433,"latest_version":604241925,"progress":0.54,"remaining":7160,"state":"updating"},"update_available":false}'
info  2023-06-07 16:55:34: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'my/R15' at 0.54%, ≈ 119 minutes remaining","meta":{"device":"my/R15","progress":0.54,"status":"update_progress"},"type":"ota_update"}'
debug 2023-06-07 16:55:52: Received Zigbee message from '0xe0798dfffec2f1a3', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":19778}' from endpoint 1 with groupID 0
info  2023-06-07 16:55:52: MQTT publish: topic 'zigbee2mqtt/0xe0798dfffec2f1a3', payload '{"battery":null,"humidity":68.34,"illuminance":19778,"illuminance_lux":95,"linkquality":149,"temperature":20.03}'
debug 2023-06-07 16:56:05: OTA update at 1.02%, remaining 6721.894658181817 seconds
info  2023-06-07 16:56:05: Update of 'my/R15' at 1.02%, ≈ 112 minutes remaining
info  2023-06-07 16:56:05: MQTT publish: topic 'zigbee2mqtt/my/R15', payload '{"action":null,"battery":60,"brightness":255,"linkquality":102,"update":{"installed_version":587728433,"latest_version":604241925,"progress":1.02,"remaining":6722,"state":"updating"},"update_available":false}'
info  2023-06-07 16:56:05: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'my/R15' at 1.02%, ≈ 112 minutes remaining","meta":{"device":"my/R15","progress":1.02,"status":"update_progress"},"type":"ota_update"}'
debug 2023-06-07 16:56:36: OTA update at 1.54%, remaining 6446.110778181818 seconds
info  2023-06-07 16:56:36: Update of 'my/R15' at 1.54%, ≈ 107 minutes remaining
info  2023-06-07 16:56:36: MQTT publish: topic 'zigbee2mqtt/my/R15', payload '{"action":null,"battery":60,"brightness":255,"linkquality":102,"update":{"installed_version":587728433,"latest_version":604241925,"progress":1.54,"remaining":6446,"state":"updating"},"update_available":false}'
info  2023-06-07 16:56:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'my/R15' at 1.54%, ≈ 107 minutes remaining","meta":{"device":"my/R15","progress":1.54,"status":"update_progress"},"type":"ota_update"}'
debug 2023-06-07 16:56:40: Received Zigbee message from 'my/n00/p01/powerplug/P04/LIDL', type 'attributeReport', cluster 'genBasic', data '{"65506":30,"appVersion":66}' from endpoint 1 with groupID 0
[... cut ...]
info  2023-06-07 16:57:01: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"my/R15"},"error":"Update of 'my/R15' failed (Update failed with reason: 'aborted by device')","status":"error","transaction":"n6o3b-3"}'
error 2023-06-07 16:57:01: Update of 'my/R15' failed (Update failed with reason: 'aborted by device')
debug 2023-06-07 16:57:01: Error: Update failed with reason: 'aborted by device'
    at /app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.js:406:24

```

bartplessers, 2023-06-07:
PS.
Just upgraded the firmware of my coordinator to
`CC2652R_coordinator_20221226.hex`

same problem

stephdust, 2023-06-10:
Hi guys, 
problem with Z2M version  1.30.4-dev as well but got a different error message
failed (Device didn't respond to OTA request)

MattWestb, 2023-06-10:
'aborted by device' is normally the remote have to low battery and is aborting the update.
Try putting in one new battery and see if its working OK.

closed 2023-06-14 by bartplessers

infabo, 2024-07-28:
I had the same error message and problem. The update started and stopped at 0% (`Update failed with reason: 'ABORT'`). Removing the battery didn’t help. Re-pairing didn’t help either. The battery level was at 74%.

In the end, the following helped: Start the OTA update via zigbee2mqtt and hold the `brightness_move_up` button until the first notification about the OTA progress appears (`z2m: Update of 'Shortcut Button' at 1.03%, ≈ 100 minutes remaining`). After that, the update continues without further without holding the button.

I hope this helps someone.