Koenkk/zigbee2mqtt #17945: Update 1.31.1-1

termit0s, 2023-06-07:
### What happened?

After update 1.31.1-1 from 1.31.0-1, all buttons TS004F (TZ3000_xabckq1v) state is single/double/etc without button number.

### Zigbee2MQTT version

1.31.1-1

### Adapter firmware version

20221226

### Adapter

CC2652P

Koenkk, 2023-06-07:
Could you check if this issue is fixed in the dev branch?. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

termit0s, 2023-06-07:
I tried it and it works normal! All buttons!

Johnsonsons, 2023-06-07:
The Zigbee switch now only transmits the commands "single/double"
Only the button numbers are missing!

Zigbee Modell  TS0044
Wireless switch with 4 buttons

Using: zigbee2mqtt 1.31.1-1
with a request for troubleshooting 🔢

1shkoda1, 2023-06-07:
I have two device Zigbee Modell TS0044 on 4 buttons and Zigbee Modell TS0043 on 3 buttons. Situation the same. No working. It send singe/double command only

Flash3dee, 2023-06-07:
> > I have two device Zigbee Modell TS0044 on 4 buttons and Zigbee Modell TS0043 on 3 buttons. Situation the same. No working. It send singe/double command only
> 
> try the edge version(1.31.0-1. Works fine. Dont forget to restart the host system -_-)

Here the same with TS0043 and TS0044 switches, rolled back to 1.31.0-1 version.
With edge version you mean the previous release 1.31.0-1?

closed 2023-06-08 by termit0s

Fougues, 2023-10-02:
the buttons go up in action but there are systematically 2 actions which go up at the same time.
One empty and another for example 1_single.

Except that the automation sees only one, either the empty one or the other (1_single) when it sees the empty one then nothing happens.

so you can press the button and see nothing happen. it's quite random.
I couldn't find a way to resolve this.

Flash3dee, 2023-10-02:
> the buttons go up in action but there are systematically 2 actions which go up at the same time. One empty and another for example 1_single.
>
> Except that the automation sees only one, either the [...]

Hi, maybe this helps: https://github.com/home-assistant/core/issues/98739