Koenkk/zigbee2mqtt #17946: Same model Aqara switches expose differently [QBKG41LM, Aqara E1 2-Gang with neutral]

alryaz, 2023-06-07:
### What happened?

I have 7 switches model mentioned (QBKG41LM), some of which (specifically 4) do not expose their button `action` and `device_temperature` in any mode of operation.

I have so far tried all expected problem-solving attempts:
- updated coordinator firmware (jumped around between 6.7.X.X and 7.2.X.X)
- re-paired devices
- reconfigured using refresh button
- put them into all kinds of couple/decouple modes

This seems weird, as the firmware on those is identical (datecode: _06-15-2022_)

### What did you expect to happen?

Decoupling switches will at least provide `single_left` or `single_right` values for the `action` enum.

### How to reproduce it (minimal and precise)

- Start monitoring `action` enum
- Press any of the buttons
- Action gets set to `unavailable` on Home Assistant (as example) side, then back to `null`

### Zigbee2MQTT version

1.30.1

### Adapter firmware version

6.10.3

### Adapter

ZB-GW04 (USB, EZSPv8)

### Debug log

_Weirdly enough, no messages related to switches appear, but I will update the issue once encountered_

DiukBogdan, 2023-06-21:
I've got the same proplem with my two QBKG41LM switches whitch I bought at different time.
In automatisations, trigger "...action_single_right" marked as unnown trigger

Zigbee2MQTT version
1.31.2

Adapter firmware version
20210708

Adapter
Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus

jomud9, 2023-07-05:
I am not sure if it helps. However from https://github.com/Koenkk/zigbee2mqtt/issues/17477 it suggested upgrade dongle firmware "might" help.

See if it works

dashmakov, 2023-08-06:
My Aqara E1 2 gang switch (with neutral) (QBKG41LM) connect Zigbee2MQTT Current version: 1.32.2-1 dont transmission single_left or single_right values for the action enum.

nightmare04, 2023-08-08:
Same problem Aqara E1 2 gang dont transmission any state

bartholomuej, 2023-09-10:
I have the same issue with both Aqara E1 2 gang and single gang sending "Empty" when pressed

closed 2023-10-19 by github-actions[bot]

welderpb, 2024-04-10:
I have same issue with one of my QBKG41LM. The switch not emitting actions in decoupled mode. I'v tried update my dongle(Sonoff Dongle E) to different firmware and different version z2m.
The strange is that it worked normally with this dongle early.. I'v didn't change firmware and z2m version.. Just stop working..

pawelbudynek, 2024-05-06:
The problem still exists. It's strange, the switch is pairing without any issues, you can switch it to decoupled mode, but then HA is not receiving any actions when you click the switch.
I never had this issue with previous versions of Aqara Wall Switches :(

nightmare04, 2024-05-12:
> The problem still exists. It's strange, the switch is pairing without any issues, you can switch it to decoupled mode, but then HA is not receiving any actions when you click the switch. I never had this issue with previous versions of Aqara Wall Switches :(

for me helps to change firmware of zigbee stick, try it

welderpb, 2024-05-12:
> for me helps to change firmware of zigbee stick, try it

What stick and firmware do you use?

nightmare04, 2024-05-12:
> What stick and firmware do you use?

SONOFF Zigbee 3.0 USB Dongle Plus and zStack3x0 version 20230507 works fine with 2 aqara switches