Koenkk/zigbee2mqtt #17947: Zigbee2mqtt doesn't start anymore due to changed adapter path/location

Schmetterfliege, 2023-06-07:
### What happened?

Came home to find that my zigbee devices didn't work anymore - with regards to any automations I made.
The zigbee2mqtt frontend was still accessible, but the devices just didn't trigger my automations when pushing buttons etc.
I'm running iobroker, so I tried updating my iobroker to see if that was the issue - nope. Couldn't start iobroker anymore, so I rebooted the Pi.
After the reboot I ran a script to fix the iobroker installation, which then worked again. But zigbee2mqtt was still not available.
I remember that I always had to manually start it after a reboot, but it just didn't start.
Throwing this error:

`
pi@raspberrypi:/opt/zigbee2mqtt $ sudo npm start

> zigbee2mqtt@1.28.4 start
> node index.js
`
Zigbee2MQTT:debug 2023-06-07 18:37:49: Loaded state from file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2023-06-07 18:37:49: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2023-06-07.18-37-49' filename: log.txt
Zigbee2MQTT:debug 2023-06-07 18:37:49: Removing old log directory '/opt/zigbee2mqtt/data/log/2023-06-07.18-29-58'
Zigbee2MQTT:info  2023-06-07 18:37:50: Starting Zigbee2MQTT version 1.28.4 (commit #52e545f)
Zigbee2MQTT:info  2023-06-07 18:37:50: Starting zigbee-herdsman (0.14.76)
Zigbee2MQTT:debug 2023-06-07 18:37:50: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"deconz","path":"/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2253809-if00"}}'
Zigbee2MQTT:error 2023-06-07 18:37:50: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-06-07 18:37:50: Failed to start zigbee
Zigbee2MQTT:error 2023-06-07 18:37:50: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-06-07 18:37:50: Exiting...
Zigbee2MQTT:error 2023-06-07 18:37:50: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2253809-if00'
    at SerialPort.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:200:28)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/@serialport/stream/lib/index.js:198:14)
    at /opt/zigbee2mqtt/node_modules/@serialport/stream/lib/index.js:242:12
`

Kind of obvious what's happening...
`pi@raspberrypi:/opt/zigbee2mqtt $ ls -l /dev/serial/by-id
ls: cannot access '/dev/serial/by-id': No such file or directory
pi@raspberrypi:/opt/zigbee2mqtt $ ls -l /dev/serial/
total 0
drwxr-xr-x 2 root root 60 Jun  7 18:44 by-path
`

For some reason, I only have "by-path" but no "by-id" in /dev/serial. No clue why or how it changed - or if it changed at all.
`pi@raspberrypi:/dev/serial/by-path $ ls
platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.1:1.0
`

There is a device, but I don't know if that is the adapter. There are no other usb devices connected to that pi.
When checking with "dmesg", I get the following output (sorry, it is veeeery long and I have no clue how I can make it "hidden")
`pi@raspberrypi:/dev/serial/by-path $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.15.61-v7l+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1579 SMP Fri Aug 26 11:13:03 BST 2022
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] random: crng init done
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ac00000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   Normal   empty
[    0.000000]   HighMem  [mem 0x0000000030000000-0x00000000fbffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] percpu: Embedded 16 pages/cpu s35788 r8192 d21556 u65536
[    0.000000] pcpu-alloc: s35788 r8192 d21556 u65536 alloc=16*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1011008
[... 474 lines cut ...]
[  908.807866] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  908.807887] usb 1-1.1: Product: ConBee II
[  908.807904] usb 1-1.1: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[  908.807921] usb 1-1.1: SerialNumber: DE2253809
[  908.817344] cdc_acm 1-1.1:1.0: ttyACM0: USB ACM device
`

### What did you expect to happen?

I have no idea what triggered it to fail while I was not at home. As far as I can tell, nothing seemed to have happened.

What can I do to fix this?
How can I tell it to use "by-path" instead of "by-id" - IF that device there is the adapter....

### Zigbee2MQTT version

don't know, doesn't run

### Adapter firmware version

don't know, doesn't run

### Adapter

usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2253809-if00

Schmetterfliege, 2023-06-07:
Well, basically the question is...
is the device in by-path my adapter? And if not, how can I fix this? Is my conbee stick suddenly broken?
dmesg shows at least info regarding dresden-electronics, which is the manufacturer of this stick

Schmetterfliege, 2023-06-07:
Updated zigbee2mqtt to 1.31.1, same issue.
in the configuration.yaml I replaced the adapter path to the path of the device that is shown in by-path.
After "npm start" I received a notification from Telegram of one of my devices, but the process still got me this error:

`pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.31.1 start
> node index.js

Zigbee2MQTT:debug 2023-06-07 19:14:46: Loaded state from file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2023-06-07 19:14:46: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2023-06-07.19-14-46' filename: log.txt
Zigbee2MQTT:debug 2023-06-07 19:14:46: Removing old log directory '/opt/zigbee2mqtt/data/log/2023-06-07.18-31-35'
Zigbee2MQTT:info  2023-06-07 19:14:47: Starting Zigbee2MQTT version 1.31.1 (commit #3572c60)
Zigbee2MQTT:info  2023-06-07 19:14:47: Starting zigbee-herdsman (0.14.117)
Zigbee2MQTT:debug 2023-06-07 19:14:47: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"deconz","path":"/dev/serial/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.1:1.0"}}'
Zigbee2MQTT:error 2023-06-07 19:14:47: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-06-07 19:14:47: Failed to start zigbee
Zigbee2MQTT:error 2023-06-07 19:14:47: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-06-07 19:14:47: Exiting...
Zigbee2MQTT:error 2023-06-07 19:14:47: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at SerialPort.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:200:28)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/@serialport/stream/dist/index.js:75:22)
    at /opt/zigbee2mqtt/node_modules/@serialport/stream/dist/index.js:111:18
`

To me this means that zigbee2mqtt was running for a second and then stopped again.

lux73, 2023-06-07:
try `/dev/ttyUSB0` instead of `/dev/serial/...`  or anything

lux73, 2023-06-07:
if the above don't work show output

`ls -l /dev/ttyU*`

EPicLURcher, 2023-06-07:
> `sudo ls -l /dev/ttyUSB0
> 
> ls: cannot access '/dev/ttyUSB0': No such file or directory
> 
> pi@raspberrypi:~ $ ls -l /dev/ttyU*
> 
> ls: cannot access '/dev/ttyU*': No such file or directory
> 
> `
> 
> 
> 
> Both don't seem to exist :S 
> 
> Well, the problem is kind of mitigated by just using the other name, and obviously had nothing to do with zigbee2mqtt, which was just affected by the underlying issue (usb device changed path and name).
> 
> 
> 
> Still, 2 questions for me:
> 
> 1. has anybody experienced this yet? What could cause something like this?
> 
> 2. even though it seems to work now, I got an error when starting with "npm start". Since it still worked, I guess it's nothing critical. Or is it?
> 
> 
> 
> @Koenkk  you can close this if you want, since a) solved/mitigated and b) not caused by zigbee2mqtt.
> 
> Would still highly appreciate if someone could shed some light on why my usb stick decided to do that on its own :D 

I'm seeing this exact same behaviour as of the last week or so - I did with lots of restarting have everything working again - so I don't think it's my dongle (sonoff) - but it is not working again tonight

Schmetterfliege, 2023-06-08:
> 

"the exact same thing" would be that your dongle's path and name has changed too and that's why zigbee2mqtt doesn't start anymore: in the configuration.yaml you specify the path, and since this has changed, it is not valid anymore.
Did you try to do what I did?
-> check if there's a device somewhere else (for me in /dev/serial/by-path) and replace the path in the configuration.yaml

EPicLURcher, 2023-06-08:
> "the exact same thing" would be that your dongle's path and name has changed too and that's why zigbee2mqtt doesn't start anymore: in the configuration.yaml you specify the path, and since this has [...]

Yes the path of my adaptor had changed - I got the new path and updated my config file and all is fixed

KiCADHugo, 2023-06-10:
It is a bug in Debian. I have updated my "old" Debian v11.6.0 with "apt upgrade". After this upgrade my Sonoff USB Dongle Plus can't found by /dev/serial/by-id but only on /dev/serial/by-path. The link on /dev/ttyACM0 is not accept by zigbee2mqtt so it not work.
I habe try a new SD-Card written with Pi Imager, after install my stick is shown on /dev/serial/by-id, after apt upgrade stick only see on by-path.
Sorry for my bad english, I hope you understud.

Schmetterfliege, 2023-06-10:
> It is a bug in Debian. I have updated my "old" Debian v11.6.0 with "apt upgrade". After this upgrade my Sonoff USB Dongle Plus can't found by /dev/serial/by-id but only on /dev/serial/by-path. The [...]

Thank you very much for your effort, and yes: it is very clear what you mean :)
Make sense that it is caused by debian, but for me it seemed to happen while I was not at home, and I'm pretty sure my Pi just didn't upgrade the OS by itself :D But at least we know now what it is, and how it can be fixed/mitigated (using the by-path path)