Koenkk/zigbee2mqtt #17948: Payload changed in TS0044 and TS0043 after upgrade to 1.31.1

shalak, 2023-06-07:
### What happened?

After Home Assistant addon upgrade to 1.31.1-1 (which ships with 1.31.1 z2m), my Tuya-compatible multi-scene remote switch are no longer  sending the same payloads.

For e.g. switch [TS0043](https://www.zigbee2mqtt.io/devices/TS0043.html#tuya-ts0043) is now only sending payloads `single`, while it used to send `single_1`/`single_2`/`single_3` before the update.

Similarly, another remote ([TS0044](https://www.zigbee2mqtt.io/devices/TS0044.html#tuya-ts0044) is no longer sending `single_bottom_left`, `single_bottom_right`, etc, but just `single`. 

Full log for the latter now looks like this:

```
info 2023-06-07 21:06:11MQTT publish: topic 'zigbee2mqtt/Kwadraciak', payload '{"action":"single","battery":100,"linkquality":255,"voltage":3000}'
info 2023-06-07 21:06:11MQTT publish: topic 'zigbee2mqtt/Kwadraciak', payload '{"action":"","battery":100,"linkquality":255,"voltage":3000}'
info 2023-06-07 21:06:11MQTT publish: topic 'zigbee2mqtt/Kwadraciak/action', payload 'single'
```

### Zigbee2MQTT version

1.31.1

### Adapter firmware version

0x26780700

### Adapter

ConBee2/

konrad861, 2023-06-07:
I have exactly the same issue with all my TuYa TS004F and TuYa TS0042 (after update to  1.31.1-1 )

andreidelait, 2023-06-07:
The same is happening with [TS004F](https://www.zigbee2mqtt.io/devices/TS004F.html)

thrhymes, 2023-06-07:
i am seeing the same thing - and i use those alot all around the house

shalak, 2023-06-07:
@andreidelait 

> I just got back to previous version until this is fixed.

Are you using HassOS? Any idea how to downgrade an add-on?

KoKolaj, 2023-06-07:
I have exactly the same issue with all my TuYa TS0044

andreidelait, 2023-06-07:
> @andreidelait
> 
> > I just got back to previous version until this is fixed.
> 
> Are you using HassOS? Any idea how to downgrade an add-on? To what exact version did you downgrade?

Yes I'm using HassOS. I created a backup when I did uptdated z2m, so then I've just restored that backup and I'm back to 1.30.4-1

Koenkk, 2023-06-07:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/17942#issuecomment-1581387469, should be fixed in z2m 1.31.2

closed 2023-06-07 by Koenkk

andreidelait, 2023-06-07:
> @Koenkk this is out of scope of this issue, but are you aware of any way to downgrade an addon in hassio?

1.31.2-1 is already out so you should try this one instead

KoKolaj, 2023-06-07:
OK, repaired. Thank you big.

thrhymes, 2023-06-07:
Release often.... i like that agility... amazing! this took no time at all...