Koenkk/zigbee2mqtt #17952: 1.31.2-dev (commit: 7df6b2b) exception

Ivano62, 2023-06-07:
### What happened?

After upgrade to latest dev version 1.31.2-dev (commit: 7df6b2b) I get this error

Zigbee2MQTT:error 2023-06-07 22:20:57: Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '17320'}
Zigbee2MQTT:error 2023-06-07 22:21:19: Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '17320'}
Zigbee2MQTT:error 2023-06-07 22:26:13: Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '17320'}
Zigbee2MQTT:error 2023-06-07 22:30:04: Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '17320'}
Zigbee2MQTT:error 2023-06-07 22:33:26: Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '17320'}
Zigbee2MQTT:error 2023-06-07 22:34:24: Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '17320'}

### What did you expect to happen?

No errors

### How to reproduce it (minimal and precise)

I upgraded and build

### Zigbee2MQTT version

1.31.2

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

### Debug log

Zigbee2MQTT:error 2023-06-07 22:20:57: Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '17320'}
Zigbee2MQTT:error 2023-06-07 22:21:19: Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '17320'}
Zigbee2MQTT:error 2023-06-07 22:26:13: Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '17320'}
Zigbee2MQTT:error 2023-06-07 22:30:04: Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '17320'}
Zigbee2MQTT:error 2023-06-07 22:33:26: Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '17320'}
Zigbee2MQTT:error 2023-06-07 22:34:24: Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '17320'}

Koenkk, 2023-06-07:
Can you provide the debug log of this and your data/database.db file?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

antibill51, 2023-06-08:
Hi, 
I've got the same error : 
`Zigbee2MQTT:error 2023-06-08 07:14:17: Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '17320'}
Zigbee2MQTT:debug 2023-06-08 07:14:17: Error: Expected one of: 1, 4, got: '17320'
    at getFromLookup (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:512:37)
    at numericAttributes2Payload (/app/node_modules/zigbee-herdsman-converters/src/lib/xiaomi.ts:193:48)
    at Object.numericAttributes2Payload (/app/node_modules/zigbee-herdsman-converters/src/lib/xiaomi.ts:703:44)
    at Object.convert (/app/node_modules/zigbee-herdsman-converters/src/converters/fromZigbee.js:3912:33)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:150:51)
    at EventEmitter.emit (node:events:525:35)
    at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22)
    at Controller.<anonymous> (/app/lib/zigbee.ts:106:27)
    at Controller.emit (node:events:513:28)
    at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)`

database.db : 
`{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0022811af8","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,10,11,12,13,47,110,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"10":{"profId":260,"epId":10,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[1281,10],"outClusterList":[1280,1282],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":1670510528971,"defaultSendRequestWhen":"immediate"}
{"id":2,"type":"Group","groupID":1,"members":[{"deviceIeeeAddr":"0x540f57fffec13767","endpointID":1},{"deviceIeeeAddr":"0x540f57fffeb9b9db","endpointID":1},{"deviceIeeeAddr":"0x540f57fffe1920a6","endpointID":1},{"deviceIeeeAddr":"0x540f57fffe192d42","endpointID":1},{"deviceIeeeAddr":"0x540f57fffeb9b991","endpointID":1},{"deviceIeeeAddr":"0x540f57fffe192cc6","endpointID":1},{"deviceIeeeAddr":"0x540f57fffe192d6f","endpointID":1},{"deviceIeeeAddr":"0x540f57fffe1920ca","endpointID":1},{"deviceIeeeAddr":"0x540f57fffe191f9f","endpointID":1},{"deviceIeeeAddr":"0x540f57fffeb9ba1a","endpointID":1},{"deviceIeeeAddr":"0x540f57fffec1360a","endpointID":1},{"deviceIeeeAddr":"0x540f57fffeb9b526","endpointID":1},{"deviceIeeeAddr":"0x540f57fffe191fa7","endpointID":1}],"meta":{}}
{"id":3,"type":"Group","groupID":2,"members":[{"deviceIeeeAddr":"0x540f57fffec13767","endpointID":1},{"deviceIeeeAddr":"0x540f57fffeb9b9db","endpointID":1},{"deviceIeeeAddr":"0x540f57fffe1920a6","endpointID":1},{"deviceIeeeAddr":"0x540f57fffe192d42","endpointID":1},{"deviceIeeeAddr":"0x540f57fffeb9b991","endpointID":1},{"deviceIeeeAddr":"0x540f57fffe192cc6","endpointID":1},{"deviceIeeeAddr":"0x540f57fffe192d6f","endpointID":1},{"deviceIeeeAddr":"0x540f57fffe1920ca","endpointID":1}],"meta":{}}
{"id":4,"type":"Group","groupID":3,"members":[{"deviceIeeeAddr":"0x540f57fffe192cc6","endpointID":1},{"deviceIeeeAddr":"0x540f57fffe192d6f","endpointID":1},{"deviceIeeeAddr":"0x540f57fffe1920ca","endpointID":1},{"deviceIeeeAddr":"0x540f57fffe191f9f","endpointID":1},{"deviceIeeeAddr":"0x540f57fffeb9ba1a","endpointID":1},{"deviceIeeeAddr":"0x540f57fffec1360a","endpointID":1},{"deviceIeeeAddr":"0x540f57fffeb9b526","endpointID":1},{"deviceIeeeAddr":"0x540f57fffe191fa7","endpointID":1}],"meta":{}}
{"id":5,"type":"Group","groupID":4,"members":[{"deviceIeeeAddr":"0x540f57fffe192d42","endpointID":1},{"deviceIeeeAddr":"0x540f57fffeb9b991","endpointID":1}],"meta":{}}
{"id":6,"type":"Group","groupID":5,"members":[{"deviceIeeeAddr":"0x540f57fffec13767","endpointID":1},{"deviceIeeeAddr":"0x540f57fffe192cc6","endpointID":1},{"deviceIeeeAddr":"0x540f57fffec1360a","endpointID":1}],"meta":{}}
[... cut ...] "outClusterList":[],"clusters":{"genBasic":{"attributes":{"65282":[{"elmType":16,"elmVal":1},{"elmType":33,"elmVal":3012},{"elmType":33,"elmVal":17320},{"elmType":36,"elmVal":[0,1]},{"elmType":33,"elmVal":222},{"elmType":32,"elmVal":89}],"modelId":"lumi.sensor_switch","appVersion":10}},"genOnOff":{"attributes":{"32768":2,"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":10,"interviewCompleted":true,"meta":{},"lastSeen":1686198311911,"defaultSendRequestWhen":"immediate"}`

Koenkk, 2023-06-08:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-06-08 by Koenkk