Koenkk/zigbee2mqtt #17953: [New device support]: LTECH

lychee512, 2023-06-08:
### Link

http://www.ltechonline.com/html/en/products/Intelligence/zigbee/SE-20-250-1000-W2Z2.html

### Database entry

{"id":11,"type":"Router","ieeeAddr":"0x9035eafffed56882","nwkAddr":35277,"manufId":4098,"manufName":"_TZE200_2gtsuokt","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_2gtsuokt","powerSource":1,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":"","appVersion":72}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1676534437},"lastSeen":1686202816998,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to write my own converter for these TuYa-based LED drivers. Everything works fine from the zigbee2mqtt front end:
[image]

However I am having trouble getting the Home Assistant auto discovery to work completely. It seems that only the on-off status of the switch gets exposed: 
[image]
Turning this light on and off from Home Assistant does work, but I can't figure out how to get the brightness and color temperature to show up.

From the debug window of the Home Assistant Device, I got the following payload for the mqtt discovery data, but I don't know if something is missing or misconfigured from this. 
```
availability:
  - topic: zigbee2mqtt/bridge/state
    value_template: '{{ value_json.state }}'
brightness: true
brightness_scale: 254
color_mode: true
command_topic: zigbee2mqtt/living_window_left/set
device:
  identifiers:
    - zigbee2mqtt_0x9035eafffed55e95
  manufacturer: LTECH
  model: Tuya LED for Vortex by Verde Lights (TY-20-250-1000-W2Z2)
  name: living_window_left
max_mireds: 454
min_mireds: 250
name: living_window_left
schema: json
state_topic: zigbee2mqtt/living_window_left
supported_color_modes:
  - color_temp
unique_id: 0x9035eafffed55e95_light_zigbee2mqtt
platform: mqtt
```
Any assistance would be appreciated!
zigbee2mqtt version: [1.30.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.30.2) commit: [cdf62ea](https://github.com/Koenkk/zigbee2mqtt/commit/cdf62ea)
homeassistant version: Home Assistant 2023.3.3, Frontend 20230309.0 - latest

Jingyee

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
  fingerprint: [
      {modelID: 'TS0601', manufacturerName: '_TZE200_2gtsuokt'},
  ],
  model: 'TY-20-250-1000-W2Z2',
  vendor: 'LTECH',
  description: 'Tuya LED for Vortex by Verde Lights',
  fromZigbee: [fz.tuya_light_wz5],
  toZigbee: [tz.tuya_dimmer_state, tz.tuya_light_wz5],
  exposes: [
      exposes.light().withBrightness().setAccess('state', ea.STATE_SET).setAccess('brightness',
          ea.STATE_SET).withColorTemp([250, 454]).setAccess('color_temp',
          ea.STATE_SET),
      //exposes.numeric('white_brightness', ea.STATE_SET).withValueMin(0).withValueMax(254).withDescription(
      //    'White brightness of this light'),
  ],
  //meta: {separateWhite: true},
};

module.exports = definition;
```

### Supported color modes

color temperature

Koenkk, 2023-06-08:
> However I am having trouble getting the Home Assistant auto discovery to work completely. It seems that only the on-off status of the switch gets exposed:

If you click on the light you should see all the other controls (brightness/color)

lychee512, 2023-06-08:
Clicking on the light toggle in Home Assistant UI does not cause the other controls to appear, although it does turn on the light itself (sending messages to the light's mqtt topic can also update the state of the light without issue). 

No entities for the light color and brightness are created by the auto discovery. The linkquality entity also doesn't work (zigbee2mqtt front end reports a number, but home assistant can't see it), this entity might have been created while I was trying out different variations of the external converter, but currently it appears to be disabled by the zigbee2mqtt integration in Home Assistant.

Koenkk, 2023-06-08:
What if you long press it? For light color and brightness now separate entities will be created

lychee512, 2023-06-09:
OK, turns out the mistake is between the keyboard and the chair, I didn't realise that the brightness and color temps in the home assistant UI are hidden behind another menu. I'm closing this issue, and will submit a PR with the converter for this device once I figure out how to do it.

closed 2023-06-09 by lychee512

Testingfor101, 2023-06-28:
Is your external converter file is without errors?
Because mine when restarting gets in loop.

Error:
[20:09:02] INFO: Preparing to start...
[20:09:02] INFO: Socat not enabled
[20:09:04] INFO: Starting Zigbee2MQTT...
/app/node_modules/zigbee-herdsman-converters/index.js:79
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:79:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)
    
Config:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const color = require('zigbee-herdsman-converters/lib/color');
const light = require('zigbee-herdsman-converters/lib/light');
const ota = require('zigbee-herdsman-converters/lib/ota');

const definition = {
	fingerprint: [
		{
			// The model ID from: Device with modelID 'TS0601' is not supported
			// You may need to add \u0000 at the end of the name in some cases
			modelID: 'TS0601',
			// The manufacturer name from: Device with modelID 'TS0601' is not supported.
			manufacturerName: '_TZE200_drs6j6m5',
		},
	],    
	model: 'TY-12-100-400-W1Z', // Vendor model number, look on the device for a model number
	vendor: 'Tuya', // Vendor of the device (only used for documentation and startup logging)
	description: 'Lifud CV Zigbee LED Driver',
	fromZigbee: [fz.tuya_light_wz5],
	toZigbee: [tz.tuya_dimmer_state, tz.tuya_light_wz5],
	exposes: [
		exposes.light().withBrightness().setAccess('state', ea.STATE_SET).setAccess('brightness',
			ea.STATE_SET).withColorTemp([250, 454]).setAccess('color_temp',
			ea.STATE_SET),
		//exposes.numeric('white_brightness', ea.STATE_SET).withValueMin(0).withValueMax(254).withDescription(
		//    'White brightness of this light'),
	],
	//meta: {separateWhite: true},
};

module.exports = definition;
```

MadMirza, 2023-07-11:
I tried out the external convertor here, but z2m fails to start with an error.

[13:59:50] INFO: Preparing to start...
[13:59:50] INFO: Socat not enabled
[13:59:51] INFO: Starting Zigbee2MQTT...
/app/node_modules/zigbee-herdsman-converters/index.js:79
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:79:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)

lychee512, 2023-07-12:
probably related to this issue: https://github.com/Koenkk/zigbee2mqtt/issues/18177
Maybe you can check which version of z2m you are using.

MadMirza, 2023-07-13:
i am using the latest version of Z2M.

I have tried creating one on my own, my code understanding skills are quite bad...

I tried mapping the datapoints as an addition to the Converter here. 

get as error that says no convertor available for color temp

MadMirza, 2023-07-13:
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
	fingerprint: [
		{
			// The model ID from: Device with modelID 'TS0601' is not supported
			// You may need to add \u0000 at the end of the name in some cases
			modelID: 'TS0601',
			// The manufacturer name from: Device with modelID 'TS0601' is not supported.
			manufacturerName: '_TZE200_2gtsuokt',
		},
	],    
	model: 'TY-12-100-400-W1Z', // Vendor model number, look on the device for a model number
	vendor: 'Tuya', // Vendor of the device (only used for documentation and startup logging)
	description: 'LTECH CCT driver',
	fromZigbee: [tuya.fz.datapoints],
	toZigbee: [tuya.tz.datapoints],
	configure: tuya.configureMagicPacket,
	exposes: [exposes.light().withBrightness().setAccess('state', ea.STATE_SET).setAccess('brightness',
	ea.STATE_SET).withColorTemp([250, 454]).setAccess('color_temp',
	ea.STATE_SET)],
	
	meta: {
		tuyaDatapoints: [
			[1, 'state', tuya.valueConverter.onOff, {skip: tuya.skip.stateOnAndBrightnessPresent}],
			[3, 'brightness', tuya.valueConverter.scale0_254to0_1000],
			[4, 'color_temp',tuya.valueConverter.scale0_254to0_1000]],
			[14, 'power_on_behavior', tuya.valueConverter.powerOnBehavior],
		],
	},
};
	
module.exports = definition;`

MadMirza, 2023-07-13:
This what I have tried so far.
```
const fz = require("zigbee-herdsman-converters/converters/fromZigbee");
const tz = require("zigbee-herdsman-converters/converters/toZigbee");
const exposes = require("zigbee-herdsman-converters/lib/exposes");
const reporting = require("zigbee-herdsman-converters/lib/reporting");
const extend = require("zigbee-herdsman-converters/lib/extend");
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
	fingerprint: [
		{
			// The model ID from: Device with modelID 'TS0601' is not supported
			// You may need to add \u0000 at the end of the name in some cases
			modelID: "TS0601",
			// The manufacturer name from: Device with modelID 'TS0601' is not supported.
			manufacturerName: "_TZE200_2gtsuokt",
		},
	],
	model: "TY-12-100-400-W1Z", // Vendor model number, look on the device for a model number
	vendor: "Tuya", // Vendor of the device (only used for documentation and startup logging)
	description: "LTECH CCT driver",
	fromZigbee: [tuya.fz.datapoints],
	toZigbee: [tuya.tz.datapoints],
	configure: tuya.configureMagicPacket,
	exposes: [
		exposes
			.light()
			.withBrightness()
			.setAccess("state", ea.STATE_SET)
			.setAccess("brightness", ea.STATE_SET)
			.withColorTemp([250, 454])
			.setAccess("color_temp", ea.STATE_SET),
	],

	meta: {
		tuyaDatapoints: [
			[
				1,
				"state",
				tuya.valueConverter.onOff,
				{ skip: tuya.skip.stateOnAndBrightnessPresent },
			],
			[3, "brightness", tuya.valueConverter.scale0_254to0_1000],
			[4, "color_temp"],
			[14, "power_on_behavior", tuya.valueConverter.powerOnBehavior],
		],
	},
};

module.exports = definition;
```

VyckaLTU, 2023-07-28:
Hi. Is it possible to make it work with LTECH TY-75-24-G2Z2 driver? Added it to home assistant but is shows as unsupported device :/ Thank you for your help.

mogim0gi, 2023-11-04:
> LTECH TY-75-24-G2Z2 is listed as supported device now : https://www.zigbee2mqtt.io/devices/TY-75-24-G2Z2.html
> 
> however my zigbee2mqtt discovery does not find it

Still cannot pair it

Testingfor101, 2023-11-05:
> > LTECH TY-75-24-G2Z2 is listed as supported device now : https://www.zigbee2mqtt.io/devices/TY-75-24-G2Z2.html
> > however my zigbee2mqtt discovery does not find it
> 
> Still cannot pair it

Add some info about device, where it shows unsupported in zigbee addon window.

VyckaLTU, 2023-11-10:
I have 2 LTECH drivers. They just have different power.
Ltech TY-75-24-G2Z2
and
Ltech TY-150-24-G2Z2

One is 75W and other is 150W.
75W is working and appears on z2m. But 150W still shows as unsupported :(

[image]

DJJo14, 2023-11-14:
> I don't know what breaking changes were introduced in the latest version(s), but the one that I wrote is for an older version of Z2M. Essentially I just modified the external converter for skydance 5-in-1 CCT lights. The latest version of that controller can be found here:
> 
> https://github.com/Koenkk/zigbee-herdsman-converters/blob/16b19d982a2e4eeb70c3cff469b52eeb2223ba0f/src/devices/skydance.ts

Tanks

Any one having this device is the external converter i used:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition =     {
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE200_2gtsuokt'}
    ],
    model: 'WZ5_cct',
    vendor: 'Skydance',
    description: 'Zigbee & RF 5 in 1 LED controller (CCT mode)',
    fromZigbee: [legacy.fz.tuya_light_wz5],
    toZigbee: [legacy.tz.tuya_dimmer_state, legacy.tz.tuya_light_wz5],
    exposes: [
        e.light().withBrightness().setAccess('state',
            ea.STATE_SET).setAccess('brightness', ea.STATE_SET).withColorTemp([250, 454]).setAccess('color_temp', ea.STATE_SET),
    ],
};

module.exports = definition;
```

mogim0gi, 2023-11-15:
> I have 2 LTECH drivers. They just have different power. Ltech TY-75-24-G2Z2 and Ltech TY-150-24-G2Z2
> 
> One is 75W and other is 150W. Which one it is can be seen in model name. 75W is working and appears on z2m. But 150W still shows as unsupported :(
> 
> In here - https://www.zigbee2mqtt.io/devices/TY-75-24-G2Z2.html It says 75W model but in description it says 150W. Which is wrong. I guess it is just a naming problem. Maybe someone form Z2M can check this one and add both models?
> 
> Thank you.
> 
> [image]

How did you manage to pair 75w version? I restat mine multiple times by 5 power cycles but it never attempts to join z2m network

mogim0gi, 2023-11-15:
Managed to pair my LM-75-24-G2Z2 however it has no light temperature control, only brightness.

[image]
[image]

bummzulu, 2023-11-24:
> I have 2 LTECH drivers. They just have different power. Ltech TY-75-24-G2Z2 and Ltech TY-150-24-G2Z2
>
> One is 75W and other is 150W. Which one it is can be seen in model name. 75W is working and [...]

I had the same issue and I fixed this with the instructions given here.

The had to modify the converter model a bit to fit to the 150W model:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition =     {
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '**_TZE200_na98lvjp**'}
    ],
    model: 'WZ5_cct',
    vendor: 'Skydance',
    description: 'Zigbee & RF 5 in 1 LED controller (CCT mode)',
    fromZigbee: [legacy.fz.tuya_light_wz5],
    toZigbee: [legacy.tz.tuya_dimmer_state, legacy.tz.tuya_light_wz5],
    exposes: [
        e.light().withBrightness().setAccess('state', ea.STATE_SET).setAccess('brightness', ea.STATE_SET).withColorTemp([250, 454]).setAccess('color_temp', ea.STATE_SET),
    ],
};

module.exports = definition;
```

VyckaLTU, 2023-11-27:
@bummzulu  Can you please help me by explaining little bit more? Where I can find this converter model? Im still in learning phase with HomeAssistant and Z2M :)

mogim0gi, 2024-01-01:
> Managed to pair my LM-75-24-G2Z2 however it has no light temperature control, only brightness.
> 
> ![IMG_3295](https://private-user-images.githubusercontent.com/64898710/283160011-53ab7113-d24d-4a13-97fe-9d089bc603bc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQxMjgyOTIsIm5iZiI6MTcwNDEyNzk5MiwicGF0aCI6Ii82NDg5ODcxMC8yODMxNjAwMTEtNTNhYjcxMTMtZDI0ZC00YTEzLTk3ZmUtOWQwODliYzYwM2JjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTAxVDE2NTMxMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZlNmEyZTdjMTA1ZTk1MDZmMTM1NGQzMTBkYjM5Mzc2MTJmMjY5ZDlmNmQ3NmM0NzAyYjEwOWJlY2E2Y2FlNTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.TxA5gujra5thpQHR13kGMyxFtLvjLdd3G4Yam1peE1Y) ![image](https://private-user-images.githubusercontent.com/64898710/283160104-2129e988-9ad3-496b-b36b-5bc0bf549160.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQxMjgyOTIsIm5iZiI6MTcwNDEyNzk5MiwicGF0aCI6Ii82NDg5ODcxMC8yODMxNjAxMDQtMjEyOWU5ODgtOWFkMy00OTZiLWIzNmItNWJjMGJmNTQ5MTYwLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTAxVDE2NTMxMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTEyMDhhMWI4MTY4NGQ3ZThiODI4OTUwNTJjNGVkNmY1YmNjODVhYzRlNGEwMDNmZDFhNTMxOWJkYTQzNWMyYjImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.sjNUhIMHkoy8QeUVxeWYLtKMci44b5gJGTALuqLSGRs)

@Koenkk  - can you please take a look into this. This model supports Color temperature, however its not available in Zigbee2mqtt

Koenkk, 2024-01-02:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

petkuniokas, 2024-01-31:
Hey,
maybe someone is lucky and paired LTECH devices?