Koenkk/zigbee2mqtt #17955: Error message does not show device

vincele, 2023-06-08:
### What happened?

I get a few of the following error messages:

    Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'power')}

But this is not really useful as I don't know which device it is that has the problem.

### What did you expect to happen?

Can the error messages be enhanced to always show the relevant device ?

### Zigbee2MQTT version

1.30.4

### Adapter firmware version

20210708

### Adapter

Sonoff Zigbee 3.0 USB Dongle P

Koenkk, 2023-06-08:
Please provide the debug log of this.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

vincele, 2023-06-08:
I'm not sure this is what you want because there's not really much more interesting info in there...

    Jun  8 11:36:25 z2m daemon.info localhost Zigbee2MQTT[4356]: Zigbee2MQTT started!
    Jun  8 11:36:25 z2m daemon.info localhost Zigbee2MQTT[4356]: MQTT publish: topic 'zigbee2mqtt/ZLinky', payload '{"active_register_tier_delivered":"HP..","apparent_power":320,"available_power":45,"current_tarif":"HC..","current_tier1_summ_delivered":19653,"current_tier2_summ_delivered":26090,"linkquality":60,"meter_serial_number":"XXXXXXXXXXXX","rms_current":1,"rms_current_max":90,"schedule_h_p_h_c":0,"update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null,"warn_d_p_s":0}'
    Jun  8 11:36:44 z2m daemon.info localhost Zigbee2MQTT[4356]: MQTT publish: topic 'zigbee2mqtt/TH - Bureau', payload '{"battery":100,"humidity":46.88,"linkquality":144,"temperature":23.71,"voltage":3000}'
    Jun  8 11:36:45 z2m daemon.info localhost Zigbee2MQTT[4356]: MQTT publish: topic 'zigbee2mqtt/TH - Bureau', payload '{"battery":100,"humidity":46.55,"linkquality":144,"temperature":23.71,"voltage":3000}'
    Jun  8 11:36:59 z2m daemon.info localhost Zigbee2MQTT[4356]: MQTT publish: topic 'zigbee2mqtt/Prise - Cuisine', payload '{"child_lock":"LOCK","current":0.05,"energy":0.05,"indicator_mode":"off","linkquality":33,"power":11,"power_apparent":null,"power_outage_memory":"off","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":226}'
    Jun  8 11:37:00 z2m daemon.info localhost Zigbee2MQTT[4356]: MQTT publish: topic 'zigbee2mqtt/Prise - Cuisine', payload '{"child_lock":"UNLOCK","current":0.05,"energy":0.05,"indicator_mode":"off","linkquality":33,"power":11,"power_apparent":null,"power_outage_memory":"off","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":226}'
    Jun  8 11:37:01 z2m daemon.info localhost Zigbee2MQTT[4356]: MQTT publish: topic 'zigbee2mqtt/Prise - Cuisine', payload '{"child_lock":"LOCK","current":0.05,"energy":0.05,"indicator_mode":"off","linkquality":33,"power":11,"power_apparent":null,"power_outage_memory":"off","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":226}'
    Jun  8 11:37:02 z2m daemon.info localhost Zigbee2MQTT[4356]: MQTT publish: topic 'zigbee2mqtt/Prise - Cuisine', payload '{"child_lock":"UNLOCK","current":0.05,"energy":0.05,"indicator_mode":"off","linkquality":36,"power":11,"power_apparent":null,"power_outage_memory":"off","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":226}'
    Jun  8 11:37:03 z2m daemon.info localhost Zigbee2MQTT[4356]: MQTT publish: topic 'zigbee2mqtt/Prise - Salon', payload '{"child_lock":"LOCK","current":0,"energy":0.12,"indicator_mode":"off","linkquality":9,"power":0,"power_outage_memory":"off","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":227}'
    Jun  8 11:37:04 z2m daemon.info localhost Zigbee2MQTT[4356]: MQTT publish: topic 'zigbee2mqtt/Prise - Salon', payload '{"child_lock":"UNLOCK","current":0,"energy":0.12,"indicator_mode":"off","linkquality":12,"power":0,"power_outage_memory":"off","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":227}'
    Jun  8 11:37:05 z2m daemon.info localhost Zigbee2MQTT[4356]: MQTT publish: topic 'zigbee2mqtt/Prise - Salon', payload '{"child_lock":"LOCK","current":0,"energy":0.12,"indicator_mode":"off","linkquality":9,"power":0,"power_outage_memory":"off","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":227}'
    Jun  8 11:37:06 z2m daemon.info localhost Zigbee2MQTT[4356]: MQTT publish: topic 'zigbee2mqtt/Prise - Salon', payload '{"child_lock":"UNLOCK","current":0,"energy":0.12,"indicator_mode":"off","linkquality":9,"power":0,"power_outage_memory":"off","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":227}'
    Jun  8 11:37:16 z2m daemon.info localhost Zigbee2MQTT[4356]: MQTT publish: topic 'zigbee2mqtt/Prise - Salon', payload '{"child_lock":"UNLOCK","current":0,"energy":0.12,"indicator_mode":"off","linkquality":6,"power":0,"power_outage_memory":"off","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":227}'
    Jun  8 11:37:16 z2m daemon.info localhost Zigbee2MQTT[4356]: MQTT publish: topic 'zigbee2mqtt/Prise - Salon', payload '{"child_lock":"UNLOCK","current":0,"energy":0.12,"indicator_mode":"off","linkquality":3,"power":0,"power_outage_memory":"off","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":228}'
    Jun  8 11:37:20 z2m daemon.info localhost Zigbee2MQTT[4356]: MQTT publish: topic 'zigbee2mqtt/Prise - Salon', payload '{"child_lock":"UNLOCK","current":0,"energy":0.12,"indicator_mode":"off","linkquality":12,"power":0,"power_outage_memory":"off","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":228}'
[... cut ...]
    Jun  8 11:37:35 z2m daemon.info localhost Zigbee2MQTT[4356]: MQTT publish: topic 'zigbee2mqtt/ZLinky', payload '{"active_register_tier_delivered":"HP..","apparent_power":340,"available_power":45,"current_tarif":"HC..","current_tier1_summ_delivered":19653,"current_tier2_summ_delivered":26090,"linkquality":60,"meter_serial_number":"XXXXXXXXXXXX","rms_current":1,"rms_current_max":90,"schedule_h_p_h_c":0,"update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null,"warn_d_p_s":0}'

vincele, 2023-06-08:
Yes, I put that in my config, then restarted the service:

    z2m:/var/log# grep -B1 debug /etc/zigbee2mqtt/configuration.yaml
    advanced:
      log_level: debug

Koenkk, 2023-06-08:
That should be correct, do you see any `daemon.debug`? There should always be some, otherwise debug logging has not been activated correctly.

vincele, 2023-06-08:
I may have misconfigured something, as I can see debug logs in Z2M web log page, but in /var/log/messages I have only daemon.info entries from Z2M...

    # cat /etc/zigbee2mqtt/configuration.yaml
    permit_join: true
    mqtt:
      base_topic: zigbee2mqtt
      server: mqtt://localhost:1883
    serial:
      port: /dev/ttyUSB0
    frontend:
      port: 8080
      host: 10.0.0.18
    homeassistant: true
    advanced:
      log_level: debug
      log_output:
        - console
        - syslog
      log_directory: /var/log/zigbee2mqtt
      log_syslog:
        facility: daemon
        path: /dev/log
        protocol: unix
      network_key: '!secret.yaml network_key'
      pan_id: '!secret.yaml pan_id'
      homeassistant_legacy_entity_attributes: false
      legacy_api: false
      legacy_availability_payload: false
    devices: devices.yaml
    groups: groups.yaml
    device_options:
      legacy: false

Koenkk, 2023-06-09:
What about the log files in `/var/log/zigbee2mqtt`?

vincele, 2023-06-09:
There's nothing created there:

    z2m:~# ls -la /var/log/zigbee2mqtt
    drwxrwxr-x    2 root     zigbee2m      4096 Jun  8 11:34 .
    drwxr-xr-x    3 root     root          4096 Jun  9 09:13 ..

    z2m:~# ps faux | grep '[z]igbee'
     4356 zigbee2m  2:33 node /usr/bin/zigbee2mqtt

despite the user who runs Z2M having group permission to write in the directory

vincele, 2023-06-09:
I have this in the logs, when restarting:

    Jun  9 09:53:35 z2m daemon.info localhost Zigbee2MQTT[4805]: Logging to console only'

Maybe my config file is not usedat all...

vincele, 2023-06-09:
Looks like this is it:

    z2m:~# su -s /bin/sh - zigbee2mqtt
    z2m:~$ zigbee2mqtt --help
    Error: ENOENT: no such file or directory, open '/var/lib/zigbee2mqtt/.z2m/configuration.yaml'

I modified the config in /etc/zigbee2mqtt

I'll double-check, and close this issue then.

vincele, 2023-06-09:
I have the config & secret files in `/var/lib/zigbee2mqtt` as symlinks to `/etc/zigbee2mqtt`.

And if the symlinks are not in `/var/lib/zigbee2mqtt` the service won't start, so I assume they are properly taken from there, despite the above output from `zigbee2mqtt --help`.

So I'm totally confused, what can I try next ?

vincele, 2023-06-09:
The `zigbee2mqtt --help` error was from not passing `ZIGBEE2MQTT_DATA`.

The OpenRC service definition file properly passes it, as:

    export ZIGBEE2MQTT_DATA="${datadir:="/var/lib/zigbee2mqtt"}"

which is also the zigbee2mqtt running user's `$HOME`.

vincele, 2023-06-09:
D'oh, I'm so stupid...

What was missing:

    log_output:
        - file

Now I get something in /var/log/zigbee2mqtt...
And the debug logs are in there.

Sorry for the time lost.

:-/

vincele, 2023-06-09:
So, back to the issue...

Here is a log file extract from around the issue message:

    info  2023-06-09 11:18:03: MQTT publish: topic 'zigbee2mqtt/Prise - Cuisine', payload '{"child_lock":"LOCK","current":0.05,"energy":0.05,"indicator_mode":"off","li
    nkquality":54,"power":11,"power_apparent":null,"power_outage_memory":"off","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"upd
    ate_available":null,"voltage":226}'
    
    debug 2023-06-09 11:18:03: Received Zigbee message from 'Prise - Cuisine', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":53}' from
     endpoint 1 with groupID 0
    
    error 2023-06-09 11:18:03: Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'power')}
    
    debug 2023-06-09 11:18:03: TypeError: Cannot read properties of undefined (reading 'power')
        at Object.convert (/usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/tuya.js:906:31)
        at Receive.onDeviceMessage (/usr/lib/zigbee2mqtt/lib/extension/receive.ts:150:51)
        at EventEmitter.emit (node:events:525:35)
        at EventBus.emitDeviceMessage (/usr/lib/zigbee2mqtt/lib/eventBus.ts:102:22)
        at Controller.<anonymous> (/usr/lib/zigbee2mqtt/lib/zigbee.ts:106:27)
        at Controller.emit (node:events:513:28)
        at Controller.selfAndDeviceEmit (/usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:417:14)
        at Controller.onZclOrRawData (/usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:595:18)
        at ZStackAdapter.<anonymous> (/usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:124:73)
        at ZStackAdapter.emit (node:events:513:28)
    
    debug 2023-06-09 11:18:09: Received Zigbee message from 'Prise - Salon', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":227}' from
    endpoint 1 with groupID 0
    
    info  2023-06-09 11:18:09: MQTT publish: topic 'zigbee2mqtt/Prise - Salon', payload '{"child_lock":"UNLOCK","current":0,"energy":0.12,"indicator_mode":"off","linkq
    uality":12,"power":0,"power_outage_memory":"off","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"volt
    age":227}'
    
    info  2023-06-09 11:18:11: MQTT publish: topic 'zigbee2mqtt/Prise - Cuisine', payload '{"child_lock":"LOCK","current":0.05,"energy":0.05,"indicator_mode":"off","li
    nkquality":0,"power":11,"power_apparent":null,"power_outage_memory":"off","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"upda
    te_available":null,"voltage":226}'
    
    debug 2023-06-09 11:18:11: Received Zigbee message from 'Prise - Cuisine', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with grou
    pID 0

Is that sufficient or do you need more ?

vincele, 2023-06-09:
The device is probably:

    Zigbee Model: TS011F
    Zigbee Manufacturer: _TZ3000_gvn91tmx

Koenkk, 2023-06-10:
Can you also check with z2m 1.31.2 and provide me the debug logging if the error still occurs?

vincele, 2023-06-10:
OK, I'll try to do that, but it'll take time as I need to learn how to make a new package on alpine.

OTOH, that issue was not about fixing this error from happening, but for fixing the error message from not giving us enough information.

vincele, 2023-06-11:
Looks like the message does not appear any more, so this issue has been sufficiently worked around.
Thanks

closed 2023-06-11 by vincele