Koenkk/zigbee2mqtt #17956: fut037z+ RGB+CCT led controller

JabaMS, 2023-06-08:
### What happened?

With zigbee2mqtt version 1.30.4 this device was recognized as TuYa TS0505B_1 and more or less it worked normally.

Problems in version 1.30.4:

- Changing between color_temp and color_xy does not recognize current brightness, but instead it uses previously used 
- Color effects does not work
- Transition time does not work
- color_sync does not work

Since update 1.31.0 it's not possible to use controller thru zigbee2mqtt frontend. 

Problems in version 1.31.0:

- Changing from color_temp to color_xy is possible, but not from color_xy to color_temp (it works as expected when doing this thru HA card). 
- Once changed from color_temp to color_xy it wan't turn off device  (it works as expected when doing this thru HA card).
- If I manage to turn of device after couple of tries, on turn on it tries to turn on color_temp, than it switches to color_xy in a half of second
- Reading current brightness not possible
- On settings page, focus in and out of transition field triggers save setting action (no matter is there a change in input field) 
- All previously problems I wrote exists

### How to reproduce it (minimal and precise)

Make a change between color modes thru Z2M page. Try turn on/off after that.

### Zigbee2MQTT version

1.31.0

### Adapter firmware version

20220301

### Adapter

zStack3x0 CC2652

### Debug log

`Debug 2023-06-08 10:24:07Publishing 'set' 'color' to 'zbLedTrakaPc'
Info 2023-06-08 10:24:07MQTT publish: topic 'zigbee2mqtt/zbLedTrakaPc', payload '{"brightness":118,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.1355,"y":0.0399},"color_mode":"xy","color_power_on_behavior":null,"color_temp":574,"do_not_disturb":null,"last_seen":"2023-06-08T10:24:08+02:00","linkquality":76,"state":"ON"}'
Debug 2023-06-08 10:24:07Publishing 'set' 'effect' to 'zbLedTrakaPc'
Info 2023-06-08 10:24:07MQTT publish: topic 'zigbee2mqtt/zbLedTrakaPc', payload '{"brightness":118,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.1355,"y":0.0399},"color_mode":"xy","color_power_on_behavior":null,"color_temp":574,"do_not_disturb":null,"last_seen":"2023-06-08T10:24:08+02:00","linkquality":80,"state":"ON"}'
Info 2023-06-08 10:24:07MQTT publish: topic 'zigbee2mqtt/zbLedTrakaPc', payload '{"brightness":118,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.1355,"y":0.0399},"color_mode":"xy","color_power_on_behavior":null,"color_temp":574,"do_not_disturb":null,"last_seen":"2023-06-08T10:24:08+02:00","linkquality":80,"state":"ON"}'
Debug 2023-06-08 10:24:08Received MQTT message on 'zigbee2mqtt/zbLedTrakaPc/set' with data '{"brightness":118,"color":{"hex":"#0000F9"},"color_temp":454,"effect":"breathe","state":"ON"}'
Debug 2023-06-08 10:24:08Publishing 'set' 'state' to 'zbLedTrakaPc'
Info 2023-06-08 10:24:08MQTT publish: topic 'zigbee2mqtt/zbLedTrakaPc', payload '{"brightness":118,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.1355,"y":0.0399},"color_mode":"xy","color_power_on_behavior":null,"color_temp":574,"do_not_disturb":null,"last_seen":"2023-06-08T10:24:09+02:00","linkquality":83,"state":"ON"}'
Debug 2023-06-08 10:24:08Publishing 'set' 'color' to 'zbLedTrakaPc'
Info 2023-06-08 10:24:08MQTT publish: topic 'zigbee2mqtt/zbLedTrakaPc', payload '{"brightness":118,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.1355,"y":0.0399},"color_mode":"xy","color_power_on_behavior":null,"color_temp":574,"do_not_disturb":null,"last_seen":"2023-06-08T10:24:09+02:00","linkquality":83,"state":"ON"}'
Debug 2023-06-08 10:24:08Publishing 'set' 'effect' to 'zbLedTrakaPc'
Info 2023-06-08 10:24:08MQTT publish: topic 'zigbee2mqtt/zbLedTrakaPc', payload '{"brightness":118,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.1355,"y":0.0399},"color_mode":"xy","color_power_on_behavior":null,"color_temp":574,"do_not_disturb":null,"last_seen":"2023-06-08T10:24:09+02:00","linkquality":80,"state":"ON"}'
Info 2023-06-08 10:24:08MQTT publish: topic 'zigbee2mqtt/zbLedTrakaPc', payload '{"brightness":118,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.1355,"y":0.0399},"color_mode":"xy","color_power_on_behavior":null,"color_temp":574,"do_not_disturb":null,"last_seen":"2023-06-08T10:24:09+02:00","linkquality":80,"state":"ON"}'
Debug 2023-06-08 10:24:09Received MQTT message on 'zigbee2mqtt/zbLedTrakaPc/set' with data '{"brightness":118,"color":{"hex":"#0000F9"},"color_temp":370,"effect":"breathe","state":"ON"}'
Debug 2023-06-08 10:24:09Publishing 'set' 'state' to 'zbLedTrakaPc'
Info 2023-06-08 10:24:09MQTT publish: topic 'zigbee2mqtt/zbLedTrakaPc', payload '{"brightness":118,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.1355,"y":0.0399},"color_mode":"xy","color_power_on_behavior":null,"color_temp":574,"do_not_disturb":null,"last_seen":"2023-06-08T10:24:10+02:00","linkquality":76,"state":"ON"}'
Debug 2023-06-08 10:24:09Publishing 'set' 'color' to 'zbLedTrakaPc'
Info 2023-06-08 10:24:09MQTT publish: topic 'zigbee2mqtt/zbLedTrakaPc', payload '{"brightness":118,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.1355,"y":0.0399},"color_mode":"xy","color_power_on_behavior":null,"color_temp":574,"do_not_disturb":null,"last_seen":"2023-06-08T10:24:10+02:00","linkquality":76,"state":"ON"}'
Debug 2023-06-08 10:24:09Publishing 'set' 'effect' to 'zbLedTrakaPc'
Info 2023-06-08 10:24:09MQTT publish: topic 'zigbee2mqtt/zbLedTrakaPc', payload '{"brightness":118,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.1355,"y":0.0399},"color_mode":"xy","color_power_on_behavior":null,"color_temp":574,"do_not_disturb":null,"last_seen":"2023-06-08T10:24:10+02:00","linkquality":80,"state":"ON"}'
Info 2023-06-08 10:24:09MQTT publish: topic 'zigbee2mqtt/zbLedTrakaPc', payload '{"brightness":118,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.1355,"y":0.0399},"color_mode":"xy","color_power_on_behavior":null,"color_temp":574,"do_not_disturb":null,"last_seen":"2023-06-08T10:24:10+02:00","linkquality":80,"state":"ON"}'
Debug 2023-06-08 10:24:10Received MQTT message on 'zigbee2mqtt/zbLedTrakaPc/get' with data '{"color_temp":""}'
Debug 2023-06-08 10:24:10Publishing get 'get' 'color_temp' to 'zbLedTrakaPc'
Debug 2023-06-08 10:24:10Received Zigbee message from 'zbLedTrakaPc', type 'readResponse', cluster 'lightingColorCtrl', data '{"colorMode":1,"colorTemperature":250,"currentHue":16,"currentSaturation":16,"currentX":8880,"currentY":2615}' from endpoint 1 with groupID 0
Info 2023-06-08 10:24:10MQTT publish: topic 'zigbee2mqtt/zbLedTrakaPc', payload '{"brightness":118,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.1355,"y":0.0399},"color_mode":"xy","color_power_on_behavior":null,"color_temp":574,"do_not_disturb":null,"last_seen":"2023-06-08T10:24:11+02:00","linkquality":80,"state":"ON"}'
Debug 2023-06-08 10:24:11Received MQTT message on 'zigbee2mqtt/zbLedTrakaPc/get' with data '{"color_temp":""}'
Debug 2023-06-08 10:24:11Publishing get 'get' 'color_temp' to 'zbLedTrakaPc'
Debug 2023-06-08 10:24:11Received Zigbee message from 'zbLedTrakaPc', type 'readResponse', cluster 'lightingColorCtrl', data '{"colorMode":1,"colorTemperature":250,"currentHue":16,"currentSaturation":16,"currentX":8880,"currentY":2615}' from endpoint 1 with groupID 0
Info 2023-06-08 10:24:11MQTT publish: topic 'zigbee2mqtt/zbLedTrakaPc', payload '{"brightness":118,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.1355,"y":0.0399},"color_mode":"xy","color_power_on_behavior":null,"color_temp":574,"do_not_disturb":null,"last_seen":"2023-06-08T10:24:12+02:00","linkquality":76,"state":"ON"}'
Debug 2023-06-08 10:24:15Received MQTT message on 'zigbee2mqtt/zbLedTrakaPc/set' with data '{"brightness":39,"color":{"hex":"#0000F9"},"color_temp":370,"effect":"breathe","state":"ON"}'
[... 30 lines cut ...]
Info 2023-06-08 10:24:20MQTT publish: topic 'zigbee2mqtt/zbLedTrakaPc', payload '{"brightness":39,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.1355,"y":0.0399},"color_mode":"xy","color_power_on_behavior":null,"color_temp":574,"do_not_disturb":null,"last_seen":"2023-06-08T10:24:21+02:00","linkquality":80,"state":"ON"}'
Debug 2023-06-08 10:24:21Received MQTT message on 'zigbee2mqtt/zbLedTrakaPc/set' with data '{"brightness":39,"color":{"hex":"#0000F9"},"color_temp":153,"effect":"breathe","state":"ON"}'
Debug 2023-06-08 10:24:21Publishing 'set' 'state' to 'zbLedTrakaPc'
[... cut ...]
Info 2023-06-08 10:24:25MQTT publish: topic 'zigbee2mqtt/zbLedTrakaPc', payload '{"brightness":39,"color":{"h":359,"hue":359,"s":100,"saturation":100,"x":0.6942,"y":0.2963},"color_mode":"xy","color_power_on_behavior":null,"color_temp":158,"do_not_disturb":null,"last_seen":"2023-06-08T10:24:26+02:00","linkquality":80,"state":"ON"}'
Info 2023-06-08 10:24:25MQTT publish: topic 'zigbee2mqtt/zbLedTrakaPc', payload '{"brightness":39,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.1355,"y":0.0399},"color_mode":"xy","color_power_on_behavior":null,"color_temp":574,"do_not_disturb":null,"last_seen":"2023-06-08T10:24:26+02:00","linkquality":80,"state":"ON"}'`

JabaMS, 2023-06-08:
[image]

lgretton, 2023-07-06:
For what it's worth I'm seeing exactly the same behaviour in 1.32.0-2 for a similar device, TS0505B_1, _TZ3210_s9lumfhn.

lgretton, 2023-08-07:
For me the possibly-related problem has gone since going to Z2M 1.32.2-1.

JabaMS, 2023-08-09:
> For me the possibly-related problem has gone since going to Z2M 1.32.2-1.

Seems to be true. Problems that starts happening since 1.31.0 version has gone, but previous ones still persists, like transition time, color effects etc.

closed 2023-09-17 by github-actions[bot]

alexsahka, 2023-10-30:
> ### What happened?
> With zigbee2mqtt version 1.30.4 this device was recognized as TuYa TS0505B_1 and more or less it worked normally.
> 
> Problems in version 1.30.4:
> 
> * Changing between color_temp and color_xy does not recognize current brightness, but instead it uses previously used
> * Color effects does not work
> * Transition time does not work
> * color_sync does not work
> 
> Since update 1.31.0 it's not possible to use controller thru zigbee2mqtt frontend.
> 
> Problems in version 1.31.0:
> 
> * Changing from color_temp to color_xy is possible, but not from color_xy to color_temp (it works as expected when doing this thru HA card).
> * Once changed from color_temp to color_xy it wan't turn off device  (it works as expected when doing this thru HA card).
> * If I manage to turn of device after couple of tries, on turn on it tries to turn on color_temp, than it switches to color_xy in a half of second
> * Reading current brightness not possible
> * On settings page, focus in and out of transition field triggers save setting action (no matter is there a change in input field)
> * All previously problems I wrote exists
> 
> ### What did you expect to happen?
> _No response_
> 
> ### How to reproduce it (minimal and precise)
> Make a change between color modes thru Z2M page. Try turn on/off after that.
> 
> ### Zigbee2MQTT version
> 1.31.0
> 
> ### Adapter firmware version
> 20220301
> 
> ### Adapter
> Egony Stick V4
> 
> ### Debug log
> `Debug 2023-06-08 10:24:07Publishing 'set' 'color' to 'zbLedTrakaPc' Info 2023-06-08 10:24:07MQTT publish: topic 'zigbee2mqtt/zbLedTrakaPc', payload '{"brightness":118,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.1355,"y":0.0399},"color_mode":"xy","color_power_on_behavior":null,"color_temp":574,"do_not_disturb":null,"last_seen":"2023-06-08T10:24:08+02:00","linkquality":76,"state":"ON"}' Debug 2023-06-08 10:24:07Publishing 'set' 'effect' to 'zbLedTrakaPc' Info 2023-06-08 10:24:07MQTT publish: topic 'zigbee2mqtt/zbLedTrakaPc', payload '{"brightness":118,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.1355,"y":0.0399},"color_mode":"xy","color_power_on_behavior":null,"color_temp":574,"do_not_disturb":null,"last_seen":"2023-06-08T10:24:08+02:00","linkquality":80,"state":"ON"}' Info 2023-06-08 10:24:07MQTT publish: topic 'zigbee2mqtt/zbLedTrakaPc', payload '{"brightness":118,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.1355,"y":0.0399},"color_mode":"xy","color_power_on_behavior":null,"color_temp":574,"do_not_disturb":null,"last_seen":"2023-06-08T10:24:08+02:00","linkquality":80,"state":"ON"}' Debug 2023-06-08 10:24:08Received MQTT message on 'zigbee2mqtt/zbLedTrakaPc/set' with data '{"brightness":118,"color":{"hex":"#0000F9"},"color_temp":454,"effect":"breathe","state":"ON"}' Debug 2023-06-08 10:24:08Publishing 'set' 'state' to 'zbLedTrakaPc' Info 2023-06-08 10:24:08MQTT publish: topic 'zigbee2mqtt/zbLedTrakaPc', payload '{"brightness":118,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.1355,"y":0.0399},"color_mode":"xy","color_power_on_behavior":null,"color_temp":574,"do_not_disturb":null,"last_seen":"2023-06-08T10:24:09+02:00","linkquality":83,"state":"ON"}' Debug 2023-06-08 10:24:08Publishing 'set' 'color' to 'zbLedTrakaPc' Info 2023-06-08 10:24:08MQTT publish: topic 'zigbee2mqtt/zbLedTrakaPc', payload '{"brightness":118,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.1355,"y":0.0399},"color_mode":"xy","color_power_on_behavior":null,"color_temp":574,"do_not_disturb":null,"last_seen":"2023-06-08T10:24:09+02:00","linkquality":83,"state":"ON"}' Debug 2023-06-08 10:24:08Publishing 'set' 'effect' to 'zbLedTrakaPc' Info 2023-06-08 10:24:08MQTT publish: topic 'zigbee2mqtt/zbLedTrakaPc', payload '{"brightness":118,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.1355,"y":0.0399},"color_mode":"xy","color_power_on_behavior":null,"color_temp":574,"do_not_disturb":null,"last_seen":"2023-06-08T10:24:09+02:00","linkquality":80,"state":"ON"}' Info 2023-06-08 10:24:08MQTT publish: topic 'zigbee2mqtt/zbLedTrakaPc', payload '{"brightness":118,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.1355,"y":0.0399},"color_mode":"xy","color_power_on_behavior":null,"color_temp":574,"do_not_disturb":null,"last_seen":"2023-06-08T10:24:09+02:00","linkquality":80,"state":"ON"}' Debug 2023-06-08 10:24:09Received MQTT message on 'zigbee2mqtt/zbLedTrakaPc/set' with data '{"brightness":118,"color":{"hex":"#0000F9"},"color_temp":370,"effect":"breathe","state":"ON"}' Debug 2023-06-08 10:24:09Publishing 'set' 'state' to 'zbLedTrakaPc' Info 2023-06-08 10:24:09MQTT publish: topic 'zigbee2mqtt/zbLedTrakaPc', payload '{"brightness":118,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.1355,"y":0.0399},"color_mode":"xy","color_power_on_behavior":null,"color_temp":574,"do_not_disturb":null,"last_seen":"2023-06-08T10:24:10+02:00","linkquality":76,"state":"ON"}' Debug 2023-06-08 10:24:09Publishing 'set' 'color' to 'zbLedTrakaPc' Info 2023-06-08 10:24:09MQTT publish: topic 'zigbee2mqtt/zbLedTrakaPc', payload '{"brightness":118,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.1355,"y":0.0399},"color_mode":"xy","color_power_on_behavior":null,"color_temp":574,"do_not_disturb":null,"last_seen":"2023-06-08T10:24:10+02:00","linkquality":76,"state":"ON"}' Debug 2023-06-08 10:24:09Publishing 'set' 'effect' to 'zbLedTrakaPc' Info 2023-06-08 10:24:09MQTT publish: topic 'zigbee2mqtt/zbLedTrakaPc', payload '{"brightness":118," [... cut ...] 2:00","linkquality":80,"state":"ON"}' Info 2023-06-08 10:24:25MQTT publish: topic 'zigbee2mqtt/zbLedTrakaPc', payload '{"brightness":39,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.1355,"y":0.0399},"color_mode":"xy","color_power_on_behavior":null,"color_temp":574,"do_not_disturb":null,"last_seen":"2023-06-08T10:24:26+02:00","linkquality":80,"state":"ON"}'`

My miboxer fut037z+ RGB+CCT led controller even not recognized as TuYa TS0505B_1
I'm on the Zigbee2MQTT version[1.33.1]
[image]

alexsahka, 2023-10-30:
Any recommendation?

alexsahka, 2023-11-02:
> ### What happened?
> With zigbee2mqtt version 1.30.4 this device was recognized as TuYa TS0505B_1 and more or less it worked normally.
>
> Problems in version 1.30.4:
>
> * Changing between color_temp and [...]

Can you recommend any converter for fut037z+?

Larsvb0, 2026-01-11:
I got transitions working using a script. From the automation I call the script:

sequence:
- repeat:
until:
- condition: device
type: is_off
device_id: eddf44ddc6bda7a3f4d9db8c46a31c9f
entity_id: 5cd77dff9f8c63a15dcd4f3b3e371bec
domain: light
sequence:
- action: light.turn_on
metadata: {}
target:
entity_id: light.traptrede_verlichting
data:
brightness_step_pct: -1
- delay:
hours: 0
minutes: 0
seconds: 0
milliseconds: 500
alias: Traptrede Brightness
description: ""
fields: {}