Koenkk/zigbee2mqtt #17958: Danfoss Icon2 Room Sensor (088U2121)

Luigi8723, 2023-06-08:
### Link

https://store.danfoss.com/en/Climate-Solutions-for-heating/Hydronic-floor-heating/Room-controls/Floor-Heating-Controls%2C-Danfoss-Icon2%2C-Room-Thermostat%2C-Attachable/p/088U2121

### Database entry

{"id":92,"type":"EndDevice","ieeeAddr":"0x94deb8fffe5fe9fa","nwkAddr":58358,"manufId":4678,"manufName":"Danfoss","powerSource":"Battery","modelId":"0x8040","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,1,3,32,513,516,1026,1029,2821],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"0x8040","manufacturerName":"Danfoss","powerSource":3,"zclVersion":8,"appVersion":0,"stackVersion":9,"hwVersion":1,"dateCode":"YYYYMMDD","swBuildId":"01.00\u0000"}},"genPollCtrl":{"attributes":{"checkinInterval":14400}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b00219fb611","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":9,"hwVersion":1,"dateCode":"YYYYMMDD","swBuildId":"01.00\u0000","zclVersion":8,"interviewCompleted":true,"meta":{},"lastSeen":1686230378372,"defaultSendRequestWhen":"active","checkinInterval":3600}

### Comments

I have added the sensor to my the network 

and it reports as unsupported. 

Warning 2023-06-08 15:23:31Received message from unsupported device with Zigbee model '0x8040' and manufacturer name 'Danfoss'
Warning 2023-06-08 15:23:31Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

Luigi8723, 2023-06-15:
@Koenkk if needed i could also Supply you with a model. Free of charge :)
I am working at Danfoss NL and they love to get this one supported, so am i :)

Koenkk, 2023-06-17:
Thanks for the offer but I currently don't have enough time to work on integration of new devices myself.

If this device follows the Zigbee spec it should be relatively easy, I've created an initial external converter:

https://gist.github.com/Koenkk/0796a9e0bbb441c1347bed7999a938e6

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

Luigi8723, 2023-06-22:
Hi,

Sorry for the late response,
i have checked, but getting an error with the startup of zigbee2mqtt

/app/data/extension/externally-loaded.js:29
        e.climate()
          ^
TypeError: e.climate is not a function
    at /app/data/extension/externally-loaded.js:29:11
    at Script.runInContext (node:vm:141:12)
    at Script.runInNewContext (node:vm:146:17)
    at Object.runInNewContext (node:vm:306:38)
    at loadModuleFromText (/app/lib/util/utils.ts:150:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:157:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:167:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)

i am not sure how to fix this

Koenkk, 2023-06-22:
I'm not getting that error, are you using the latest z2m version?

Luigi8723, 2023-06-23:
I was a bit behind, but now running the latest version.

Still getting the error.

I am running Zigbee2Mqtt in Home Assistant OS on a Rasp 4B. 
And i see the are link to files behind certain paths 
I cannot trace the paths due to limted SSH posibilities in HAos.
maby the paths are diffrent in my settup? I make use of the addon Zigbee2MQTT Current version: 1.31.2-1

[11:31:59] INFO: Starting Zigbee2MQTT...
/app/data/extension/externally-loaded.js:29
        e.climate()
          ^
TypeError: e.climate is not a function
    at /app/data/extension/externally-loaded.js:29:11
    at Script.runInContext (node:vm:141:12)
    at Script.runInNewContext (node:vm:146:17)
    at Object.runInNewContext (node:vm:306:38)
    at loadModuleFromText (/app/lib/util/utils.ts:150:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:157:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:167:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)

Koenkk, 2023-06-23:
Can you confirm the following 2 lines are present in your ext converter?

```js
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
```

Luigi8723, 2023-06-23:
Yep, confirm

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['0x8040'],
    model: '088U2121',
    vendor: 'Danfoss',
    description: 'Room sensor',
    fromZigbee: [fz.thermostat, fz.hvac_user_interface],
    toZigbee: [
        tz.thermostat_local_temperature,
        tz.thermostat_system_mode,
        tz.thermostat_occupied_heating_setpoint,
        tz.thermostat_unoccupied_heating_setpoint,
        tz.thermostat_occupied_cooling_setpoint,
        tz.thermostat_running_state,
    ],
    exposes: [
        e.climate()
            .withSetpoint('occupied_heating_setpoint', 7, 28, 0.5)
            .withSetpoint('unoccupied_heating_setpoint', 7, 28, 0.5)
            .withLocalTemperature()
            .withSystemMode(['off', 'heat', 'auto'])
            .withRunningState(['idle', 'heat']),
    ],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'hvacThermostat']);
        await reporting.thermostatTemperature(endpoint);
        await reporting.thermostatRunningState(endpoint);
        await reporting.thermostatOccupiedHeatingSetpoint(endpoint);
        await reporting.thermostatUnoccupiedHeatingSetpoint(endpoint);
    },
};

module.exports = definition;`

Koenkk, 2023-06-24:
I have no clue why it isn't working, can you setup z2m outside HA and see if that works? (https://www.zigbee2mqtt.io/guide/installation/01_linux.html)

Luigi8723, 2023-06-27:
Oke, than it has to be somewhere in my setup.
Will try to get is running in HA first.

I have somewhere a RPI with an install of domoticz and Zigbee2mqtt on that, I  will try it out.

Thanks :)

bjornjonsson, 2023-07-07:
@Luigi8723 if you need some help I could try to solve it for you. I'm running four Danfoss Ally radiator thermostats in my house upstairs and I'm considering replacing my downstairs Roth floor thermostats with Danfoss Icon, so I have some interest in getting this device working in zigbee2mqtt as well.

Luigi8723, 2023-07-12:
@bjornjonsson sorry for my late reply
It should already be working with the ext_converter.js from Koen.
But I have a somewhere in my setup a flaw, not found it yet.
You could test it with the converter from Koen, if its working on your side.

JohnConnett, 2023-08-09:
I have just obtained an **Icon2 RT** (088U2121).
Is there some special magic to put it into pairing mode? I have nothing in the Zigbee2MQTT log that looks like it is from it and the device displays "Err NEt".

Luigi8723, 2023-08-09:
Out the top of my head, it goes in pairing mode after the batteries are in.
So set your zigbee network in pairing mode, and then put the battery in.
If that doesnt work, if have to check again how i did it

JohnConnett, 2023-08-09:
> Out the top of my head, it goes in pairing mode after the batteries are in. So set your zigbee network in pairing mode, and then put the battery in. If that doesnt work, if have to check again how i did it

That's what I tried. Couldn't find a separate reset switch. Was wondering if **step 2** of **Mounting and inclusion** did something clever with the QR code or Bluetooth. I have a **SkyConnect** which I can try it with to see if that is more successful.

JohnConnett, 2023-08-10:
> That's what I tried. Couldn't find a separate reset switch. Was wondering if **step 2** of **Mounting and inclusion** did something clever with the QR code or Bluetooth. I have a **SkyConnect** [...]

Removed both batteries over night. Reinserted them. Couldn't connect **Icon2 RT** to **SkyConnect** (**Philips Hue wall switch module** connected with no problems).

Dug out my old **Ally Gateway** (014G2400). Connected with **Ally app** and checked that **Ally Gateway** firmware is up-to-date. Clicked **+ Add subdevice**. No obvious picture/name combination for **Icon2 RT**. Tried **Icon2** but it expects a fully configured **Icon2** system. Tried **Ally Room Sensor**. Reinserted batteries. App says _Search timed out_. Briefly touched top left of device face. Displays temperature, RH, then "Err NEt". Packaging clearly displays **Danfoss Ally Compatible** and **Frequency: 2,4GHz (Zigbee)**. Strange ...

JohnConnett, 2023-08-10:
Re-pairing instructions: [Application guide Danfoss Icon2™](https://assets.danfoss.com/documents/282053/AB432956914381en-010202.pdf, page 82.

1. Activate the thermostat by pressing on the screen
2. Use both thumbs to press the temperature adjustment arrow keys **^v** until **deL ALL** appears, as shown, as shown in the adjacent image
3. Press the **checkmark** next to **deL ALL**

The thermostat has now been removed from the system.

JohnConnett, 2023-08-10:
> Re-pairing instructions: [Application guide Danfoss Icon2™](https://assets.danfoss.com/documents/282053/AB432956914381en-010202.pdf, page 82.
>
>     1. Activate the thermostat by pressing on the [...]

Works successfully to pair with **Zigbee2MQTT** (and **SkyConnect**). For **Zigbee2MQTT** I had to touch the screen which then displayed a _worm running round a rectangle_ then paired.

Luigi8723, 2023-08-14:
@JohnConnett nice!
Are you running Zigbee2MQTT on Home Assistant? and is your RS icon2 now fully supported in HA?

JohnConnett, 2023-08-14:
> @JohnConnett nice!
> Are you running Zigbee2MQTT on Home Assistant?

Yes. On a Raspberry Pi 4 Model B.

> and is your RS icon2 now fully supported in HA?

Not quite. Here's what is in the Zigbee2MQTT log:
```
Zigbee2MQTT:debug 2023-08-14 22:44:28: Received Zigbee message from '0x8cf681fffe824a00', type 'attributeReport', cluster 'hvacThermostat', data '{"localTemp":2951}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-08-14 22:44:28: Configuring '0x8cf681fffe824a00'
Zigbee2MQTT:info  2023-08-14 22:44:28: MQTT publish: topic 'zigbee2mqtt/0x8cf681fffe824a00', payload '{"linkquality":167,"local_temperature":29.51,"running_state":null}'
Zigbee2MQTT:error 2023-08-14 22:44:29: Failed to configure '0x8cf681fffe824a00', attempt 2 (Error: ConfigureReporting 0x8cf681fffe824a00/1 hvacThermostat([{"attribute":"runningState","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:419:28)
    at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:760:22)
    at Object.thermostatRunningState (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:176:5)
    at Object.configure (/app/data/extension/externally-loaded.js:40:9)
    at Configure.configure (/app/lib/extension/configure.ts:117:13))
```

JohnConnett, 2023-09-15:
I'm still keen for the Icon2 room Sensor family of devices to be supported.
How can this issue be progressed?
Can the Compliance Documents available from https://csa-iot.org/csa_product/danfoss-icon2-room-thermostat-sensor/ be used to craft a better "ext_converter.js"?

closed 2023-10-24 by github-actions[bot]