Koenkk/zigbee2mqtt #17961: Exception while calling fromZigbee converter: Expected one of: 1, 4, got:

smarteshome2020, 2023-06-08:
### What happened?

After update to 1.31.2-dev (frontend 0.6.129)  i recieved some errors and i don`t know what happen.

`<html>
<body>

zigbee2mqtt.0 | 2023-06-08 19:33:36.341 | error | Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '17320'}
-- | -- | -- | --
zigbee2mqtt.0 | 2023-06-08 19:31:47.599 | error | Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '17320'}
zigbee2mqtt.0 | 2023-06-08 19:31:40.282 | error | Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '17320'}
zigbee2mqtt.0 | 2023-06-08 19:22:29.337 | error | Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '5032'}
zigbee2mqtt.0 | 2023-06-08 19:20:08.249 | error | Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '5032'}
zigbee2mqtt.0 | 2023-06-08 19:16:13.676 | error | Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '17320'}
zigbee2mqtt.0 | 2023-06-08 19:13:29.595 | error | Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '5032'}
zigbee2mqtt.0 | 2023-06-08 19:11:31.749 | error | Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '13224'}
zigbee2mqtt.0 | 2023-06-08 19:11:11.481 | error | Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '13224'}
zigbee2mqtt.0 | 2023-06-08 19:09:12.503 | error | Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '17320'}
zigbee2mqtt.0 | 2023-06-08 18:59:19.921 | error | Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '5032'}
zigbee2mqtt.0 | 2023-06-08 18:52:33.596 | error | Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '5032'}
zigbee2mqtt.0 | 2023-06-08 18:48:19.937 | error | Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '17320'}
zigbee2mqtt.0 | 2023-06-08 18:48:14.838 | error | Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '5032'}
zigbee2mqtt.0 | 2023-06-08 18:43:39.109 | error | Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '17320'}
zigbee2mqtt.0 | 2023-06-08 18:41:24.262 | error | Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '17320'}
zigbee2mqtt.0 | 2023-06-08 18:37:23.120 | error | Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '17320'}
zigbee2mqtt.0 | 2023-06-08 18:36:21.432 | error | Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '17320'}
zigbee2mqtt.0 | 2023-06-08 18:32:06.339 | error | Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '5032'}
zigbee2mqtt.0 | 2023-06-08 18:25:53.981 | error | Exception while calling fromZigbee converter: Expected one of: 1, 4, got: '17320'}

</body>
</html>`

### Zigbee2MQTT version

1.31.2

### Adapter firmware version

20221226

### Adapter

Sonoff 3.0

Koenkk, 2023-06-08:
Please check if this is already fixed in the latest-dev, if not provide the debug log.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

skkost, 2023-07-01:
1.32.0-2

Zigbee2MQTT:debug 2023-07-02 01:46:38: Received Zigbee message from 'Реле Aqara', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[100,16,0,3,40,16,5,33,12,0,154,32,0,8,33,24,1,11,32,0,10,33,218,113,12,32,1],"type":"Buffer"}}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2023-07-02 01:46:38: SSM-U02: Processed buffer into data {"3":16,"5":12,"8":280,"10":29146,"11":0,"12":1,"100":0,"154":0}
Zigbee2MQTT:error 2023-07-02 01:46:38: Exception while calling fromZigbee converter: Expected one of: 1, 2, got: '29146'}
Zigbee2MQTT:debug 2023-07-02 01:46:38: Error: Expected one of: 1, 2, got: '29146'
    at getFromLookup (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:512:37)
    at numericAttributes2Payload (/app/node_modules/zigbee-herdsman-converters/src/lib/xiaomi.ts:213:52)
    at Object.numericAttributes2Payload (/app/node_modules/zigbee-herdsman-converters/src/lib/xiaomi.ts:478:42)
    at Object.convert (/app/node_modules/zigbee-herdsman-converters/src/converters/fromZigbee.js:3967:33)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:150:51)
    at EventEmitter.emit (node:events:525:35)
    at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22)
    at Controller.<anonymous> (/app/lib/zigbee.ts:106:27)
    at Controller.emit (node:events:513:28)
    at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)
Zigbee2MQTT:info  2023-07-02 01:46:38: MQTT publish: topic 'zigbee2mqtt/Реле Aqara', payload '{"device_temperature":19,"last_seen":"2023-07-02T01:46:38+03:00","linkquality":65,"power_outage_count":11,"power_outage_memory":false,"state":"OFF","switch_type":null}'

Koenkk, 2023-07-02:
@skkost fixed that one

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-08-09 by github-actions[bot]

FvanEgmond, 2024-01-24:
FYI: Got this message again, using version 1.35.1.