Koenkk/zigbee2mqtt #17962: [New device support]: Bingoelec ZigBee Thermostat

dmitriy-sqrt, 2023-06-08:
### Link

https://www.aliexpress.com/item/1005005362303229.html?

### Database entry

{"id":26,"type":"Router","ieeeAddr":"0xa4c138880f149305","nwkAddr":51806,"manufId":4417,"manufName":"_TZE200_tu0bri7p","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65506":54,"65508":1,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_tu0bri7p","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1686249994193,"defaultSendRequestWhen":"immediate"}

### Comments

First i thought Bseed converter should work, as Bingoelec and Bseed devices are mostly similar, but Bseed converter didnt work at all.

So I spent some time with Tuya Dev portal, could make most things work. Things not working: schedule, only selected temperature probe works (in case you want both room+NTC sensor - only one is reported via `local_temperature`)

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const bingoelec_thermostat_floor_sensor_calibration = {
    key: ['floor_sensor_calibration'],
    convertSet: async (entity, key, value, meta) => {
        // negative values are set incorrectly
        await tuya.sendDataPointValue(entity, 101, temp);
    },
};

const bingoelec_thermostat_state = {
    key: ['state'],
    convertSet: async (entity, key, value, meta) => {
        await tuya.sendDataPointBool(entity, 1,  value === 'ON');
    },
};

const bingoelec_thermostat_away_mode = {
    key: ['away_mode'],
    convertSet: async (entity, key, value, meta) => {
        await tuya.sendDataPointBool(entity, 103,  value === 'ON');
    },
};

const bingoelec_thermostat_child_lock = {
    key: ['child_lock'],
    convertSet: async (entity, key, value, meta) => {
        await tuya.sendDataPointBool(entity, 9,  value === 'LOCK');
    },
};

const bingoelec_thermostat = {
    cluster: 'manuSpecificTuya',
    type: ['commandDataResponse', 'commandDataReport'],
    convert: (model, msg, publish, options, meta) => {
        const dpValue = tuya.firstDpValue(msg, meta, 'bingoelec_thermostat');
        const dp = dpValue.dp;
        const value = tuya.getDataValue(dpValue);

        // meta.logger.warn(`debug ${dp} - ${value}`)

        switch (dp) {
            case 1:
                return {state: value ? 'ON' : 'OFF'};
            case 9:
                return {child_lock: value ? 'LOCK' : 'UNLOCK'};
            case 101:
                return {floor_sensor_calibration: value};
            case 102:
                return {running_state: value ? 'idle' : 'heat'};
            case 103:
                return {away_mode: value ? 'ON' : 'OFF'};
            default:
                return {};
        }
    }
}

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE200_tu0bri7p',
        },
    ],
    model: 'TS0601_bingoelec_thermostat',
    vendor: 'Bingoelec',
    description: 'Bingoelec Room Thermostat Electric Heating',
    fromZigbee: [
        tuya.fz.datapoints,
        bingoelec_thermostat,
    ],
    toZigbee: [
        bingoelec_thermostat_floor_sensor_calibration,
        bingoelec_thermostat_away_mode, bingoelec_thermostat_child_lock,
        bingoelec_thermostat_state,
        tuya.tz.datapoints,
    ],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        e.switch(),

        exposes.climate()
                .withSetpoint('current_heating_setpoint', 5, 50, 1, ea.STATE_SET)
                .withLocalTemperature(ea.STATE)
                .withLocalTemperatureCalibration(-9, 9, 1, ea.STATE_SET) // Should it use room or floor sensor as a source?
                .withRunningState(['idle', 'heat'], ea.STATE),

        e.away_mode(),
        e.min_temperature().withValueMin(5).withValueMax(35),
        e.max_temperature().withValueMin(20).withValueMax(50), // Device supports: +35 - +50°C from internal menu, but accepts other values from datapoint

        exposes.numeric('floor_sensor_calibration', ea.STATE_SET)
                .withUnit('°C')
                .withValueMin(-9)
                .withValueMax(9)
                .withValueStep(1)
                .withDescription('Temperature calibration for the exernal floor sensor, between -9 and 9 in 1°C.  Default: 0'),

        e.child_lock(),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverter.raw],
            [2, 'current_heating_setpoint', tuya.valueConverter.raw],
            [3, 'local_temperature', tuya.valueConverter.divideBy10],
            [9, 'child_lock', tuya.valueConverter.raw],
            [15, 'max_temperature', tuya.valueConverter.raw],
            [16, 'min_temperature', tuya.valueConverter.raw],
            [19, 'local_temperature_calibration', tuya.valueConverter.raw],
            [101, 'floor_sensor_calibration', tuya.valueConverter.raw],
            [102, 'running_state', tuya.valueConverter.trueFalse0], // Is thermostat heating now or is idle
            [103, 'away_mode', tuya.valueConverter.raw],

            // deadzone_temperature - looks like its not exposed to zigbee
            // TODO: DPID - 11, fault
        ],
    },
};

module.exports = definition;
```

Koenkk, 2023-06-08:
I see you are still using some converters besides `tuya.tz.datapoints`, can you merge these (e.g. `bingoelec_thermostat_state`) into `tuyaDatapoints`?

dmitriy-sqrt, 2023-06-21:
> I see you are still using some converters besides `tuya.tz.datapoints`, can you merge these (e.g. `bingoelec_thermostat_state`) into `tuyaDatapoints`?

Sorry could you explain a bit?

Koenkk, 2023-06-22:
You can find an example of a thermostat which is integrated with just `tuyaDataPoints` here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/30e6fcd2eec8098282056615bb7db14f4b352787/src/devices/tuya.ts#L2524 (so no other converters in toZigbee/fromZigbee besides `tuya.tz.datapoints` and `tuya.fz.datapoints`)

PioKow2023, 2023-06-23:
Hi, great that you addressed this thermostat. The value for money is really great, just... it doesn't work with HA yet. :)
I lack the knowledge to add this device myself, so thank you very much for your work.
So far I have not managed to connect with HA at all using these instructions https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-adding-your-device
I'm curious how did you manage to do it? :)
Greetings!

PioKow2023, 2023-06-27:
One error - current_heating_setpoing should also be divided by 10:
meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverter.raw],
            [2, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
            [3, 'local_temperature', tuya.valueConverter.divideBy10],

PioKow2023, 2023-08-14:
Hi, did you by chance managed to finish this converter? I don't have enough knowledge to make it myself, unfortunately, and still my 6 Bingoelec thermostats do not work properly... :( Thanks for any support you can provide!

closed 2023-09-22 by github-actions[bot]

domiksmi, 2023-10-28:
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition =
{
    fingerprint: [{
            modelID: 'TS0601',
            manufacturerName: '_TZE200_tu0bri7p',
        },
    ],
    model: 'bingoelec_thermostat',
    vendor: 'Bingoelec',
    description: 'TS0601 series thermostat',

    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEvent({timeStart: '2000'}),
    configure: tuya.configureMagicPacket,
    exposes: [
        e.binary('state', ea.STATE_SET, 'ON', 'OFF').withDescription('Turn the thermostat ON/OFF'),
        e.child_lock(),
        e.climate()
            .withLocalTemperature(ea.STATE)
            .withSetpoint('current_heating_setpoint', 5, 35, 1, ea.STATE_SET)
            .withRunningState(['idle', 'heat'], ea.STATE)
            .withLocalTemperatureCalibration(-5, 5, 0.5, ea.STATE_SET),
        e.min_temperature().withValueMin(5).withValueMax(15),
        e.max_temperature().withValueMin(15).withValueMax(45),
        e.numeric('floor_sensor_calibration', ea.STATE_SET).withUnit('°C').withValueMin(-5).withValueMax(5).withValueStep(1).withDescription('Floor sensor calibration, between -5 and 5 in 1°C. Default: 0'),
    ],
    meta: {
        tuyaDatapoints:
        [
            [1, 'state', tuya.valueConverter.onOff],
            [2, 'current_heating_setpoint', tuya.valueConverter.raw],
            [3, 'local_temperature', tuya.valueConverter.divideBy10],
            [9, 'child_lock', tuya.valueConverter.lockUnlock],
            [15, 'max_temperature', tuya.valueConverter.raw],
            [16, 'min_temperature', tuya.valueConverter.raw],
            [19, 'local_temperature_calibration', tuya.valueConverter.localTemperatureCalibration],
            [101, 'floor_sensor_calibration', tuya.valueConverter.raw],
            [102, 'running_state', tuya.valueConverter.trueFalse0],
        ],
    },
};

module.exports = definition;

domiksmi, 2023-10-28:
Try this one)

PioKow2023, 2023-11-02:
Thank you very much. Your converter works and its code is much "nicer" than the one I made by trial and error.
The only thing that still doesn't work is the time zone update - the time on the controllers is 6 hours ahead.
Do you know perhaps how to fix this?

dadock, 2023-11-29:
Great, thanks so much, works like a charme. I have the same device, the only difference ist "manufacturerName: '_TZE200_lndsb16m'".
One question however: As far as I understand, the device has two temperature sensors: One inside (Local temperature) and another one external, which comes with a long white cable that needs to be connected to the ports of the device. This sensor is configurable via zigbee2mqtt ("Floor sensor calibration"), but it seems that there is no extra temperature reading for it.

dadock, 2026-02-28:
The device is supported now thanks to custom firmware: https://github.com/slacky1965/tuya_thermostat_zrd - see also https://www.zigbee2mqtt.io/devices/THERM_SLACKY_DIY_R08.html - I managed to update the firmware without problems.