Koenkk/zigbee2mqtt #17963: E1743 super powered at 200% battery level

Gonioul, 2023-06-08:
### What happened?

200% level battery since recent update

### Zigbee2MQTT version

1.31.2

### Adapter firmware version

0x26780700

### Adapter

ConBee2/RaspBee2

Gonioul, 2023-06-08:
[image]

Gonioul, 2023-06-08:
Note that the firmware version is also wrong. It was updated recently.

sjorge, 2023-06-08:
The battery info is wrong because for ikea we decide if we should divide or not based on the fw version.

You can try rereading it via the dev console in the frontend. It's on genBasic swBuildId (i think)

closed 2023-07-17 by github-actions[bot]