Koenkk/zigbee2mqtt #17965: [New device support]: OFL 122 C- light, power_on_behavior

xav31190, 2023-06-08:
### Link

https://www.innr.com/en/product/outdoor-flex-light-colour-2021/

### Database entry

"id":40,"type":"Router","ieeeAddr":"0x00158d00027d5d2f","nwkAddr":30195,"manufId":4454,"manufName":"innr","powerSource":"Mains (single phase)","modelId":"OFL 122 C","epList":[1,2],"endpoints":{"1":{"profId":49246,"epId":1,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[25],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":49246,"epId":2,"devId":4096,"inClusterList":[4096],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":41,"stackVersion":33,"hwVersion":2,"dateCode":"20210908","swBuildId":"2.9.9","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1686256495619,"defaultSendRequestWhen":"immediate"}

### Comments

Hi,
similar to "OFL 142 C- light, power_on_behavior" which is already supported

### Supported color modes

1800K - 6500K RGBW

Koenkk, 2023-06-08:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

xav31190, 2023-06-10:
Hi,
Thank you very much for your reactivity, I will test