Koenkk/zigbee2mqtt #17969: [New device support]: Mercator Ikuu SSW05G-ZB

kiwizznz, 2023-06-08:
### Link

https://www.ikuu.com.au/product/five-switch-2/

### Database entry

{"id":51,"type":"Router","ieeeAddr":"0xe0798dfffed35758","nwkAddr":4692,"manufId":4098,"manufName":"_TZE200_leaqthqq","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"l\u001c\u0015,\u0013l\u001c\u0015,el\u001c\u0015,\u0012m\u001c\u0015,\u0012","65506":31,"65508":0,"appVersion":65,"modelId":"TS0601","manufacturerName":"_TZE200_leaqthqq","powerSource":1,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1686266007612,"defaultSendRequestWhen":"immediate"}

### Comments

5 Gang Light Switch (Neutral required)

Zigbee Model
TS0601
Zigbee Manufacturer
_TZE200_leaqthqq

Koenkk, 2023-06-09:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/50423d7fd0fe249e5b6e72554b3adff6

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

kiwizznz, 2023-06-09:
@Koenkk yep - that worked. Cheers!

closed 2023-06-09 by kiwizznz

Koenkk, 2023-06-10:
Integrated the changes!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)