Koenkk/zigbee2mqtt #17970: GLEDOPTO GL-B-007P add PowerOnBehavior

deaxx, 2023-06-09:
### What happened?

There is no PowerOnBehavior in the converter

{
        zigbeeModel: ['GL-B-007P'],
        model: 'GL-B-007P',
        vendor: 'Gledopto',
        ota: ota.zigbeeOTA,
        description: 'Zigbee 6W E26/E27 Bulb RGB+CCT (pro)',
        extend: gledoptoExtend.light_onoff_brightness_colortemp_color({**_disablePowerOnBehavior: false_** }),
    },

### Zigbee2MQTT version

1.31.2

### Adapter firmware version

20230507

### Adapter

СС2652P

Koenkk, 2023-06-09:
Does it work after this change?

deaxx, 2023-06-09:
> Does it work after this change?

works great

Koenkk, 2023-06-10:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-06-10 by Koenkk