Koenkk/zigbee2mqtt #17971: QS-Zigbee-S04-2C-LN - Not working after update

kizio1000, 2023-06-09:
### What happened?

After update to 1.31.2-1 two pole switch QS-Zigbee-S04-2C-LN discovered as TS0003 stopped executing commands from HA. It shows current state set manually but does not execute state changes done from HA/Z2M.

Log after attempt to change state in Z2M (light did not went on) :

Debug 2023-06-09 07:32:58Received MQTT message on 'zigbee2mqtt/Stol i kominek/set' with data '{"state_l1":"ON"}'
Debug 2023-06-09 07:32:58Publishing 'set' 'state' to 'Stol i kominek'

Log after manual switching state on the switch:

bug 2023-06-09 07:34:53Received Zigbee message from 'Stol i kominek', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 2 with groupID 0
Info 2023-06-09 07:34:53MQTT publish: topic 'zigbee2mqtt/Stol i kominek', payload '{"linkquality":96,"state_l1":"OFF","state_l2":"ON"}'
Debug 2023-06-09 07:34:53Received Zigbee message from 'Stol i kominek', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 2 with groupID 0
Info 2023-06-09 07:34:53MQTT publish: topic 'zigbee2mqtt/Stol i kominek', payload '{"linkquality":96,"state_l1":"OFF","state_l2":"ON"}'
Debug 2023-06-09 07:34:53Received Zigbee message from 'Stol i kominek', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 2 with groupID 0
Info 2023-06-09 07:34:53MQTT publish: topic 'zigbee2mqtt/Stol i kominek', payload '{"linkquality":100,"state_l1":"OFF","state_l2":"ON"}'
Debug 2023-06-09 07:34:54Received Zigbee message from 'Stol i kominek', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 2 with groupID 0
Info 2023-06-09 07:34:54MQTT publish: topic 'zigbee2mqtt/Stol i kominek', payload '{"linkquality":88,"state_l1":"OFF","state_l2":"ON"}'
Debug 2023-06-09 07:34:54Received Zigbee message from 'Stol i kominek', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 2 with groupID 0
Info 2023-06-09 07:34:54MQTT publish: topic 'zigbee2mqtt/Stol i kominek', payload '{"linkquality":96,"state_l1":"OFF","state_l2":"ON"}'

### What did you expect to happen?

Switch will change states as always before

### Zigbee2MQTT version

1.31.2-1

### Adapter firmware version

7.1.1.0 build 273

### Adapter

ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

Koenkk, 2023-06-09:
Can you provide the data/database.db entry of this device?

kizio1000, 2023-06-09:
You mean /config/zigbee2mqtt/database.db ?

This is what I have for this device in database.db:

{"id":2,"type":"Router","ieeeAddr":"0x804b50fffee7a33f","nwkAddr":6580,"manufId":4098,"manufName":"_TYZB01_zsl6z0pw","powerSource":"Mains (single phase)","modelId":"TS0003","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"inClusterList":[0,4,5,6,6],"outClusterList":[25,10],"clusters":{"genOnOff":{"attributes":{"onOff":0,"offWaitTime":0,"onTime":0}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"inClusterList":[4,5,6,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"offWaitTime":0,"onTime":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":false,"meta":{},"lastSeen":1686291539138,"defaultSendRequestWhen":"immediate"}

Koenkk, 2023-06-10:
I cannot find the cause yet.

> does not execute state changes done from HA/Z2M.

Can you provide the debug log of this on both z2m 1.31.2 and 1.31.0?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

kizio1000, 2023-06-10:
Hi Koenkk,

I had already logging level on debug when posting logs in my first post, this is all that apers in log while trying to operate switch (both manually and from z2m/ha. So logs from 1.31.2 are in my initial post. Below logs (filtered by device) showing what's going on while operating switch when version 1.31.0-1 is installed.

Operated from HA/Z2M:

Fom OFF to ON:
`Debug 2023-06-10 09:37:15Received MQTT message on 'zigbee2mqtt/Stol i kominek/l2/set' with data 'ON'
Debug 2023-06-10 09:37:15Publishing 'set' 'state' to 'Stol i kominek'
Info 2023-06-10 09:37:15MQTT publish: topic 'zigbee2mqtt/Stol i kominek', payload '{"linkquality":104,"state_l1":"OFF","state_l2":"ON"}'
Debug 2023-06-10 09:37:16Received Zigbee message from 'Stol i kominek', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 2 with groupID 0
Info 2023-06-10 09:37:16MQTT publish: topic 'zigbee2mqtt/Stol i kominek', payload '{"linkquality":100,"state_l1":"OFF","state_l2":"ON"}'`

Fom ON to OFF:

`ebug 2023-06-10 09:39:30Received MQTT message on 'zigbee2mqtt/Stol i kominek/l2/set' with data 'OFF'
Debug 2023-06-10 09:39:30Publishing 'set' 'state' to 'Stol i kominek'
Info 2023-06-10 09:39:30MQTT publish: topic 'zigbee2mqtt/Stol i kominek', payload '{"linkquality":100,"state_l1":"OFF","state_l2":"OFF"}'
Debug 2023-06-10 09:39:30Received Zigbee message from 'Stol i kominek', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 2 with groupID 0
Info 2023-06-10 09:39:30MQTT publish: topic 'zigbee2mqtt/Stol i kominek', payload '{"linkquality":104,"state_l1":"OFF","state_l2":"OFF"}'`

Operated from wall switch:

From OFF to ON:

`Debug 2023-06-10 09:40:16Received Zigbee message from 'Stol i kominek', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 2 with groupID 0
Info 2023-06-10 09:40:16MQTT publish: topic 'zigbee2mqtt/Stol i kominek', payload '{"linkquality":104,"state_l1":"OFF","state_l2":"ON"}'`

From ON to OFF:

`Debug 2023-06-10 09:41:01Received Zigbee message from 'Stol i kominek', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 2 with groupID 0
Info 2023-06-10 09:41:01MQTT publish: topic 'zigbee2mqtt/Stol i kominek', payload '{"linkquality":108,"state_l1":"OFF","state_l2":"OFF"}'`

Koenkk, 2023-06-12:
Can you provide the herdsman debug log of this case on 1.31.0-1 vs 1.31.2-1

> Log after attempt to change state in Z2M (light did not went on)
   Debug 2023-06-09 07:32:58Received MQTT message on 'zigbee2mqtt/Stol i kominek/set' with data '{"state_l1":"ON"}'
   Debug 2023-06-09 07:32:58Publishing 'set' 'state' to 'Stol i kominek'

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

kizio1000, 2023-06-19:
Sorry for long reply time. Can you help me a bit with "start Zigbee2MQTT with: DEBUG=zigbee-herdsman* npm start" How do I do it on Z2M as HA addon ? Add it to addon configuration.yaml or add to configuration tab of addon ?

kizio1000, 2023-06-20:
I do not know what happened but after another update that I did today to latest 1.31.2-1 it started to work as expected.

Koenkk, 2023-06-20:
Great!

closed 2023-06-20 by Koenkk