Koenkk/zigbee2mqtt #17977: [Feature request]: ts0210 is too sensitive

JokeRsevilla, 2023-06-10:
### Is your feature request related to a problem? Please describe

ts0210 is too sensitive

### Describe the solution you'd like

be able to make a numeric adjustment in the menu.

### Describe alternatives you've considered

I have been able to do it by modifying the value in the development console but it is uncomfortable.

Cluster = sslasZone
Attribute = currentZoneSensitivityLevel
write number 20

### Additional context

current values are:
High = 0
Medium = 2
Low = 6

Koenkk, 2023-06-12:
Do I understand correctly that only value 0, 2 and 6 are allowed?

JokeRsevilla, 2023-06-12:
Yes, in format High , Medium and Low, but is too sensitive.
I have it placed on a door in case they try to break it.
I have set it to 20.
the maximum value is 50 according numZoneSensitivityLevelsSupported attribute.

P.D. sorry for my english

Koenkk, 2023-06-12:
So all values from 0 to 50 are allowed?

JokeRsevilla, 2023-06-12:
> So all values from 0 to 50 are allowed?

yes

JokeRsevilla, 2023-06-12:
[image]

here

Koenkk, 2023-06-13:
Thanks, implemented!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-06-13 by Koenkk