Koenkk/zigbee2mqtt #17978: Support for Innr RB 255 C

B08Z, 2023-06-10:
### Link

https://www.innr.com/en/product/innr-mini-bulb-white-colour-e14-uk/

### Database entry

{"id":9,"type":"Router", "ieeeAddr": "0xc49886000097821 5", "nwkAddr":44610, "manufId":4454, "manufName": "innr", "powerSource": "Mains (single phase)","modelId":"RB 25 5 C","epList":[1,242],"endpoints":{"1":["profId":260, "epId":1,"devId":269, "inClusterList":[0.3.4,5,6,8,768 ,4096,64599,64642], "outClusterList":[25], "clusters": [ "genBasic": ["attributes":{"modelId": "RB 255 C", "manuf acturerName": "innr", "powerSource" : 1, "zclVersion": 8, "a ppVersion":36, stackVersion":21,"hwVersion": 1, "dateCo de":"20230208", "swBuildId":"2.04.01"]}], "binds":[],"c onfiguredReportings": [], "meta":[]},"242":{"profId":41 440,"epId":242, "devId":97, "inClusterList":[],"outClus terList":[33], "clusters":{},"binds": [], "configured Rep ortings": [], "meta":[]}},"appVersion":36, "stackVersion ":21,"hwVersion" : 1, "dateCode": "20230208","swBuildId": 2.04.01", "zclVersion": 8, "interviewCompleted":true,"m eta":{},"lastSeen" :1686346485196, "defaultSendRequestW hen": "immediate"}

### Comments

I could not work out why this was not supported. Seems like there are several in the same family that are.

It pairs just not suppted:

[Zigbee2MQTT](http://192.168.2.157:9442/#/)

0xc498860000978215
[About](http://192.168.2.157:9442/#/device/0xc498860000978215/info)
[Exposes](http://192.168.2.157:9442/#/device/0xc498860000978215/exposes)
[Bind](http://192.168.2.157:9442/#/device/0xc498860000978215/bind)
[Reporting](http://192.168.2.157:9442/#/device/0xc498860000978215/reporting)
[Settings](http://192.168.2.157:9442/#/device/0xc498860000978215/settings)
[Settings (specific)](http://192.168.2.157:9442/#/device/0xc498860000978215/settings-specific)
[State](http://192.168.2.157:9442/#/device/0xc498860000978215/state)
[Clusters](http://192.168.2.157:9442/#/device/0xc498860000978215/clusters)
[Scene](http://192.168.2.157:9442/#/device/0xc498860000978215/scene)
[Dev console](http://192.168.2.157:9442/#/device/0xc498860000978215/dev-console)

Friendly name
0xc498860000978215
Description
Last seen
N/A
Availability
[Disabled](https://www.zigbee2mqtt.io/guide/configuration/device-availability.html#availability-advanced-configuration)
Device type
Router
Zigbee Model
RB 255 C
Zigbee Manufacturer
innr
Support status
Not supported

IEEE Address
0xc498860000978215
Network address
0xAE42
Firmware build date
20230208
Firmware version
2.04.01
Power
Interview completed
True

### External converter

```shell
Not tried, sorry not technical enough...
```

### Supported color modes

light (state, brightness, color_temp, color_temp_startup, color_xy, color_hs), effect, power_on_behavior, linkquality

B08Z, 2023-06-12:
> Try changing `zigbeeModel: ['RB 251 C'],` to `zigbeeModel: ['RB 25 5 C'],` and your ext converter should work.
> 
> 

Thanks I have changed
Unfortunately I get an error when I try and use the external converter

● zigbee2mqtt.service - zigbee2mqtt
     Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Mon 2023-06-12 16:51:38 BST; 42ms ago
    Process: 3692 ExecStart=/usr/bin/npm start (code=exited, status=1/FAILURE)
   Main PID: 3692 (code=exited, status=1/FAILURE)
        CPU: 1.174s

Koenkk, 2023-06-12:
Remove this line: ` ota: ota.zigbeeOTA,`

B08Z, 2023-06-13:
> Remove this line: ` ota: ota.zigbeeOTA,`

Ok, so inhave removed the above. 
Zigbee2mqtt starts fine but when I repair the light I get the same error. (Device not supported)

Koenkk, 2023-06-13:
Can you try with `zigbeeModel: ['RB 255 C'],` ? (removed the space)

B08Z, 2023-06-13:
That's what I currently have,
I thought the space was a typo.

Koenkk, 2023-06-13:
You also tried with the space?

B08Z, 2023-06-13:
Doing it now!

B08Z, 2023-06-13:
> You also tried with the space?

Unfortunately neither worked. 

I am not convinced my external converter is actually being used...how can I check?

B08Z, 2023-06-14:
> You also tried with the space?

I have confirmed that:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
         zigbeeModel: ['RB 255 C'],
         model: 'RB 255 C',
         vendor: 'Innr',
         description: 'E14 mini bulb RGBW',
         extend: extend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 555], supportsHueAndSaturation: true}),
         meta: {supportsEnhancedHue: false, applyRedFix: true, turnsOffAtBrightness1: true},
         ota: ota.zigbeeOTA,
};

module.exports = definition;

Works for the E14 mini bulb. 

Thanks for all your help.

closed 2023-06-14 by Koenkk