Koenkk/zigbee2mqtt #17982: TS0601_dimmer_1 - Moes MS-105Z Groups / Scene / Power on behavior issue

marcbernis, 2023-06-10:
### What happened?

Hello,
I've 2 Moes MS-105Z devices which have same issues:
- They do not answer to group actions like scene while being part of the group without error when joining. The device switch to ON or OFF in the Z2M UI but  it does not change its real power state (need to do a ON/OFF cycle manually out of group scene) 
- When reading power on behavior report an issue
Out of this devices  are working ok being able to correctly DIMM Leds
Also picture is wrong for the MOES version as this is a micro module and not a switch (picture attached)
[image]

### What did you expect to happen?

Calling back a scene in a group device should change its ON/OFF status and light should be ON or OFF while only the Z2M UI status change.

### How to reproduce it (minimal and precise)

Add device to a group
Store a scene for this group
Change device ON/OFF status to the opposite
Recall Scene 
UI Change to new status, device doe not do anything and remains in its previous status

For power on behavior click on status refresh and get an error 

### Zigbee2MQTT version

1.31.2-1

### Adapter firmware version

 20221226

### Adapter

SONOFF ZBDONGLE-P

### Debug log

Group Scene Recall All ON (Failed while device appears ON but still OFF with no light)

Debug 2023-06-10 14:23:15Received MQTT message on 'zigbee2mqtt/Chambre/set' with data '{"scene_recall":1}'
Debug 2023-06-10 14:23:15Publishing 'set' 'scene_recall' to 'Chambre'
Debug 2023-06-10 14:23:15Requested color_temp 455 is higher than maximum supported 454, using maximum!
Debug 2023-06-10 14:23:15Requested color_temp 455 is higher than maximum supported 454, using maximum!
Debug 2023-06-10 14:23:15Requested color_temp 455 is higher than maximum supported 454, using maximum!
Info 2023-06-10 14:23:15Successfully recalled group scene
Info 2023-06-10 14:23:15MQTT publish: topic 'zigbee2mqtt/Chambre', payload '{"brightness":254,"color":{"x":0.5056,"y":0.4152},"color_mode":"xy","color_temp":454,"state":"ON"}'
Info 2023-06-10 14:23:15MQTT publish: topic 'zigbee2mqtt/Chevet de la Chambre', payload '{"brightness":254,"color":{"x":0.5056,"y":0.4152},"color_mode":"xy","color_temp":454,"color_temp_startup":454,"linkquality":145,"power_on_behavior":"previous","state":"ON","update":{"installed_version":16786688,"latest_version":16783874,"state":"idle"},"update_available":null}'
Info 2023-06-10 14:23:15MQTT publish: topic 'zigbee2mqtt/Spot de la Lampe de la Chambre', payload '{"brightness":254,"color":{"x":0.5056,"y":0.4152},"color_mode":"xy","color_temp":454,"linkquality":87,"power_on_behavior":"on","state":"ON","update":{"installed_version":16786688,"latest_version":16783874,"state":"idle"},"update_available":null}'
Info 2023-06-10 14:23:15MQTT publish: topic 'zigbee2mqtt/Halogène de la Chambre', payload '{"brightness":254,"color":{"x":0.5056,"y":0.4152},"color_mode":"xy","color_temp":454,"countdown":null,"light_type":"led","linkquality":98,"max_brightness":255,"min_brightness":1,"power_on_behavior":"on","state":"ON"}'
Info 2023-06-10 14:23:15MQTT publish: topic 'zigbee2mqtt/Halogène de la Chambre/type', payload '{"light":"led"}'

Works by manually switching it (Off) the ON (works, looks identical):

Info 2023-06-10 15:37:05MQTT publish: topic 'zigbee2mqtt/Halogène de la Chambre', payload '{"brightness":254,"color":{"x":0.5056,"y":0.4152},"color_mode":"xy","color_temp":454,"countdown":null,"light_type":"led","linkquality":98,"max_brightness":255,"min_brightness":1,"power_on_behavior":"on","state":"ON"}'
Info 2023-06-10 15:37:05MQTT publish: topic 'zigbee2mqtt/Halogène de la Chambre/type', payload '{"light":"led"}'

power_on_behavior ERROR:

Debug 2023-06-10 14:13:52Received MQTT message on 'zigbee2mqtt/Halogène de la Chambre/set' with data '{"power_on_behavior":"on"}'
Debug 2023-06-10 14:13:52Publishing 'set' 'power_on_behavior' to 'Halogène de la Chambre'
Error 2023-06-10 14:14:11Publish 'set' 'power_on_behavior' to 'Halogène de la Chambre' failed: 'Error: Command 0xdc8e95fffeb01013/1 manuSpecificTuya.dataRequest({"seq":1,"dpValues":[{"dp":14,"datatype":2,"data":[0,0,0,1]}]}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'
Debug 2023-06-10 14:14:11Error: Command 0xdc8e95fffeb01013/1 manuSpecificTuya.dataRequest({"seq":1,"dpValues":[{"dp":14,"datatype":2,"data":[0,0,0,1]}]}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:415:23) at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

closed 2023-07-18 by github-actions[bot]

mbratukha, 2023-11-24:
I have troubles with this device too. Is there any chance to get it work with z2m?

tgoetten, 2024-09-08:
I’m having the exact same issue with my Moes MS-105Z dimmer as described here. When part of a group, the device does not respond correctly to on/off actions. The Z2M UI shows the state change, but the physical device doesn’t actually turn on or off unless I manually toggle it.

Another problem I’ve noticed is that sometimes, when using the physical switch connected to the dimmer, instead of simply turning the light on or off, it dims the light in steps, going all the way down to 0, and then back up to 255. To stop this, I have to manually toggle the dimmer on and off through the Z2M UI.

Other than these issues, the dimmer works fine on its own, and I’m able to dim LEDs without any problems.

This is definitely frustrating when trying to control multiple devices through scenes. If anyone has found a workaround or if there’s any update from the manufacturer, it would be great to hear.