Koenkk/zigbee2mqtt #17983: [New device support]: FRIENT - ZIGBEE 3.0 INTELLIGENT SMOKE DETECTOR (DIN EN 14604 AND DIN 14676)

baretd, 2023-06-10:
### Link

https://www.domadoo.fr/en/peripheriques/5409-frient-zigbee-30-intelligent-smoke-detector-5713594002330.html

### Database entry

"nodes": [         {             "definition": null,             "failed": [],             "friendlyName": "Coordinator",             "ieeeAddr": "0x84fd27fffe0e43c2",             "lastSeen": 1686389490903,             "networkAddress": 0,             "type": "Coordinator"         },         {             "definition": {                 "description": "Smart carbon monoxide sensor",                 "model": "HS1CA-M",                 "supports": "carbon_monoxide, battery_low, battery, linkquality",                 "vendor": "HEIMAN"             },             "friendlyName": "0x086bd7fffeaeb2a3",             "ieeeAddr": "0x086bd7fffeaeb2a3",             "lastSeen": 1686388694274,             "manufacturerName": "_TYZB01_wpmo3ja3",             "modelID": "TS0212",             "networkAddress": 63451,             "type": "EndDevice"         },         {             "definition": null,             "failed": [                 "lqi"             ],             "friendlyName": "0x0015bc003100bdba",             "ieeeAddr": "0x0015bc003100bdba",             "lastSeen": 1686389035731,             "networkAddress": 5489,             "type": "Unknown"         },         {             "definition": null,             "failed": [                 "lqi"             ],             "friendlyName": "0x0015bc003100be8f",             "ieeeAddr": "0x0015bc003100be8f",             "lastSeen": 1686388947429,             "networkAddress": 23945,             "type": "Unknown"         }     ]

### Comments

FRIENT - ZIGBEE 3.0 INTELLIGENT SMOKE DETECTOR (DIN EN 14604 AND DIN 14676)

FabriceC01, 2023-06-10:
+1 for this support

Iroamstaff, 2023-06-11:
I was planning to buy this detector, I hope it will be taken care of
 thanks a lot

sjorge, 2023-06-11:
The heat detector variant is currently already support so it should not to too hard to add a converter, but the posted database entry seems to be for a HEIMAN HS1CA-M.

FabriceC01, 2023-06-11:
Hello,
For me the database.db contain for my frient smoke detector but I'm afraid it's much too short :

{"id":15,"type":"Unknown","ieeeAddr":"0x0015bc003100c2b0","nwkAddr":5910,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{},"lastSeen":1686414832145,"defaultSendRequestWhen":"immediate"}

sjorge, 2023-06-12:
The field needed to give it an attempt is `modelId` which is also not listed.

FabriceC01, 2023-06-24:
up

Rappelkiste98, 2023-08-03:
The unit is recognised as "Develco SMSZB-120", but there seems to be an error configuring it. The parameters temperature, reliability & fault cannot be used. The following error is displayed:

Failed to configure 'Rauchmelder 2', attempt 3 (Error: Bind 0x0015bc003101c521/35 genBasic from '0x00212effff08285a/1' failed (Error: status: 140) at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:766:19) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:643:13) at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:43:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/develco.ts:470:13) at Configure.configure (/app/lib/extension/configure.ts:117:13))

cespar, 2023-08-08:
I have the same error, is there a way to fix it?

jsa91, 2023-08-08:
Successfully paired SMSZB-120 with SONOFF zbdongle-E. Firmware 7.3.1.0, build 176. No reported errors.

closed 2023-10-12 by github-actions[bot]

Sweti934, 2023-11-05:
Hi,

are the devices now supported (frient) ? 
https://www.amazon.de/frient-20209600-Intelligent-Batteriebetrieben-funktioniert/dp/B08WXV3G8P/ref=sr_1_1?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=1MYJU4C68BGCI&keywords=SMSZB-120&qid=1699213520&s=diy&sprefix=smszb-120%2Cdiy%2C113&sr=1-1&th=1

thanks so much..

PovilasID, 2023-11-06:
@Sweti934 It works fine just shows up as Develco not frient maybe it is rebrand or something... yesterday I had false positive (1st in 3 months) and handled all alarms and notifications as it should.

garret, 2024-02-04:
I just bought 4 of these Frient smoke detectors from amazon.de (recognized by Z2M as Develco SMSZB-120). They seem to work in general (I can see the smoke state in home assistant) but I often get a configuration error alert on the Z2M webgui and, additionally, the temperature field is always N/A. 

Is this an expected behavior? It has been mentioned in previous comments but I don't understand if these issues can be solved by using a SONOFF zbdongle-E 6.10.3? 

At the moment, I am using mosquitto/Z2M/home assistant on a raspberry pi 3 with a conbee2 usb stick.

PovilasID, 2024-02-05:
I have SONOFF zbdongle-E 6.10.3 and it is been reporting temps rock solid for several months now. However, I have it plugged in x86 (old PC) and a USB 3 port and those can supply a bit more power. 
I am very pleased with update frequency and sensitivity of temperature sensor.